Koenkk/zigbee2mqtt #16783: TS0601 Attribute limits (temperature_calibration)

JuMi2006, 2023-02-22:
### What happened?

Temperature calibration can not set to values higher than a delta of 3 degrees. But the TRV supports up to 10 degrees. I remember in the native app it was possible to set a delta of 5 degrees.
For those TRVs such a big delta is necessary. 

See also: https://github.com/Koenkk/zigbee2mqtt/discussions/15497

### What did you expect to happen?

I can set a offset from 0-10 (+/-) or at min 0-5 (+/-)

### How to reproduce it (minimal and precise)

Setting the offset bigger than creates an error.

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

DennisVonDerBey, 2023-02-22:
The original [suggestion](https://github.com/twhittock/avatto_me167/blob/main/me167.js#L303) that, I think, has been the example implementation (referenced [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5081)) even had a limit of +-10 degrees celsius.

DennisVonDerBey, 2023-02-23:
The original committer has [answered](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5081#issuecomment-1440847856) that it hasn't been tested and the limit has just been copied from the user interface of Tuya.

Is there an easy way to test it with a Home Assistant Addon installation?

DennisVonDerBey, 2023-02-23:
I just tried adding the referenced `me167.js` as an external converter. After restart my devices switched over to that converter and I was able to set the calibration to from -10 to +10 degree celsius. It also successfully changed the exposed temperature. Even with -20 to +20 it worked correctly it seems. @Koenkk do you know of any potential problems doing that? How do other devices behave and or why is there a limit for so many TRVs?

Koenkk, 2023-02-25:
The temperature_calibration limit is determined by the FW of the TRV. I don't know the reason why manufacturers differ the min/max value.

closed 2023-04-04 by github-actions[bot]

Exellent1988, 2023-10-04:
If the fw would limit the Delta, why would you limit it in software then, if there are fws with later possible Delta? I had no issues with the custom Js. But since the official integration is in. I can't use my custom converter anymore. 
How can I resolve that?

IvanRebroff, 2023-10-04:
Same here. The external converter works much better! Another problem: know I have in some TRV's schedule settings. And I can not delete them (Error message ..."invalid schedule. there should be 4 transitions..."
Isn't it possible to prefer external converters?

Exellent1988, 2023-10-04:
But I get the following wir with the external me167.js 
tuya.getDataValue is not a function
Any suggestions?

wookash, 2023-10-06:
> But I get the following wir with the external me167.js tuya.getDataValue is not a function Any suggestions?

I have exactly the same issue when using me167.js file. Can the calibration limits be changed in z2m? Alternatively, how do I fix 'tuya.getDataValue is not a function' error?