Koenkk/zigbee2mqtt #16793: Documentation "Flashing the CC2531 stick" contains wrong information about firmware-file

markus-lorenz, 2023-02-23:
### What happened?

The information about which firmware to use to flash the CC2531-USB-stick are doubled.

**Correct**
On page [Sniff Zigbee traffic](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)
With CC2531 > 1. Flashing the CC2531 stick > Windows (and possibly Ubuntu).
Mentioned is the firmware included in ZBOSS.

**Incorrect**
On page [Flashing the CC2531 USB stick](https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_the_cc2531.html)
Windows > Step 6: "Download the firmware [CC2531_DEFAULT_20211115.zip](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20211115.zip)".

I think under step 6 it should point again to the ZBOSS-included hex-file.

### Zigbee2MQTT version

-

### Adapter firmware version

-

### Adapter

-

closed 2023-04-02 by github-actions[bot]

Masterx0202, 2024-12-31:
can you send download link

markus-lorenz, 2025-01-03:
There is no direct download link. You have to register first.
The best link is https://dsr-iot.com/downloads/

Therefor I would copy the whole text with description:
[image]

copy from https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html
copy to: https://www.zigbee2mqtt.io/guide/adapters/flashing/flashing_the_cc2531.html