Koenkk/zigbee2mqtt #16802: ZNJLBL01LM wrong state reported.

michtam, 2023-02-23:
### What happened?

ZNJLBL01LM always reports state **OFF**.
No matter in what states it really is.

### What did you expect to happen?

Received correct state from motor.

### How to reproduce it (minimal and precise)

Publish `{"state": ""}` message on `zigbee2mqtt/NAME/get`.
Or open webui.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

Logs from following actions: https://github.com/Koenkk/zigbee2mqtt/issues/12647#issuecomment-1143623963

```debug 2023-02-23 21:00:34: Received Zigbee message from '0x54ef4410004d0d59', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.curtain.acn002"}' from endpoint 1 with groupID null
debug 2023-02-23 21:00:34: Received Zigbee message from '0x54ef4410004d0d59', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,23,5,33,1,0,9,33,0,1,10,33,239,242,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,100,101,32,94,102,32,128,103,32,16,105,32,2,106,33,77,31,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID null
debug 2023-02-23 21:00:34: lumi.curtain.acn002: Processed buffer into data {"3":23,"5":1,"9":256,"10":62191,"11":0,"12":1,"13":3870,"17":1,"100":100,"101":94,"102":128,"103":16,"105":2,"106":8013,"107":0}
debug 2023-02-23 21:00:34: lumi.curtain.acn002: unknown key 13 with value 3870
debug 2023-02-23 21:00:34: lumi.curtain.acn002: unknown key 17 with value 1
debug 2023-02-23 21:00:34: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":0,"state":"OFF","battery":94}
info  2023-02-23 21:00:34: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:00:34+01:00","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":0,"running":false,"state":"OFF"}'
debug 2023-02-23 21:00:34: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":0,"state":"OFF","battery":94}
info  2023-02-23 21:00:34: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:00:34+01:00","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":0,"running":false,"state":"OFF"}'
debug 2023-02-23 21:00:49: Received Zigbee message from '0x54ef4410004d0d59', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,23,5,33,1,0,9,33,0,1,10,33,239,242,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,100,101,32,94,102,32,128,103,32,16,105,32,2,106,33,73,31,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID null
debug 2023-02-23 21:00:49: lumi.curtain.acn002: Processed buffer into data {"3":23,"5":1,"9":256,"10":62191,"11":0,"12":1,"13":3870,"17":1,"100":100,"101":94,"102":128,"103":16,"105":2,"106":8009,"107":0}
debug 2023-02-23 21:00:49: lumi.curtain.acn002: unknown key 13 with value 3870
debug 2023-02-23 21:00:49: lumi.curtain.acn002: unknown key 17 with value 1
debug 2023-02-23 21:00:49: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":0,"state":"OFF","battery":94}
debug 2023-02-23 21:00:49: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":0,"state":"OFF","battery":94}
info  2023-02-23 21:00:49: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:00:49+01:00","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":0,"running":false,"state":"OFF"}'
debug 2023-02-23 21:00:50: Received MQTT message on 'zigbee2mqtt/0x54ef4410004d0d59/set' with data '{"position":0}'
debug 2023-02-23 21:00:50: Publishing 'set' 'position' to '0x54ef4410004d0d59'
info  2023-02-23 21:00:51: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:00:51+01:00","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":0,"running":false,"state":"OFF"}'
debug 2023-02-23 21:00:51: Received Zigbee message from '0x54ef4410004d0d59', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":117440522,"presentValue":0}' from endpoint 1 with groupID null
info  2023-02-23 21:00:51: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:00:51+01:00","linkquality":255,"motor_state":"declining","position":100,"power_outage_count":0,"running":true,"state":"OFF"}'
debug 2023-02-23 21:01:08: Received Zigbee message from '0x54ef4410004d0d59', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":15159552,"presentValue":2}' from endpoint 1 with groupID null
info  2023-02-23 21:01:08: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:01:08+01:00","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":0,"running":false,"state":"OFF"}'
debug 2023-02-23 21:01:08: Received Zigbee message from '0x54ef4410004d0d59', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID null
info  2023-02-23 21:01:08: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:01:08+01:00","linkquality":255,"motor_state":"pause","position":0,"power_outage_count":0,"running":false,"state":"OFF"}'
debug 2023-02-23 21:01:08: Received Zigbee message from '0x54ef4410004d0d59', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":0}' from endpoint 1 with groupID null
info  2023-02-23 21:01:08: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":94,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","z [... cut ...]
info  2023-02-23 21:02:48: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004d0d59', payload '{"battery":93,"charging_status":false,"device":{"applicationVersion":30,"dateCode":"10-09-2022","friendlyName":"0x54ef4410004d0d59","hardwareVersion":1,"ieeeAddr":"0x54ef4410004d0d59","manufacturerID":4447,"manufacturerName":"LUMI","model":"ZNJLBL01LM","networkAddress":59217,"powerSource":"Battery","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":23,"last_seen":"2023-02-23T21:02:48+01:00","linkquality":255,"motor_state":"pause","position":0,"power_outage_count":0,"running":false,"state":"OFF"}'
```

mamrai1, 2023-02-24:
have you repaired the device after upgrading conbee 2?

michtam, 2023-02-24:
> have you repaired the device after upgrading conbee 2?

Yes, I did.

michtam, 2023-03-27:
great, I can't remove label.
I have to put comment 🤦

closed 2023-09-09 by github-actions[bot]