Koenkk/zigbee2mqtt #16804: TS110E Dimmer has many issues with brightness

chojnicki, 2023-02-23:
### What happened?

This device is supported by #15372, but for me it has many issues.

1. Refreshing state when dimmer is off, causes showing it as on in dashboard, even that is is really off - or it just sets on and brightness to 0, not sure.
2. After that, when I really turn state on - brightness is set to 0, so light does not turn on.
3. After turning off, it also changes brightness slider to 0 (which does not happen in any other dimmer or light bulb in Z2M).
4. After changing brightness, it changes value -1, so if I set 200, it jumps to 199 after second.
5. In [#15372](https://github.com/Koenkk/zigbee2mqtt/issues/15372) it was mentioned that power on behavior is not supported so it will be removed. But in current version it is exposed so I tested it since is must have feature for me. Setting is saved successfully and persisted. If I set state to on and brightness to 50%, after cutting power it brings back state on and 50%. But after setting state off, it recovers with state on and brightness 0%. So indeed light is off, but I cannot toggle lights anymore since brightness is always set to 0 :/ So this is related to 1 and 2. For me it looks like indeed device supports power_on_behavior, but again something is messed up with brightness. 

Probably other issues, it's hard do describe because it is chaos, everything related to brightness so maybe single fix would fix them all ;)

Some issues were fixed by @Koenkk already for similar device here [#11853](https://github.com/Koenkk/zigbee2mqtt/issues/11853)

### Zigbee2MQTT version

1.30.1  tried also edge

### Adapter firmware version

0x26680700

### Adapter

ConBee2

### Debug log

Refresh state:

Debug 2023-02-23 23:47:49Publishing get 'get' 'state' to 'living_room_ceiling_lamp'
Debug 2023-02-23 23:47:49Received Zigbee message from 'living_room_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-02-23 23:47:49MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_lamp', payload '{"brightness":0,"last_seen":"2023-02-23T22:47:49.798Z","linkquality":119,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'

Changing brightness:
Info 2023-02-23 23:49:43MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_lamp', payload '{"brightness":185,"last_seen":"2023-02-23T22:49:43.590Z","linkquality":111,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Debug 2023-02-23 23:49:44Received Zigbee message from 'living_room_ceiling_lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":726,"currentLevel":185}' from endpoint 1 with groupID null
Info 2023-02-23 23:49:44MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_lamp', payload '{"brightness":184,"last_seen":"2023-02-23T22:49:44.411Z","linkquality":119,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'

Berlusconi13, 2023-02-25:
Device works quite ok for me.
@Koenkk Could you expose min_brightness and max_brightness so that can be set in the menu?
This following converter works also/maybe even better for that device - just a few tweaks and you‘re fine.
I just noticed that there‘s still an issue with the min/max brightness.
The brightnes slider has to be limited to the min/max values, otherwise it wont work correct. Unfortunately I didn‘t get any further.

https://github.com/Koenkk/zigbee2mqtt/issues/15120

Berlusconi13, 2023-02-25:
Did you try it with the latest converter?

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    TS110E: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64515')) {
                result['min_brightness'] = utils.mapNumberRange(msg.data['64515'], 0, 1000, 1, 255);
            }
            if (msg.data.hasOwnProperty('64516')) {
                result['max_brightness'] = utils.mapNumberRange(msg.data['64516'], 0, 1000, 1, 255);
            }
            if (msg.data.hasOwnProperty('61440')) {
                result['brightness'] = utils.mapNumberRange(msg.data['61440'], 0, 1000, 0, 254);
            }
            return result;
        },
    },
    TS110E_light_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64514')) {
                const lookup = {0: 'led', 1: 'incandescent', 2: 'halogen'};
                result['light_type'] = lookup[msg.data['64514']];
            }
            return result;
        },
    },
    TS110E_switch_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64514')) {
                const lookup = {0: 'momentary', 1: 'toggle', 2: 'state'};
                const propertyName = utils.postfixWithEndpointName('switch_type', msg, model, meta);
                result[propertyName] = lookup[msg.data['64514']];
            }
            return result;
        },
    },
};

const tzLocal = {
    TS110E_options: {
        key: ['min_brightness', 'max_brightness', 'light_type', 'switch_type'],
        convertSet: async (entity, key, value, meta) => {
            let payload = null;
            if (key === 'min_brightness' || key == 'max_brightness') {
                const id = key === 'min_brightness' ? 64515 : 64516;
                payload = {[id]: {value: utils.mapNumberRange(value, 1, 255, 0, 1000), type: 0x21}};
            } else if (key === 'light_type' || key === 'switch_type') {
                const lookup = key === 'light_type' ? {led: 0, incandescent: 1, halogen: 2} : {momentary: 0, toggle: 1, state: 2};
                payload = {64514: {value: lookup[value], type: 0x20}};
[... 46 lines cut ...]
    },
};

const definition = {
        fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_zxbtub8r', '_TZ3210_k1msuvg6']),
        model: 'TS110E_1gang_1',
        vendor: 'TuYa',
        description: '1 channel dimmer',
        fromZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
            .fromZigbee.concat([tuya.fz.power_on_behavior, fzLocal.TS110E_switch_type, fzLocal.TS110E]),
        toZigbee: utils.replaceInArray(
            extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
                .toZigbee.concat([tuya.tz.power_on_behavior, tzLocal.TS110E_options]),
            [tz.light_onoff_brightness],
            [tzLocal.TS110E_light_onoff_brightness],
        ),
        exposes: [e.light_brightness().withMinBrightness().withMaxBrightness(), e.power_on_behavior(), tuya.exposes.switchType(), tuya.exposes.lightType()],
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
    };

module.exports = definition;`

Berlusconi13, 2023-02-25:
Did you try it with the latest converter?
`````const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    TS110E: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64515')) {
                result['min_brightness'] = utils.mapNumberRange(msg.data['64515'], 0, 1000, 1, 255);
            }
            if (msg.data.hasOwnProperty('64516')) {
                result['max_brightness'] = utils.mapNumberRange(msg.data['64516'], 0, 1000, 1, 255);
            }
            if (msg.data.hasOwnProperty('61440')) {
                result['brightness'] = utils.mapNumberRange(msg.data['61440'], 0, 1000, 0, 254);
            }
            return result;
        },
    },
    TS110E_light_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64514')) {
                const lookup = {0: 'led', 1: 'incandescent', 2: 'halogen'};
                result['light_type'] = lookup[msg.data['64514']];
            }
            return result;
        },
    },
    TS110E_switch_type: {
        cluster: 'genLevelCtrl',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('64514')) {
                const lookup = {0: 'momentary', 1: 'toggle', 2: 'state'};
                const propertyName = utils.postfixWithEndpointName('switch_type', msg, model, meta);
                result[propertyName] = lookup[msg.data['64514']];
            }
            return result;
        },
    },
};

const tzLocal = {
    TS110E_options: {
        key: ['min_brightness', 'max_brightness', 'light_type', 'switch_type'],
        convertSet: async (entity, key, value, meta) => {
            let payload = null;
            if (key === 'min_brightness' || key == 'max_brightness') {
                const id = key === 'min_brightness' ? 64515 : 64516;
                payload = {[id]: {value: utils.mapNumberRange(value, 1, 255, 0, 1000), type: 0x21}};
            } else if (key === 'light_type' || key === 'switch_type') {
                const lookup = key === 'light_type' ? {led: 0, incandescent: 1, halogen: 2} : {momentary: 0, toggle: 1, state: 2};
                payload = {64514: {value: lookup[value], type: 0x20}};
            }
[... 45 lines cut ...]
    },
};

const definition = {
        fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_zxbtub8r', '_TZ3210_k1msuvg6']),
        model: 'TS110E_1gang_1',
        vendor: 'TuYa',
        description: '1 channel dimmer',
        fromZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
            .fromZigbee.concat([tuya.fz.power_on_behavior, fzLocal.TS110E_switch_type, fzLocal.TS110E]),
        toZigbee: utils.replaceInArray(
            extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
                .toZigbee.concat([tuya.tz.power_on_behavior, tzLocal.TS110E_options]),
            [tz.light_onoff_brightness],
            [tzLocal.TS110E_light_onoff_brightness],
        ),
        exposes: [e.light_brightness().withMinBrightness().withMaxBrightness(), e.power_on_behavior(), tuya.exposes.switchType(), tuya.exposes.lightType()],
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
    };

module.exports = definition;`````

Berlusconi13, 2023-02-25:
For me it seems, that brightness in that case is the 'normal' brightness setting AND the power_on_brightness at the same time.

Berlusconi13, 2023-02-25:
Does anybody know a way to limit the brightness slider? I set min- and max_brightness, but still can go below min/max treshold with the normal brightness slider.
The range should be either 0 or a value between min and max.

chojnicki, 2023-02-26:
> Did you try it with the latest converter?
> 

@Berlusconi13 I tried edge branch without success, but now I tried this converter and yeah, it's better (thank you!) but there are still some problems, and even new ones.

1. If I set brightness, it still decreases -1 by itself.
2. If I refresh brightness, it shows 254 which is not true (light still stays at correct brightness level).
3. If I turn light off, it still shows brightness level as 0, which does not happen in any other dimmer/bulb/controller I have in Z2M.

This is that I got by quick testing, maybe there are more errors too.

> Does anybody know a way to limit the brightness slider? I set min- and max_brightness, but still can go below min/max treshold with the normal brightness slider.

This is annoying for me too, but I tested other dimmers with min/max and it was never respected in Z2M and Home Assistant dashboard cards :/ Only in manual usage. I think it could be limited easy in converter, but I'm not sure about Z2M dashboard. It would be great if Z2M instead limiting brightness range, it could actually calculate correct range and send that to device. So If we have min_brightness set to 100 and max to 200, after setting in Z2M or HA brightness value to 1, it could send 100 to device, 255 would send 200, and 127 (50%) = 150 etc. I do not need max_brightness, but min_brightness is must for dimmable bulbs. Every dimmable led bulb has different level when it actually powers on. In my case I need brightness 65 to start, if I dim it lower it's just off and misleading. For now I'm fixing this somehow by nodered automations, but it's annoying and mess. Would be great if we could just use 1-255 range just like with led bulbs and led strips controllers. Is is maybe already option somewhere, implemented in some device? What do you think @Koenkk?

Berlusconi13, 2023-02-26:
@chojnicki 
There‘s a way to limit min/max brightness through a template in homeassistant.
Check that thread: 
https://community.home-assistant.io/t/template-light-limit-maximum-and-minimum-brightness-of-light/246128

Give it a friendly name and use it for all the stuff. That‘s how I solve it until now.

closed 2023-04-07 by github-actions[bot]

enricodeleo, 2023-04-19:
Same issue (I think) here with my [TS110E_1gang_1](https://www.zigbee2mqtt.io/devices/TS110E_1gang_1.html#tuya-ts110e_1gang_1) devices. Weirdly enough if I turn on or off a light from home assistant or Alexa (both are always as expected) the button works, it seems like it changes the brightness on its own after a while (dunno exactly how long it takes but often enough to make buttons unreliable).

I tried changing power on behavior, disabling legacy home assistant actions, enabling state actions but nothing changes. Any ideas?