Koenkk/zigbee2mqtt #16808: [New device support]: Ikea E2123

djansen1987, 2023-02-24:
### Link

https://www.ikea.com/nl/en/p/symfonisk-sound-remote-gen-2-90524607/

### Database entry

{"id":72,"type":"EndDevice","ieeeAddr":"0x94deb8fffe5d6381","nwkAddr":14104,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"SYMFONISK sound remote gen2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,1,3,32,4096,64599],"outClusterList":[3,4,6,8,25,4096,64639],"clusters":{"genBasic":{"attributes":{"swBuildId":"1.0.012"}},"genPollCtrl":{"attributes":{"checkinInterval":172800}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002281a561","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20211214","swBuildId":"1.0.012","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1677249054005,"defaultSendRequestWhen":"active"}

### Comments

https://www.ikea.com/ma/fr/manuals/symfonisk-sound-remote-gen-2__AA-2356375-1-101.pdf

djansen1987, 2023-02-24:
i have updated the External converter, it is not yet perfect as the dot buttons are not yet working and this is my first attempt to add an device.

Below i specified the output per button:

```
2023-02-24 17:50:50No converter available for 'E2123' with cluster 'genOnOff' and type 'commandToggle' and data '{}'
Play

2023-02-24 17:31:32No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":1,"stepsize":1,"transtime":0}'
previous

2023-02-24 17:31:33No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":0,"stepsize":1,"transtime":0}'
Next

2023-02-24 17:39:49No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":1,"rate":255}'
Volume Up

2023-02-24 17:39:49No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":0,"rate":255}'
Volume Down

2023-02-24 17:45:52No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,2,1,2,1],"type":"Buffer"}'
1 dot

2023-02-24 17:45:53No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,3,1,1,1],"type":"Buffer"}'
2 dots
```

djansen1987, 2023-02-24:
[image]

Koenkk, 2023-02-25:
Good work! For some of the events you can use standard converters: `fz.command_move, fz.command_toggle, fz.command_step`

bjekel, 2023-02-25:
I was also trying to get the E2123 to work, but didn't get any further then the play/pause button and battery.

After seeing @djansen1987's solution I've copy/pasted my battery solution in there. It's definitely not a clean solution, but this is my first attempt. Also not sure if the custom battery converter is needed, but it looked like all ikea devices use it.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const ikea = {
    fz: {
        ikea_dots_click: {
            cluster: '64639',
            type: 'raw',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.data === '[21,124,17,2,1,2,1]' ? '1' : '2';
                return {action: `dots_${direction}`};
            },
        },
        ikea_volume_click: {
            cluster: 'genLevelCtrl',
            type: 'commandMoveWithOnOff',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.movemode === 1 ? 'down' : 'up';
                return {action: `volume_${direction}`};
            },
        },
        ikea_track_click: {
            cluster: 'genLevelCtrl',
            type: 'commandStep',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.stepmode === 1 ? 'previous' : 'next';
                return {action: `track_${direction}`};
            },
        },
        ikea_track_play: {
            cluster: 'genOnOff',
            type: 'commandToggle',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                const arrowReleaseAgo = Date.now() - globalStore.getValue(msg.endpoint, 'track_play', 0);
                if (arrowReleaseAgo > 700) {
                    return {action: 'track_play'};
                }
            },
        },
        battery: {
            cluster: 'genPowerCfg',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const payload = {};
                if (msg.data.hasOwnProperty('batteryPercentageRemaining')) {
                    // Some devices do not comply to the ZCL and report a
                    // batteryPercentageRemaining of 100 when the battery is full (should be 200).
                    //
                    // IKEA corrected this on newer remote fw version, but many people are still
                    // 2.2.010 which is the last version supporting group bindings. We try to be
                    // smart and pick the correct one for IKEA remotes.
                    let dontDividePercentage = false;
                    let percentage = msg.data['batteryPercentageRemaining'];
                    const fwVer = meta.device.softwareBuildID.split('.');
                    if ((fwVer[0] < 2) || (fwVer[0] == 2 && fwVer[1] <= 3)) {
                        dontDividePercentage = true;
                    }
                    percentage = dontDividePercentage ? percentage : percentage / 2;
                    payload.battery = precisionRound(percentage, 2);
                }

                return payload;
            },
        }
    },
};

const definition = {
    zigbeeModel: ['SYMFONISK sound remote gen2'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'E2123', // Vendor model number, look on the device for a model number
    vendor: 'Ikea', // Vendor of the device (only used for documentation and startup logging)
    description: 'SYMFONISK sound remote gen2', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [ikea.fz.battery, ikea.fz.ikea_track_play, ikea.fz.ikea_track_click, ikea.fz.ikea_volume_click, ikea.fz.ikea_dots_click],
    exposes: [e.battery().withAccess(ea.STATE_GET), e.action(['track_play','track_previous','track_next','volume_up','volume_down','dots_1','dots_2'])],
    toZigbee: [tz.battery_percentage_remaining],
    ota: ota.tradfri,
};

module.exports = definition;
```

rendyhd, 2023-02-26:
I didn't get the battery to work, but do have all the other buttons working with this external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const ikea = {
    fz: {
        dot_clicks: {
            cluster: 64639,
            type: ['raw'],
            convert: (model, msg, publish, options, meta) => {
                const button_lookup = {
                    1: 'dot1',
                    2: 'dot2',
                };
                const action_lookup = {
                    1: 'single',
                    2: 'double',
                    3: 'hold',
                };
                const button = button_lookup[msg.data[5]];
                const action = action_lookup[msg.data[6]];
                if (button) {
                    return {action: `${button}_${action}`};
                }
            },
        },
    },
};

const definition = {
    zigbeeModel: ['SYMFONISK sound remote gen2'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'E2123', // Vendor model number, look on the device for a model number
    vendor: 'Ikea', // Vendor of the device (only used for documentation and startup logging)
    description: 'SYMFONISK sound remote gen2', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_toggle, fz.command_move, fz.command_step, ikea.fz.dot_clicks],
    exposes: [e.action(['toggle', 'brightness_move_down','brightness_move_up','brightness_step_down','brightness_step_up', 'dot1_single', 'dot1_double', 'dot1_hold', 'dot2_single', 'dot2_double', 'dot2_hold'])],
    toZigbee: [tz.battery_percentage_remaining],
    // ota: ota.tradfri,
};

module.exports = definition;
```

Volume up -> brightness_move_up
Volume down -> brightness_move_down
Prev track -> brightness_step_down
Next track -> brightness_step_up
Play / Pause -> toggle
Dots -> 'dot1_single', 'dot1_double', 'dot1_hold', 'dot2_single', 'dot2_double', 'dot2_hold'

This is my first time writing JS, so might not be optimal. Thanks @Koenkk for all the code I could copy from :)

guilhermednt, 2023-02-26:
I just bought this controller Friday and was very happy to see people are already working on it! Thanks!

On the implementation itself, I'd like to provide some input, however please keep in mind it's my first time experimenting with these converters, so I'm still trying to wrap my hear around it.

On @rendyhd we have the "dots" working perfectly, however I disagree on how the volume up/down and next/previous track buttons are implemented.

First of all in my opinion these buttons should have an action named according: `next_track`, `previous_track`, `volume_move_up`, `volume_step_up`, `volume_move_down` and `volume_step_down` (the names are mere suggestions, I'm not familiar with eventual naming conventions in this repository)

My next point is related to the volume buttons. In my opinion they each have two different actions: click and hold, like IKEA's small rocker controllers, so it'd be nice if we could expose these differently likewise (hence the `volume_move_*` and `volume_step_*` I mentioned previously)

I'll try to come up with something to address those in the next days.

Again, thanks a lot for the awesome collaboration!

rendyhd, 2023-02-26:
@guilhermednt I hear you with the naming. 
The remote uses default Zigbee commands (fz.command_toggle, fz.command_move, fz.command_step) that are already defined [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js
). I don't know how to how to change the naming without duplicating it. 
I followed the practice that was also used for the previous Symfonisk remote ([E1744](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/ikea.js)) by just exposing brightness.

I created a blueprint in Home Assistant this morning and the remote works perfectly. Holding or pressing volume does it what it needs to without having to split up the commands.

Hope this information helps in your research, good luck!

guilhermednt, 2023-02-26:
@rendyhd I meant that the volume in your converter is not behaving like the brightness button on other IKEA remotes I have.

With the existing remotes I have when I hold the brightness up, for example, I get a single `brightness_move_up` action and then when I release the button I get a single `brightness_stop`.

With your converter when I hold the volume up I get multiple `brightness_move_up`, also when I click it once I also get a `brightness_move_up`, so I can't differentiate between two different actions.

Naming aside, I'd be happy to have the `brightness_step_up` and `brightness_move_up`+`brightness_stop` for such cases when using the volume up (click vs hold respectively).

This would allow, for example, to change the volume either faster or slower than the step used for the click action, or perform other actions entirely.

guilhermednt, 2023-02-26:
After checking the Zigbee messages from the E2123 it looks like it doesn't behave like other IKEA controllers, so implementing the `_stop` logic would be extra tricky. 😢 

Unlike the E2001/E2002 controller that won't flood the network with `commandMove` messages when we hold the brightness up/down buttons, the E2123 will.

Implementing a start/stop logic similar to E2001 would require the converter to detect the "flood" and throw a single action when it's detected and then a `_stop` when it stops. Not sure if it'd be adequate for Z2M to do that.

I guess I'll just implement it manually on Node RED instead then.

djansen1987, 2023-03-02:
I didn't yet have time to implement it, but the code for rolling/holding volume is below:

Volume Up Hold:
No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandMove' and data '{"movemode":0,"rate":255}'

Volume down Hold:
No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandMove' and data '{"movemode":1,"rate":255}'

djansen1987, 2023-03-02:
Added the Hold Function:

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const globalStore = require('zigbee-herdsman-converters/lib/store');

const ikea = {  
    fz: {
```
        ikea_dots_click: {
            cluster: '64639',
            type: 'raw',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.data === '[21,124,17,2,1,2,1]' ? '1' : '2';
                return {action: `dots_${direction}`};
            },
        },
        ikea_volume_click: {
            cluster: 'genLevelCtrl',
            type: 'commandMoveWithOnOff',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.movemode === 1 ? 'down' : 'up';
                return {action: `volume_${direction}`};
            },
        },
        ikea_volume_hold: {
            cluster: 'genLevelCtrl',
            type: 'commandMove',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.movemode === 1 ? 'down_hold' : 'up_hold';
                return {action: `volume_${direction}`};
            },
        },
        ikea_track_click: {
            cluster: 'genLevelCtrl',
            type: 'commandStep',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                if (msg.data.value === 2) {
                    // This is send on toggle hold, ignore it as a toggle_hold is already handled above.
                    return;
                }
    
                const direction = msg.data.stepmode === 1 ? 'previous' : 'next';
                return {action: `track_${direction}`};
            },
        },
        ikea_track_play: {
            cluster: 'genOnOff',
            type: 'commandToggle',
            convert: (model, msg, publish, options, meta) => {
                if (utils.hasAlreadyProcessedMessage(msg, model)) return;
                const arrowReleaseAgo = Date.now() - globalStore.getValue(msg.endpoint, 'track_play', 0);
                if (arrowReleaseAgo > 700) {
                    return {action: 'track_play'};
                }
            },
        },
    },
};

const definition = {
    zigbeeModel: ['SYMFONISK sound remote gen2'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'E2123', // Vendor model number, look on the device for a model number
    vendor: 'Ikea', // Vendor of the device (only used for documentation and startup logging)
    description: 'SYMFONISK sound remote gen2', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [ikea.fz.ikea_track_play, ikea.fz.ikea_track_click, ikea.fz.ikea_volume_click,ikea.fz.ikea_volume_hold, ikea.fz.ikea_dots_click],
    exposes: [e.action(['track_play','track_previous','track_next','volume_up','volume_down','volume_up_hold','volume_down_hold','dots_1','dots_2'])],
    toZigbee: [],
    ota: ota.tradfri,
};

module.exports = definition;

// 2023-02-24 17:50:50No converter available for 'E2123' with cluster 'genOnOff' and type 'commandToggle' and data '{}'
// Play

// 2023-02-24 17:31:32No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":1,"stepsize":1,"transtime":0}'
// previous

// 2023-02-24 17:31:33No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandStep' and data '{"stepmode":0,"stepsize":1,"transtime":0}'
// Next

// 2023-02-24 17:39:49No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":1,"rate":255}'
// Volume Up

// 2023-02-24 17:39:49No converter available for 'E2123' with cluster 'genLevelCtrl' and type 'commandMoveWithOnOff' and data '{"movemode":0,"rate":255}'
// Volume Down

// 2023-02-24 17:45:52No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,2,1,2,1],"type":"Buffer"}'
// 1 dot

// 2023-02-24 17:45:53No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,3,1,1,1],"type":"Buffer"}'
// 2 dots

darkkatarsis, 2023-03-03:
@djansen1987 and what about battery ? He has been trying since yesterday, but unfortunately without success

djansen1987, 2023-03-03:
@darkkatarsis  i did not yet get the battery working. Currently do not as much time for it as i like. Hoping we can combine code from people here to make a complete code.

djansen1987, 2023-03-05:
Hi @Koenkk, could you guid me on how to handle RAW data ?
 
 Below the return of the same 2 buttons pressed at different times:
 
```
// Debug 2023-03-05 14:28:15Received Zigbee message from 'Media Remote', type 'raw', cluster '64639', data '{"data":[21,124,17,31,1,1,1],"type":"Buffer"}' from endpoint 1 with groupID 0
// Debug 2023-03-05 14:28:15No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,31,1,1,1],"type":"Buffer"}'
// Debug 2023-03-05 14:28:16Received Zigbee message from 'Media Remote', type 'raw', cluster '64639', data '{"data":[21,124,17,32,1,2,1],"type":"Buffer"}' from endpoint 1 with groupID 0
// Debug 2023-03-05 14:28:16No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,32,1,2,1],"type":"Buffer"}'

// 2023-03-05 15:13:05Received Zigbee message from 'Media Remote', type 'raw', cluster '64639', data '{"data":[21,124,17,48,1,1,1],"type":"Buffer"}' from endpoint 1 with groupID 0
// 2023-03-05 15:13:05No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,48,1,1,1],"type":"Buffer"}'
// 2023-03-05 15:13:06Received Zigbee message from 'Media Remote', type 'raw', cluster '64639', data '{"data":[21,124,17,49,1,2,1],"type":"Buffer"}' from endpoint 1 with groupID 0
// 2023-03-05 15:13:06No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,49,1,2,1],"type":"Buffer"}'

// 2023-02-24 17:45:52No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,2,1,2,1],"type":"Buffer"}'
// 1 dot

// 2023-02-24 17:45:53No converter available for 'E2123' with cluster '64639' and type 'raw' and data '{"data":[21,124,17,3,1,1,1],"type":"Buffer"}'
// 2 dots
```

rendyhd, 2023-03-05:
> @djansen1987 and what about battery ? Im trying since yesterday, but unfortunately without success

I wonder if they expose it, I've been looking more and can't find it. Don't have a IKEA hub to check, but in [their documentation](https://manuals.plus/ikea/symfonisk-2nd-gen-sound-remote-manual) they mention that it'll last two years and then blink red on the remote itself.
```
Changing the batteries
When SYMFONISK sound remote is used regularly and as intended, the batteries will last for approximately 2 years. When it’s time to replace the batteries, the red LED indicator on the back of the sound remote will flash when you use it.
```

@djansen1987 in the code I posted last week there's a solution for the dots. 

Been using the remote in Home Assistant for a week now with the code I posted above and it's really like this device for 15€. It's always on our dining table and we use it multiple times a day. Besides volume and track control, using the dots in single press for light scenes, double press for 2 playlists, and hold for grouping/ungrouping.

djansen1987, 2023-03-05:
@rendyhd Thanks for your feedback, will have another try but I was getting errors when putting in your code. Will post once I know the outcome.

It is indeed a very nice device! but for sure different then a light switch so some function need to be redesigned if you ask me (Volume instead brightness)

rendyhd, 2023-03-05:
@djansen1987 Let me share to help you trouble shoot your errors, I don't know if these raw data dumps can differ per remote:

type 'raw', cluster '64639', data '{"data":[21,124,17,46,1,1,1],"type":"Buffer"}' -> Dot 1, single
type 'raw', cluster '64639', data '{"data":[21,124,17,47,1,1,2],"type":"Buffer"}' -> Dot 1, double
type 'raw', cluster '64639', data '{"data":[21,124,17,48,1,1,3],"type":"Buffer"}' -> Dot 1, hold
type 'raw', cluster '64639', data '{"data":[21,124,17,50,1,2,1],"type":"Buffer"}' -> Dot 2, single
type 'raw', cluster '64639', data '{"data":[21,124,17,51,1,2,2],"type":"Buffer"}' -> Dot 2, double
type 'raw', cluster '64639', data '{"data":[21,124,17,52,1,2,3],"type":"Buffer"}' -> Dot 2, hold

Based on that:
Second to last digit (msg[5]) = Dot #
Last digit (msg[6]) = 1:single, 2:double, 3:hold

The code used around it I copied from the [fromZigbee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/419b500022c62353108f0f7b15834bde264a8c6c/converters/fromZigbee.js)

My other references are:
- The Zigbee commands (fz.command_toggle, fz.command_move, fz.command_step) from [this file](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js). 
- Symfonisk remote ([E1744](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/ikea.js))

Hope that helps you!

MattWestb, 2023-03-05:
If updating the device with Dirigera or the zigpy sniffed OT update its Surtcut clusters is 95% Matter compatible and shall working with HAs Matter integration.
The Matter commands is:
```
1.11.7. Events
This cluster SHALL support these events:
Id Name Access Conformance
0 SwitchLatched V LS
1 InitialPress V MS
2 LongPress V MSL
3 ShortRelease V MSR
4 LongRelease V MSL
5 MultiPressOngoing V MSM
6 MultiPressComplete V MSM
```
Only command 5 is not implanted in firmware version  1.0.32.
Stock firmware is not supported by IKEA TF and Dirigera and is forced update for getting compatible and working.

Koenkk, 2023-03-06:
@djansen1987 you can find an example `raw` converter here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/a654b8426ab13e04cd14c4806b009219c1b56b8d/converters/fromZigbee.js#L3107

flowcool, 2023-03-09:
Hello Everyone,

Just for my understanding as I received mine and I'm puzzled.
Are you able to pair it at least with Z2M ? 
The small light on the remote is not turning on at anytime

Put logs to debug in Z2M and trying to pair, nothing is showing up... I'm wondering if it's "normal" or if my device is not in good shape :/

Thanks !

ilueckel, 2023-03-09:
The manual states to do a long press on the connect button to start pairing. But for me it was 4 or 5 short presses, like with other Ikea remotes. However there is no LED indication that the pairing mode is active.

MattWestb, 2023-03-09:
Long press is tush link pairing = direct with lights or TF GW and 4 fast is normal coordinator pairing mode.

ilueckel, 2023-03-12:
@MattWestb Did you update your remote and connected it with zigbee2mqtt? Do they work together?

MattWestb, 2023-03-12:
Nop I was paring one with Dirigera and was sniffing the update and putting it together with zigpy-CLI so we have one signed OTA file that is working OK.
Im  using both factory (V1) and production (V2) with ZHA firmware and both is working OK and V1 is having 13 device automation and V2 17 in ZHA and its ready for PR to make it in ZHA production.

V2 is is having all normal commands the same as V1 but the chortcut buttons is different and is send from EP 2 and 3 (V1 from EP1) with more or less `Matter Switch` ZCL compatible commands.

Both versions command is being sniffed and documented in ZHA issue for the device support request.

Im sorry but i cant helping adding support for them to Z2M then Java for my is for drinking.

flowcool, 2023-03-13:
> The manual states to do a long press on the connect button to start pairing. But for me it was 4 or 5 short presses, like with other Ikea remotes. However there is no LED indication that the pairing mode is active.

Here we are !! Was following the manual and tried to do 4/5 short presses and.... Tada !! Here we are ! many thanks

cgilis, 2023-03-13:
Nice! Does this mean this device will be supported on the next zigbee2mqtt update?

djansen1987, 2023-03-13:
Look great, hope to find some time tomorrow to test

MattWestb, 2023-03-13:
One very brave user !!!
One warning with this and all new manufactured devices have IKEA disable debug so its **not** possible flashing the devices with one older firmware with SWD probe or dumping it and the firmware for controllers is blocked for downgrading.
Its possible but then unlocking the chip is the flash and RAM erased so you have one dead device and need one copy of one working device (that is not possible dumping so not possible) but we have taking the bootloader from Styrbar and the userdat and patching it and putting the update in it so its working but is not easy and is not possible getting the shipping version only the OTA updated.

Great work @ilueckel fixing the support for the new device and you is getting more to do in some weeks then the new OnOff, OpenClose and ShurtCut button Gen 2 is being released.

FCC application [RODRET](https://fccid.io/FHO-E2201/Letter/Model-Declaration-Letter-6321983)

darkkatarsis, 2023-03-13:
I don't know if I understood. Is it safe to update OTA from zigbee-OTA project ?

MattWestb, 2023-03-13:
The OTA files is 100% OK then IKEA have signing the file and we have getting all parts (from Dirigera) together and the signing is OK so the device is flashing it that is not doing if the signing is broken if some parts is not 100% OK.
Only thing you can **not** reversion it to shipping version (V1).

junosuarez, 2023-03-14:
I picked up one of these devices today, paired it to my zigbee2mqtt 1.30.2 network, and tried the most recent converter code in this thread, https://github.com/Koenkk/zigbee2mqtt/issues/16808#issuecomment-1466899337 - and got zigbee-herdsman-converter errors on startup: 

```
Using '/app/data' as data directory

/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

This is my first time using an External Converter, so there might be something silly I'm doing wrong. I copied the code from https://github.com/Koenkk/zigbee2mqtt/issues/16808#issuecomment-1466899337 into a file called `symfonisk.js` in the same directory as my configuration.yaml. I'm using the zigbee2mqtt docker image, so both my config and symfonisk.js are in `/app/data`. Finally, I added this to my configuration.yaml:
```
external_converters:
  - symfonisk.js
```

If anyone has suggestions, I'm happy to try them out and test this converter on a real device.

ilueckel, 2023-03-14:
Does it start up when you remove the whole `configure`-block?

ilueckel, 2023-03-14:
I updated the code, can you try again?

darkkatarsis, 2023-03-14:
@ilueckel after updating OTA, and updated code from above - Everything looks like it works as it should - including the battery status

MattWestb, 2023-03-14:
PR made for both firmware versions in ZHA and all is working great only device automatons for long press up and down is stalling HA automatons then its sending the command very / to fast :-)))

And the best de(F)CONZ have **not** start implanting the device so 12 points is going to the open Zigbee community for implanting support for it in under 2 weeks !!

flowcool, 2023-03-14:
> external_converters:
>   - symfonisk.js

Hello !

I tried the same (thanks for help @junosuarez - we have the same setup so done the same) ! 

FYI, my firmware is 1.0.012... (and didn't get how to update firmware with comments mentioned in this thread (newbie...)
so rebooted my container and here we are:

`03/14/2023 2:46:17 PM
Zigbee2MQTT:error 2023-03-14 14:46:17: Failed to configure 'Télécommande ikea', attempt 1 (TypeError: Cannot read properties of undefined (reading 'bind')
03/14/2023 2:46:17 PM
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
03/14/2023 2:46:17 PM
    at Object.configure (/app/data/extension/externally-loaded.js:137:25)
03/14/2023 2:46:17 PM
    at Configure.configure (/app/lib/extension/configure.ts:117:13))
03/14/2023 2:46:34 PM
Zigbee2MQTT:error 2023-03-14 14:46:34: Failed to configure 'Télécommande ikea', attempt 2 (TypeError: Cannot read properties of undefined (reading 'bind')
03/14/2023 2:46:34 PM
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
03/14/2023 2:46:34 PM
    at Object.configure (/app/data/extension/externally-loaded.js:137:25)
03/14/2023 2:46:34 PM
    at Configure.configure (/app/lib/extension/configure.ts:117:13))
03/14/2023 2:46:56 PM
Zigbee2MQTT:error 2023-03-14 14:46:56: Failed to check if update available for 'Télécommande ikea' (Device didn't respond to OTA request)`

I hope it helps anyone... :)

MattWestb, 2023-03-14:
The latest `(Device didn't respond to OTA request)` is the devise sleeping the you is sending the command to it. 
Press one button for waking it up then sending command to it so it can receiving it and it shall working OK

flowcool, 2023-03-14:
Unfortunately... doesn't change anything (I put in debug and capture the before to show I pushed button)

```
Debug 2023-03-14 15:16:26Received Zigbee message from 'Télécommande ikea', type 'commandMoveWithOnOff', cluster 'genLevelCtrl', data '{"movemode":0,"rate":255}' from endpoint 1 with groupID 0
Info 2023-03-14 15:16:26MQTT publish: topic 'zigbee2mqtt/Télécommande ikea', payload '{"action":"volume_up","battery":50,"linkquality":156,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-14 15:16:26MQTT publish: topic 'zigbee2mqtt/Télécommande ikea', payload '{"action":"","battery":50,"linkquality":156,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-14 15:16:26MQTT publish: topic 'zigbee2mqtt/Télécommande ikea/action', payload 'volume_up'
Debug 2023-03-14 15:16:32Received Zigbee message from 'Télécommande ikea', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":1,"transtime":0}' from endpoint 1 with groupID 0
Info 2023-03-14 15:16:32MQTT publish: topic 'zigbee2mqtt/Télécommande ikea', payload '{"action":"track_previous","battery":50,"linkquality":148,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-14 15:16:32MQTT publish: topic 'zigbee2mqtt/Télécommande ikea', payload '{"action":"","battery":50,"linkquality":148,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-14 15:16:32MQTT publish: topic 'zigbee2mqtt/Télécommande ikea/action', payload 'track_previous'
Debug 2023-03-14 15:16:39Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Télécommande ikea","transaction":"ox4l8-6"}'
Info 2023-03-14 15:16:39Checking if update available for 'Télécommande ikea'
Info 2023-03-14 15:16:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Télécommande ikea'","meta":{"device":"Télécommande ikea","status":"checking_if_available"},"type":"ota_update"}'
Debug 2023-03-14 15:16:39Check if update available for '0x540f57fffe58b77f' (SYMFONISK sound remote gen2)
Debug 2023-03-14 15:16:39Using endpoint '1'
Debug 2023-03-14 15:17:38Received Zigbee message from 'Télécommande ikea', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":1,"transtime":0}' from endpoint 1 with groupID 0
Info 2023-03-14 15:17:38MQTT publish: topic 'zigbee2mqtt/Télécommande ikea', payload '{"action":"track_previous","battery":50,"linkquality":148,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-14 15:17:38MQTT publish: topic 'zigbee2mqtt/Télécommande ikea', payload '{"action":"","battery":50,"linkquality":148,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-14 15:17:38MQTT publish: topic 'zigbee2mqtt/Télécommande ikea/action', payload 'track_previous'
Info 2023-03-14 15:17:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Télécommande ikea' (Device didn't respond to OTA request)","meta":{"device":"Télécommande ikea","status":"check_failed"},"type":"ota_update"}'
Info 2023-03-14 15:17:39MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Télécommande ikea"},"error":"Failed to check if update available for 'Télécommande ikea' (Device didn't respond to OTA request)","status":"error","transaction":"ox4l8-6"}'
Error 2023-03-14 15:17:39Failed to check if update available for 'Télécommande ikea' (Device didn't respond to OTA request)
Debug 2023-03-14 15:17:39Error: Device didn't respond to OTA request at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:191:15) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:246:25) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
```

hashtagsandro, 2023-03-14:
SYMFONISK sound remote gen2 found with the latest.js posted above and firmware release 1.0.012 ( 20211214 ). Now I'm try to flash ota that after some battery unplug correctly started ( seems the same about other ikea's battery devices ).

MQTT publish: topic 'zigbee2mqtt/0xIKEA_SONOFF_TEST', payload '{"action":null,"battery":null,"linkquality":185,"update":{"installed_version":65554,"latest_version":16777266,"progress":3.13,"remaining":12004,"state":"updating"},"update_available":null}'

junosuarez, 2023-03-15:
I tried the code in . My device is on firmware 20211214 (as it came out of the box. Actions are working for `track_play`, `volume_up`, `volume_down`, and `volume_previous`. I haven't been able to OTA update the firmware, but if I get that working, I'll try the other button modes.

Is it possible to support double click or hold variants for the buttons on this device?

junosuarez, 2023-03-15:
Further debugging why my device "dot" buttons don't work - the first time I press either the 1 dot or 2 dot buttons, I get the following in my logs (running zigbee2mqtt 1.30.2):

```
Zigbee2MQTT:error 2023-03-15 15:16:16: Failed to configure 'c-1036-ikea-media-remote', attempt 1 (TypeError: Cannot read properties of undefined (reading 'bind')
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:24)
    at Object.configure (/app/data/symfonisk:139:25)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Configure.configure (/app/lib/extension/configure.ts:117:13))
```

junosuarez, 2023-03-15:
Ok, a bit more hacking and I got the dot buttons fully working on firmware 20211214!

I decoded some of the buffers I was seeing:

```

single 1 - [ 21, 124,  17, 126,   1,   1,   1]
double 1 - [ 21, 124,  17, 127,   1,   1,   2]
hold 1   - [ 21, 124,  17,   0,   1,   1,   3]
            A    B     C   D    E    F    G
single 2 - [ 21, 124,  17,   2,   1,   2,   1]
double 2 - [ 21, 124,  17,   3,   1,   2,   2]
hold 2   - [ 21, 124,  17,   4,   1,   2,   3]

 
A0 - header? always 0x21
B1 - header? always 0x124
C2 - header? always 0x17
D3 - counter, increments with each press
E4 - ? always 0x01
F5 - which button - 0x01 for 1 dot, 0x02 for 2 dots
G6 - type of press: 0x01 for single, 0x02 for double, 0x03 for hold
```

So here's the relevant part of the converter:
```
        ikea_dots_click_v1: {
            cluster: 64639,
            type: 'raw',
            convert: (model, msg, publish, options, meta) => {
                if (!Buffer.isBuffer(msg.data)) return;

                const button = msg.data.readInt8(5);
                let actionType;
                switch (msg.data.readInt8(6)) {
                        case 1:
                                actionType = 'single';
                                break;
                        case 2:
                                actionType = 'double';
                                break;
                        case 3:
                                actionType = 'hold';
                                break;
                        default:  // unknown press type
                                return;
                        }

                return {action: `dots_${button}_${actionType}`};
            },
        },
```
The other tricky thing is `zigbee-herdsman` requires numeric clusters to be type `number` not `string` - ie `64639` not `'64639'`.

Finally, I added some conditionals to only try binding the endpoints if they're available (on the new firmware) - this prevents the error logs from my earlier comment and makes this converter work with the old firmware (I can't test the new firmware, if anyone has it, please test to ensure it works!)

Putting it all together, and updating the exposes section to match the event names:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const logger = require('/app/dist/util/logger');

const ikea = {
[... 123 lines cut ...]
        model: 'E2123', // Vendor model number, look on the device for a model number
        vendor: 'IKEA', // Vendor of the device (only used for documentation and startup logging)
        description: 'SYMFONISK sound remote gen2', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
        fromZigbee: [ikea.fz.battery, ikea.fz.ikea_track_play, ikea.fz.ikea_track_click, ikea.fz.ikea_volume_click, ikea.fz.ikea_volume_hold, ikea.fz.ikea_dots_click_v1, ikea.fz.ikea_dots_click_v2],
        exposes: [e.battery().withAccess(ea.STATE_GET), e.action(['track_play', 'track_previous', 'track_next', 'volume_up', 'volume_down', 'volume_up_hold', 'volume_down_hold', 'dots_1', 'dots_2', 'dots_1_initial_press', 'dots_2_initial_press', 'dots_1_long_press', 'dots_2_long_press', , 'dots_1_short_release', 'dots_2_short_release', 'dots_1_long_release', 'dots_2_long_release', 'dots_1_double_press', 'dots_2_double_press', 'dots_1_single', 'dots_1_double', 'dots_1_hold', 'dots_2_single', 'dots_2_double', 'dots_2_hold'])],
        toZigbee: [tz.battery_percentage_remaining],
        ota: ota.zigbeeOTA,
        configure: async (device, coordinatorEndpoint, logger) => {
                const endpoint1 = device.getEndpoint(1);
                const endpoint2 = device.getEndpoint(2);
                const endpoint3 = device.getEndpoint(3);
                await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genPollCtrl', 64639]);
                if (endpoint2) {
                        await reporting.bind(endpoint2, coordinatorEndpoint, ['heimanSpecificScenes']);
                }
                if (endpoint3) {
                        await reporting.bind(endpoint3, coordinatorEndpoint, ['heimanSpecificScenes']);
                }
                await reporting.batteryVoltage(endpoint1);
        },
};

module.exports = definition;

```

ilueckel, 2023-03-16:
Thanks @junosuarez for debugging and fixing the code for the stock firmware. I will update the code in my pull request.

ilueckel, 2023-03-16:
@junosuarez Does your remote report battery values?

krusse, 2023-03-16:
I don't know if this is helpful to add the conversation or not as I admittedly am new to this, but...

I updated the firmware to 1.0.32 through OTA. Using either the @junosuarez and @ilueckel converters does not enable the dots buttons for me. They do not show up in the device trigger options. Everything else works including battery reporting. This was the case before I updated the firmware as well. 

Not sure if I am missing something. Does everyone have the dots buttons working?

ilueckel, 2023-03-16:
Does your binding look like this

[image]

Maybe you need to reconfigure your device after the firmware update as the bindings for Endpoint 2 and Endpoint 3 are only available with the updated version

junosuarez, 2023-03-16:
For refrence, here's what my binding looks like, using my device with the stock (old) firmware, configured with the External Converted code I shared above:

[image]

krusse, 2023-03-16:
It looks like the one @junosuarez shows:
[image]

When changing the converter or updating the firmware, is there anything I should be doing aside from restarting z2mqtt?

ilueckel, 2023-03-16:
Try reconfiguring the device
[image]
(orange button). 

If this does not help, try rejoining the device.

mex1254, 2023-03-17:
Hello, 

i am pretty new, so please forgive me my stupid questions. I managed to update my Symfonisk gen2 with OTA and I can see it in z2m as supported device. i also implemented the external converter. i can see my attempts on volume up and down in the logs. Unfortunately they do nothing on my Symfonsik Speaker. Only thing i can do is to Switch on/off my aquara power plug with the "Play" button. So i think i got the binding wrong - where can i fix this.

Thank you in advance

krusse, 2023-03-17:
Rejoining the device worked beautifully. Now, all the buttons work as expected. My bindings now look like the ones @ilueckel posted. Many thanks!

However, now the battery values are not reporting.

shawsky, 2023-03-17:
I noticed the same thing, an earlier iteration of the script on the later firmware showed battery life but since re-adding to pick up the changes it doesn't appear at all.

Good work so far everyone :)

pmfernandes, 2023-03-18:
Hi,
Just to leave feedback about last file.
 - After updating the FW of the remote I'v disabled v1 fromZigbee. Also removed the actions.
 - Battery reporting successfuly 100%

Suming up: Everything working fine here.

Thanks.

NoClaim, 2023-03-20:
I'm totally blond with no hair, but how to update FW through Z2M? Looking for a manual, but can't find (dummy proof)
Thnx

cgilis, 2023-03-21:
Will this work out of the box with zigbee2mqtt addon? (in an update..) 
Or do we need to update the firmware manually?

shawsky, 2023-03-21:
The firmware is in the repo already. If you add a custom converter as per [this](https://github.com/Koenkk/zigbee2mqtt/issues/16808#issuecomment-1466899337) comment you can update your device

NoClaim, 2023-03-21:
@shawsky The big Q for me is: where?
I have created a json file in the z2m directory and copy the complete code into the file. I assume this is not the correct way

shawsky, 2023-03-21:
> @shawsky The big Q for me is: where? I have created a json file in the z2m directory and copy the complete code into the file. I assume this is not the correct way

Have you added the following to your Z2M configuration.yaml?
`external_converters:
  - symfonisk.js`
 
Once done, re-start Z2M and it's worth removing and re-adding the Symfonisk, it should then show an update available

NoClaim, 2023-03-21:
I had no external converters in my configuration.yaml
I will add it. The symfonisk.js file is the one I have [the code](https://github.com/Koenkk/zigbee2mqtt/issues/16808#issuecomment-1466899337) added to?

shawsky, 2023-03-21:
@NoClaim Yes that's right, "symfonisk.js" is just the filename I've used for the converter in this case

[10 more replies]

closed 2023-03-24 by djansen1987