Koenkk/zigbee2mqtt #16809: Sonoff temperature/humidity SNZB-02 not repairing after force remove

gaggio, 2023-02-24:
### What happened?

The sensor was not sending updates, so after reading other solved issues I tried the following:

- force remove the sonoff temperature sensor
- remove the battery
- restarted z2m
- put the battery back in
- tried re-pairing

I tried multiple times, even trying to pair with coordinator only and with each router one at a time. New battery.

### What did you expect to happen?

Sensor to reconnect to the network and start reporting.

### How to reproduce it (minimal and precise)

Not sure, the sensor was working correctly for almost a year then after a damaged SD started to have hiccups up to the current situation.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221226

### Adapter

Electrolama zzh

### Debug log

No message at all relevant to this sensor in debug log

Wireheadbe, 2023-03-03:
To reset the Zigbee SonOff SNZB-02 temperature and humidity sensor, you must press and hold the reset button for more than 10 seconds. (5 secs only re-pairs it)
Have you tried a new battery? I have six of these, and they work flawless (even able to set specific reporting settings). So they are very well supported.

gaggio, 2023-03-07:
> To reset the Zigbee SonOff SNZB-02 temperature and humidity sensor, you must press and hold the reset button for more than 10 seconds. (5 secs only re-pairs it) 
I tried to follow your suggestion, but no improvement. Honestly, I'm not sure the sensor is doing anything different when holding the button for longer. Is it supposed to flash the led to confirm it's been reset?

> Have you tried a new battery? I have six of these, and they work flawless (even able to set specific reporting settings). So they are very well supported.
The one currently on is quite new and was reporting above 80% before the issue happened. I will buy another one and try o replace it anyway. By the way my sensor was working well too, and I didn't have any issue at the first pairing.

Wireheadbe, 2023-03-08:
Could be that the sensor is just damaged.
Force remove + reset sensor by long button press.. should do it. 

Are you close to the coordinator to pair, or are you pairing via routing?

gaggio, 2023-03-13:
I've tried to pair closer to the coordinator, even disabling pairing via routers, but nothing seemed to work.

News is that I upgraded to 1.30.2 and tried again, the sensor paired flawlessly and so far (24h) it's working and updating correctly.

closed 2023-04-20 by github-actions[bot]

nmaster2042, 2023-11-26:
I just got the same issue with a sensor I replaced the battery.
No more temp/hum/batt info reports anymore.
My USB coordinator is the Sonof Dongle  P.

marcelammerlaan, 2023-11-29:
Same here: stops updating. Won't repair but on a conbee stick with deconz. Seems the hardware/firmware is a tad buggy.

AlexH269, 2023-11-30:
I have the same issue with my sensor since a few days ago.

nmaster2042, 2023-12-02:
I found a way to solve the issue in my case.
Initially, The sensor was in a room, far from my main zigbee controller.
Changing battery and no more data coming from sensor.
Yesterday I took the sensor, came near the main zigbee controller, removed the battery, putting it after 5 sec and sensor reconnected to the zigbee network without doing anything else.
No zigbee2mqtt restart needed.
I will have to check with another sensor when I'll need to replace another battery.

hahlin90, 2023-12-08:
I had the same issue and removing the device and pairing it again worked after repeating for 5 times, right next to the Sonoff USB dongle.

Hudibras, 2024-05-18:
Replace battery 3feet from bridge wont restart updating, is the only way really to start new pairring

terbx, 2025-02-07:
Had similar or same problem after changing battery, appeared to be paied including wave indicator on device but all entities showing unavailable,, Tried everything above to 'reactivate' entities, no joy all seemd to be lost. However, patience seems to play a part , device adjacent to bridge try leave for a while 10/15 minutes, close home assistant and then restart. Worked for me a number of times.

tbart, 2026-09-14:
For anyone landing here: see https://github.com/Koenkk/zigbee2mqtt/issues/33115.
At least with newer firmware versions, registers have changed. That produced exactly the same issue as described here.
Fixed for me by following the info in the mentioned issue.