Koenkk/zigbee2mqtt #16810: wrong TUYA device detected : switch gang TS0011 recognize as TS011F_plug_1

Electronlibre2012, 2023-02-24:
### What happened?

Hello,

i have a TS0002 was a 2 gang wall switch, work like a charm, but the TS0011 which is a one gang wall switch was recognize as a plug : TS011F_plug_1.

### What did you expect to happen?

TS0011 recognize as a TS0011 : one gang wall switch 

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

Koenkk, 2023-02-25:
- Can you provide the data/database.db entry of this device?
- Are you sure it is a TS0011?

Electronlibre2012, 2023-02-25:
thanks for your reply! i think its like a TS011F _TZ3000_aim0ztek here is the link where i bought it : [https://fr.aliexpress.com/item/4000460643839.html?spm=a2g0o.order_list.order_list_main.37.7c525e5b9PmEbU&gatewayAdapt=glo2fra](url) the strange thing is the 2 gang is recognize well : 
[image]

but the 1 gang : is a socket lol : 
[image]

in the database.db there is 47 but in zigbee2mqtt i have only 46 device.

Can you provide the data/database.db entry of this device?
this is the id 47:
````
{"id":47,"type":"Router","ieeeAddr":"0xa4c13890d7826476","nwkAddr":20442,"manufId":4417,"manufName":"_TZ3000_aim0ztek","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�ϋ+\u0003","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_aim0ztek","zclVersion":3,"appVersion":192,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":237,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":2,"switchType":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad96a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad96a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad96a","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1677321850787,"defaultSendRequestWhen":"immediate"}
````
and this is the id 46:
````
{"id":46,"type":"Router","ieeeAddr":"0xa4c1383e1926906f","nwkAddr":13310,"manufId":4417,"manufName":"_TZ3000_gvn91tmx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��~+\u0003","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_gvn91tmx","zclVersion":3,"appVersion":192,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":237,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,648]}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2914147,"53251":124,"53252":10522,"53253":1235,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad96a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad96a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad96a","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1677325380731,"defaultSendRequestWhen":"immediate"}
`````

does i need to delete both in the database? Thanks

Koenkk, 2023-02-26:
`Led bureau` is a TS011F, not a TS0001:

```js
{"id":47,"type":"Router","ieeeAddr":"0xa4c13890d7826476","nwkAddr":20442,"manufId":4417,"manufName":"_TZ3000_aim0ztek","powerSource":"Mains (single phase)","modelId":"TS011F"
```

So these are not the same devices.

Electronlibre2012, 2023-02-26:
its physically a gang switch mount on wall, with phase, neutral and a phase for the lamp, i dont understand...its falsy recognize as a socket, but its not a power socket...what can i do please? any idea?

Koenkk, 2023-02-27:
This is just a cosmetic thing, the device should work fine.

Electronlibre2012, 2023-02-27:
yes,  you are right...

 i hope some day it will added to the supported devices.

closed 2023-04-06 by github-actions[bot]