Koenkk/zigbee2mqtt #16818: FrontEnd with different than default port is not working on version 1.30.1

AdamWeglarz, 2023-02-25:
### What happened?

I used Z2M version 1.25.2 with following config:
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
  user: xxx
  password: yyy
serial:
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2218951-if00
  adapter: deconz
advanced:
  log_level: debug
  frontend: true
    port: 8081
    host: 0.0.0.0
	
  channel: 25`

Everything was fin until I have updated version.

I have updated Z2M to latest stable and application stopped working.
It is not able to parse frontend settings.
`/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.30.1 start
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: bad indentation of a mapping entry (18:9)

 15 |   log_level: debug
 16 |   channel: 25
 17 |   frontend: true
 18 |     port: 8081
--------------^
 19 |     host: 0.0.0.0
 20 | devices:
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1182:7)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
    at Object.load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1714:19)
    at Object.read (/opt/zigbee2mqtt/lib/util/yaml.ts:7:21)`

### What did you expect to happen?

To start correctly application

### How to reproduce it (minimal and precise)

provide non standard port for FrontEnd.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

not relevant

### Adapter

deconz

Koenkk, 2023-02-26:
The indentation of your yaml file is incorrect. Paste your config into e.g. https://www.yamllint.com to find the errors.

closed 2023-02-26 by Koenkk

lux73, 2023-02-26:
checked this on my TestVM - works as expected

[image]

changed Port to 888