Koenkk/zigbee2mqtt #16820: Aqara S1 Air Monitor

xeon6, 2023-02-26:
### What happened?

The system successfully added the Xiaomi KQJCMB11LM (Aqara S1 Air Monitor); updated firmware through OTA to 09-19-2022.
Only can read the Link quality (signal strength); the rest of the data is unknown.

After enabling debug log level, I started seeing the "No converter available for 'KQJCMB11LM'".
Downloaded https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js to /config/zigbee2mqtt.
Modified the filed and changed the paths from .. to zigbee-herdsman-converters

Configuration.yaml:
homeassistant: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: homeio
  password: *****
serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  port: 8099
devices:
  '0xa4c138879fb7d970':
    friendly_name: Grandpa
  '0xa4c138a2a7a2be5d':
    friendly_name: LED stripe
  '0xa4c138819483d198':
    friendly_name: Kettle
  '0x00124b0029322ed4':
    friendly_name: Sonoff PIR
  '0x54ef4410005d5200':
    friendly_name: Aqara Air Monitor
permit_join: false
advanced:
  log_level: debug
external_converters:
  - xiaomi.js

Restarted Zigbee2MQTT, but the same carries on happening = no data, and the logs repeating the No converter errors.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

2023-02-25 23:59:30Received Zigbee message from 'Aqara Air Monitor', type 'attributeReport', cluster 'aqaraOpple', data '{"210":0}' from endpoint 1 with groupID null
2023-02-25 23:59:30No converter available for 'KQJCMB11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"210":0}'
2023-02-25 23:59:30Received Zigbee message from 'Aqara Air Monitor', type 'attributeReport', cluster 'aqaraOpple', data '{"344":2}' from endpoint 1 with groupID null
2023-02-25 23:59:30No converter available for 'KQJCMB11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"344":2}'
2023-02-26 00:00:32Received Zigbee message from 'Aqara Air Monitor', type 'attributeReport', cluster 'aqaraOpple', data '{"210":0}' from endpoint 1 with groupID null
2023-02-26 00:00:32No converter available for 'KQJCMB11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"210":0}'
2023-02-26 00:00:32Received Zigbee message from 'Aqara Air Monitor', type 'attributeReport', cluster 'aqaraOpple', data '{"344":0}' from endpoint 1 with groupID null
2023-02-26 00:00:32No converter available for 'KQJCMB11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"344":0}'
2023-02-26 00:00:59Saving state to file /config/zigbee2mqtt/state.json
2023-02-26 00:01:35Received Zigbee message from 'Aqara Air Monitor', type 'attributeReport', cluster 'aqaraOpple', data '{"210":0}' from endpoint 1 with groupID null
2023-02-26 00:01:35No converter available for 'KQJCMB11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"210":0}'
2023-02-26 00:01:35Received Zigbee message from 'Aqara Air Monitor', type 'attributeReport', cluster 'aqaraOpple', data '{"344":2}' from endpoint 1 with groupID null
2023-02-26 00:01:35No converter available for 'KQJCMB11LM' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"344":2}'

xeon6, 2023-03-29:
Changed the ConBee 2 Zigbee adapter to Skyconnect, which solved the problem.

closed 2023-03-29 by xeon6

dankarization, 2023-09-01:
I get the same problem with Sonoff Zigbee Dongle P. No data after OTA update