Koenkk/zigbee2mqtt #16823: Unable to upgrade to custom firmware

fred-gb, 2023-02-26:
### What happened?

Hello, 👋🏻 

I have problem with my [Zlinky TIC](https://www.zigbee2mqtt.io/devices/ZLinky_TIC.html#lixee-zlinky_tic) 

I would like to upgrade to a special firware version [limited](https://github.com/fairecasoimeme/Zlinky_TIC#route-or-limited-route-from-v7).

But I can't...

### What did you expect to happen?

Upgrade from UI or anything else.

### How to reproduce it (minimal and precise)

I used zigbee-OTA script. to create an index.json, with last entry:

```
    {
        "fileVersion": 13,
        "fileSize": 245406,
        "manufacturerCode": 4151,
        "imageType": 2,
        "sha512": "dd77b28a3b4664e7ad944fcffaa9eca9f3adb0bbe598e12bdd6eece8070a8cdda6792bed378d173dd5b4532b4cdb88cebda0ef0c432c4c4d6581aa9f2bbba54d",
        "url": "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v13.0/ZLinky_router_v13_limited.ota",
        "force": true
    }
```

I add `force` configuration.

In `configuration.yml` I add: 
```
ota:
  zigbee_ota_override_index_location: my_index.json
  ikea_ota_use_test_url: false
```

I renamed `index.json` in `my_index.json` an put in same directory like configuration.yaml.

I restart my docker Zigbee2Mqtt.

And send message: `domotique/zigbee2mqtt/bridge/request/device/ota_update/update` with ID of device `0x00158d0006a4018c`.

But... No update avaiblable when I try in UI.
```
Update of '0x00158d0006a4018c' failed (No new image available)
```

### Zigbee2MQTT version

1.30.1 commit: eb878d3 (docker)

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZBDongle-E (EZSP v8)

### Debug log

```
Zigbee2MQTT:debug 2023-02-26 17:09:50: Responded to OTA request of '0x94deb8fffe8a1ec6' with 'NO_IMAGE_AVAILABLE'
```

Danit2, 2023-02-26:
This is beqause of this problem. 

https://github.com/Koenkk/zigbee2mqtt/issues/12554

It is not fix at the moment. 

You can make a custom converter of this device with the normal ota directory and then the override will work.

molusk, 2023-02-28:
@Danit2 when I look at the lixee converter, I see not much about OTA : ``` ota: ota.lixee,``` . 

What do you mean we should do with this custom converter ?

Danit2, 2023-02-28:
You must replace the `ota: ota.lixee` with `ota: ota.zigbeeOTA` then the local ota index is working.

molusk, 2023-02-28:
Thank's a lot, I will test this as soon as possible :clap:

molusk, 2023-02-28:
If I understand well, the next step would be to modify lixee.js to handle custom index (and/or alternative firmware as we are currently trying to push). Huge step for the JS rookie I am :laughing:

Danit2, 2023-02-28:
yes, when you will use the local index you have to write the same function as in this file:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/a3e11ce5ffa9a02f61f296ecf990dc410e2691f0/lib/ota/zigbeeOTA.js#L34

But there i can not help. Sorry.

molusk, 2023-04-04:
The issue should be kept open until someone finds a suitable solution.
I don't know JS well and don't have enough time right know to try to solve this issue.

molusk, 2023-05-09:
This issue should be kept open.

molusk, 2023-06-01:
Indeed, but there was a discussion about how the creator of the device should manage the firmware variants to allow an elegant way of managing upgrades. Representatives of various zigbee managers (including z2m) were involved.
Nothing was done for z2m yet.
The workaround works but this issue should be kept open until somebody competent enough and willing to do it passes by :)

molusk, 2023-07-02:
Do not close this issue, please.

molusk, 2023-08-04:
Please keep this open.

molusk, 2023-10-09:
Not stale

KipK, 2024-10-11:
I'm reponing this topic, seeing people still living with this issue.

@Koenkk , lixee device has 2 different firmwares, but booth use different 'type' meta. type 1 for norma fw, type 2 for limited.
manufacturer code and all other metas are the same for booth.

Checking the [lixee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ota/lixee.ts) , I see there's no such test for "type" meta.

Would it be possible to add such test like in [zigbeeota.js ](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ota/zigbeeOTA.ts) and download the appropriated file then ? 

Or simpler, just remove the lixee.js from ota lib and replace in [lixee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a3e11ce5ffa9a02f61f296ecf990dc410e2691f0/devices/lixee.js) ( from device folder )  line 646 the ota: ota.lixee, by ota: ota.zigbeeOTA, and ask for the author if the firmware can be hosted on your git like the other devices. This would solve the issue without too much hassle.

Koenkk, 2024-10-12:
@KipK good idea, but how do you want the user to select the correct firmware?

KipK, 2024-10-12:
User don't have to select. Current installed firmware already had type meta, can't we get that information from the device ?

KipK, 2024-10-12:
I remember on Deconz, It was displaying all the OTA meta for each running devices

KipK, 2024-10-12:
Related to what you've said on the other `[post](https://github.com/Koenkk/zigbee2mqtt/issues/24295)`

>PS: that ZLinky_TIC is in the official OTA index from what I can see: https://github.com/Koenkk/zigbee-OTA/blob/b8dee47ba1ff6e79ec6f249b666b9631d71c406e/index.json#L1668

So we could add a second entry after this one:
```
    {
        "fileVersion": 14,
        "fileSize": 249694,
        "manufacturerCode": 4151,
        "imageType": 1,
        "sha512": "cc69b0745c72daf8deda935ba47aa7abd34dfcaaa4bc35bfa0605cd7937b0ecd8582ba0c08110df4f620c8aa87798d201f407d3d7e17198cfef1a4aa13c5013d",
        "url": "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v14.0/ZLinky_router_v14.ota"
    },
```

that would gives for the limited version:

```
    {
        "fileVersion": 14,
        "fileSize": 249694,
        "manufacturerCode": 4151,
        "imageType": 2,
        "sha512": "cc69b0745c72daf8deda935ba47aa7abd34dfcaaa4bc35bfa0605cd7937b0ecd8582ba0c08110df4f620c8aa87798d201f407d3d7e17198cfef1a4aa13c5013d",
        "url": "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v14.0/ZLinky_router_v14_limited.ota"
    },
```

But I don't think it use the commn index, in [lixee.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a3e11ce5ffa9a02f61f296ecf990dc410e2691f0/devices/lixee.js) line 646:

  ota: ota.lixee, 
should be then:
ota: ota.zigbeeOTA

And this should work theorically ain't it ?

KipK, 2024-10-14:
So i have tested and it works as wanted.

I"ve created a PR to fix this: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8136

and submitted limited firmware: https://github.com/Koenkk/zigbee-OTA/pull/568

closed 2024-10-15 by Koenkk