Koenkk/zigbee2mqtt #16826: [Feature request]: Save Temperature in JSON on txt file

lucaesse, 2023-02-26:
### Is your feature request related to a problem? Please describe

I have this Tuya sensor WSD500A and I cannot get temperature over MQTT.
I can use another plugin in homebridge to read the temperature in JSON format from a .txt file and use it for another accessory in homekit.

### Describe the solution you'd like

It could be great to have an option for saving the temperature in JSON format on a .txt file

### Describe alternatives you've considered

I tried to read it from log file but it didn't worked

### Additional context

Could be useful also with humidity

junosuarez, 2023-03-29:
This seems outside the scope of zigbee2mqtt.

Koenkk, 2023-03-29:
Indeed out of scope.

closed 2023-03-29 by Koenkk