Koenkk/zigbee2mqtt #16827: [Feature request]: S520567position states [Stopped/Opening/Closing]

Pepeanuts, 2023-02-26:
### Is your feature request related to a problem? Please describe

Actually we have two states : open/close, how can we have opening and closing states ?

### Describe the solution you'd like

More states than "Open/close" when the cover is opening or closing

### Describe alternatives you've considered

Movement state

### Additional context

None

Drafteed, 2023-03-02:
You can use `motor_state` / `moving` or `running` ([example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8878a5b7925949e8bb5755e587b4a180c5a826f3/devices/xiaomi.js#L2198-L2201)). 

Home Assistant extension will handle this:
https://github.com/Koenkk/zigbee2mqtt/blob/cdf62ea5601f74af19b1626b72afc88edbd8e4ab/lib/extension/homeassistant.ts#L393-L436.

Pepeanuts, 2023-03-02:
> You can use `motor_state` / `moving` or `running` ([example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8878a5b7925949e8bb5755e587b4a180c5a826f3/devices/xiaomi.js#L2198-L2201)).
>
> Home [...]

Thanks for your reply, but what i'm supposed to do with this ?
It's impossible to be in default settings for this device integration ?

Drafteed, 2023-03-03:
What curtain device are you using?

Pepeanuts, 2023-03-03:
> What curtain device are you using?

I use Schneider cover buttons S520567W

Drafteed, 2023-03-03:
Currently zigbee2mqtt does not handle the moving state of your device. Since I don't have this device I can't help. You can try to do some research yourself to see if the device is reporting motion state.

Pepeanuts, 2023-03-03:
> Currently zigbee2mqtt does not handle the moving state of your device. Since I don't have this device I can't help. You can try to do some research yourself to see if the device is reporting motion state.

In zigbee2mqtt not but with Wiser app and Home assistant Wiser HACS intégration yes.

[image]

https://forum.hacf.fr/t/solution-pour-bien-integrer-linterrupteur-volet-roulant-connecte-zigbee-3-0-wiser-odace-blanc-s520567w/7035/44?u=pepeanuts

Pepeanuts, 2023-03-24:
With Wiser HACS intégration there is "lift_movement: Stopped".

How z2m can use this attributs ?

[image]

Pepeanuts, 2023-04-24:
How to integrate "lift_movement" attribut ?

closed 2023-06-01 by github-actions[bot]

fjumeaux, 2025-05-16:
Same question : How z2m can use this attributs ? (attributs exposed with wiser). I have 8 S520567 at home. It would be really useful for managing manual interactions during automated roller shutter movements.