Koenkk/zigbee2mqtt #16834: Unable to get/set transmit power for ZBDongle-P with the latest firmware

onsprng, 2023-02-27:
### What happened?

When trying to get or set transmit power , the command will timeout. But managed to get 1 of the dongle to transmit power , 2 dongle not able to.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20221102

### Adapter

Sonoff ZBDongle-p

### Debug log

Publish 'set' 'transmit_power' to 'zbdongle2-l1walkway' failed: 'Error: Write 0x00124b002a2f52c8/242 genBasic({"4919":{"value":20,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 24282 - 242 - 122 - 0 - 4 after 10000ms)'

jerrm, 2023-03-03:
Has z2m been restarted since pairing?  It may not be related, but I've seen issues where unexpected behavior with a newly paired router stick goes awY after a z2m container restart or reboot.

bromarc, 2023-03-24:
I am having the same issue, on z2m 1.30.2

bromarc, 2023-04-04:
My issue appears to be fixed on 1.30.3-1

closed 2023-05-13 by github-actions[bot]