Koenkk/zigbee2mqtt #16838: problem with tuya motion sensor keep_time

1HyperZ, 2023-02-27:
### What happened?

the keep time doesnt change for the tuya motion sensor

https://user-images.githubusercontent.com/77121655/221612747-89fcf2e9-7317-45e0-9db5-098e230b6db9.mp4

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20220507

### Adapter

sonoff stick

### Debug log

Debug 2023-02-27 17:50:37Received MQTT message on 'zigbee2mqtt/Home Entrance Motion Sensor/get' with data '{"keep_time":""}'
Debug 2023-02-27 17:50:37Publishing get 'get' 'keep_time' to 'Home Entrance Motion Sensor'
Debug 2023-02-27 17:50:37Received MQTT message on 'zigbee2mqtt/Home Entrance Motion Sensor/get' with data '{"keep_time":""}'
Debug 2023-02-27 17:50:37Publishing get 'get' 'keep_time' to 'Home Entrance Motion Sensor'
Debug 2023-02-27 17:50:38Received MQTT message on 'zigbee2mqtt/Home Entrance Motion Sensor/get' with data '{"keep_time":""}'
Debug 2023-02-27 17:50:38Publishing get 'get' 'keep_time' to 'Home Entrance Motion Sensor'
Debug 2023-02-27 17:50:38Received MQTT message on 'zigbee2mqtt/Home Entrance Motion Sensor/get' with data '{"keep_time":""}'
Debug 2023-02-27 17:50:38Publishing get 'get' 'keep_time' to 'Home Entrance Motion Sensor'
Debug 2023-02-27 17:50:38Received MQTT message on 'zigbee2mqtt/Home Entrance Motion Sensor/get' with data '{"keep_time":""}'
Debug 2023-02-27 17:50:38Publishing get 'get' 'keep_time' to 'Home Entrance Motion Sensor'
Debug 2023-02-27 17:50:56Received Zigbee message from 'Kitchen 4 Gang Switch', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":76}' from endpoint 1 with groupID 0
Debug 2023-02-27 17:51:05Received Zigbee message from '0xa4c138fbc37e7724', type 'attributeReport', cluster 'genBasic', data '{"65506":53,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
Debug 2023-02-27 17:51:19Received Zigbee message from 'Home Entrance Ceiling Light', type 'attributeReport', cluster 'genBasic', data '{"appVersion":98}' from endpoint 1 with groupID 0
Error 2023-02-27 17:51:29Publish 'set' 'keep_time' to 'Home Entrance Motion Sensor' failed: 'Error: Write 0xa4c138dac753b087/1 ssIasZone({"61441":{"value":2,"type":32}},

arripeo, 2023-03-01:
Here same device same problem, more over also for the "sensitivity" option is quite impossible to make any change here below the error
Publish 'set' 'sensitivity' to 'Motion bath gf2' failed: 'Error: Command 0xa4c138abb7bcfb39/1 manuSpecificTuya.dataRequest({"seq":10,"dpValues":[{"dp":9,"datatype":4,"data":[0]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

1HyperZ, 2023-03-10:
Yes I have that too both keep time and sensitivity cant change

kevindegrootgrave, 2023-04-07:
This is not a problem. To safe power, changes to keep time and sensitivity are only applied when device is active (e.g. because it detected motion).

See https://www.zigbee2mqtt.io/devices/ZG-204ZL.html under Reading and Setting Values.

closed 2023-05-15 by github-actions[bot]

uniquePWD, 2024-11-11:
Did anything ever come of this?