Koenkk/zigbee2mqtt #16842: [Feature request]: Make base topic for devices configurable

junosuarez, 2023-02-27:
### Is your feature request related to a problem? Please describe

Problem: I'd like to subscribe to all device topics, but not bridge topics. Specifically, logging is more verbose than I want my subscribers to handle.

The base `zigbee2mqtt/` topic is configurable, but because of how wildcard subscriptions work in MQTT, I'd like to configure a separate base topic for device topics.

I'd like to be able to subscribe to eg `zigbee2mqtt/devices/#` to ignore all of `zigbee2mqtt/bridge/#`.

### Describe the solution you'd like

Add a configuration `mqtt.device_topic` or `mqtt.device_topic_prefix` (open to other names if you think of a better one)

By default, if it's not specified in config, no prefix is applied and the current topic naming behavior is unchanged.

If it's supplied, this value is prefixed to the topic.

### Example: Default behavior
```yaml
mqtt:
  base_topic: zigbee2mqtt
```
A device with the friendly name `light` would publish to `zigbee2mqtt/light`

### Example: Configured behavior
```yaml
mqtt:
  base_topic: zigbee2mqtt
  device_topic: devices
```
A device with the friendly name `light` would publish to `zigbee2mqtt/devices/light`

### Describe alternatives you've considered

I can modify the FRIENDLY_NAMES to include a `device/FRIENDLY_NAME` prefix, but this requires changing the device friendly names and changes how they show up in the zigbee2mqtt UI itself, and relies on me manually setting friendly names to follow this convention.

Alternately, I can filter out the `bridge` messages in my subscriber application. However, this creates additional network traffic. 

As another alternative, I could add an mqtt relay layer to subscribe to the zigbee2mqtt topics, apply filtering, and republish certain topics to another name. However, this adds additional complexity to the setup.

Lastly, I could configure my MQTT broker to ignore those topics, or set up an ACL to filter access.

It seems like the best place to control this behavior is in the publisher: zigbee2mqtt.

### Additional context

None

junosuarez, 2023-03-01:
As a workaround, I wrote an extension to meet my current needs. However, this duplicates some code from `controller.ts` for how entity state messages are published - since that might change in the future, this isn't sustainable.

Sharing as a proof of concept:
```js
const EXCLUDE_COORDINATOR = false // this is a constant that makes it exclude... not saying not to exclude. confusing.
const DEVICE_BASE_TOPIC = "devices" // added to MQTT.BASE_TOPIC, eg /zigbee2mqtt/devices/FRIENDLY_NAME

const stringify = require('json-stable-stringify-without-jsonify');
const utils = require('../util/utils').default;

class DeviceBaseTopic {
    #eventBus

    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.#eventBus = eventBus

        this.#eventBus.on('publishEntityState', async (data) => {
            const {entity, message} = data
            
            // same as what happens on publish in https://github.com/Koenkk/zigbee2mqtt/blob/eb878d3d8ee47f77e27c771452e2d2c77ca83bb5/lib/controller.ts#L275-L286
            if (Object.entries(message).length) {
                const topic = `${DEVICE_BASE_TOPIC}/${entity.name}`
                // logger.info('ES: ' + topic + ": " + JSON.stringify(message))    
         
                const options = {
                    retain: utils.getObjectProperty(entity.options, 'retain', false),
                    qos: utils.getObjectProperty(entity.options, 'qos', 0),
                };

                const retention = utils.getObjectProperty(entity.options, 'retention', false);
                if (retention !== false) {
                    options.properties = {messageExpiryInterval: retention};
                }
                
                await mqtt.publish(topic, stringify(message), options);
            }
            
        }, this.constructor.name);
    }

    async stop() {
        this.#eventBus.removeListeners(this.constructor.name);
    }
}

module.exports = DeviceBaseTopic;
```

Timmmy001, 2023-03-09:
My request goes in a similar direction.
https://github.com/Koenkk/zigbee2mqtt/issues/16912

closed 2023-04-17 by github-actions[bot]