Koenkk/zigbee2mqtt #16845: Power meter devices spamming attributeReport / genBasic messages and making network slow/ crash

NietSinterklaas, 2023-02-27:
### What happened?

So I got a lot of TS011F_plug_1 power meter plugs, a lot of them are spamming wired messages and making the whole network slow/ unresponsive and some times even crash Zigbee2MQTT. 

I have tried to update devices, repair, change interval in reporting settings and even think about removing them from the network because they make everything unreliable.

[image]
[image]
[image]

### What did you expect to happen?

Not crash or spam

### How to reproduce it (minimal and precise)

Get my power meter blocks from my house

### Zigbee2MQTT version

1.30.1-1

### Adapter firmware version

20210708

### Adapter

Sonoff | Zigbee 3.0 | Dongle Plus-E

adorobis, 2023-02-28:
Interesting observation, in my case it seems that a Thermostat is flooding the network with messages:
```
debug 2023-02-28 13:56:06: Received Zigbee message from 'Termostat Szary', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":25091}' from endpoint 1 with groupID null
info  2023-02-28 13:56:06: MQTT publish: topic 'zigbee2mqtt/Termostat Szary', payload '{"child_lock":"UNLOCK","current_heating_setpoint":22,"deadzone_temperature":1,"heat":"ON","linkquality":255,"local_temperature":21,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":26,"min_temperature_limit":null,"preset":"program","preset_mode":"program","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":21,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":21,"saturday_p3_hour":14,"saturday_p3_minute":0,"saturday_p3_temperature":21,"saturday_p4_hour":19,"saturday_p4_minute":0,"saturday_p4_temperature":19,"sunday_p1_hour":8,"sunday_p1_minute":0,"sunday_p1_temperature":21,"sunday_p2_hour":12,"sunday_p2_minute":0,"sunday_p2_temperature":21,"sunday_p3_hour":14,"sunday_p3_minute":0,"sunday_p3_temperature":21,"sunday_p4_hour":19,"sunday_p4_minute":0,"sunday_p4_temperature":19,"weekdays_p1_hour":4,"weekdays_p1_minute":30,"weekdays_p1_temperature":21,"weekdays_p2_hour":12,"weekdays_p2_minute":0,"weekdays_p2_temperature":21,"weekdays_p3_hour":15,"weekdays_p3_minute":0,"weekdays_p3_temperature":21,"weekdays_p4_hour":19,"weekdays_p4_minute":0,"weekdays_p4_temperature":19},"running_state":"heat","sensor":"IN","system_mode":"heat"}'
debug 2023-02-28 13:56:06: Received Zigbee message from 'Termostat Szary', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":25347}' from endpoint 1 with groupID null
info  2023-02-28 13:56:06: MQTT publish: topic 'zigbee2mqtt/Termostat Szary', payload '{"child_lock":"UNLOCK","current_heating_setpoint":22,"deadzone_temperature":1,"heat":"ON","linkquality":255,"local_temperature":21,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":26,"min_temperature_limit":null,"preset":"program","preset_mode":"program","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":21,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":21,"saturday_p3_hour":14,"saturday_p3_minute":0,"saturday_p3_temperature":21,"saturday_p4_hour":19,"saturday_p4_minute":0,"saturday_p4_temperature":19,"sunday_p1_hour":8,"sunday_p1_minute":0,"sunday_p1_temperature":21,"sunday_p2_hour":12,"sunday_p2_minute":0,"sunday_p2_temperature":21,"sunday_p3_hour":14,"sunday_p3_minute":0,"sunday_p3_temperature":21,"sunday_p4_hour":19,"sunday_p4_minute":0,"sunday_p4_temperature":19,"weekdays_p1_hour":4,"weekdays_p1_minute":30,"weekdays_p1_temperature":21,"weekdays_p2_hour":12,"weekdays_p2_minute":0,"weekdays_p2_temperature":21,"weekdays_p3_hour":15,"weekdays_p3_minute":0,"weekdays_p3_temperature":21,"weekdays_p4_hour":19,"weekdays_p4_minute":0,"weekdays_p4_temperature":19},"running_state":"heat","sensor":"IN","system_mode":"heat"}'
debug 2023-02-28 13:56:06: Received Zigbee message from 'Termostat Szary', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":25603}' from endpoint 1 with groupID null
info  2023-02-28 13:56:06: MQTT publish: topic 'zigbee2mqtt/Termostat Szary', payload '{"child_lock":"UNLOCK","current_heating_setpoint":22,"deadzone_temperature":1,"heat":"ON","linkquality":255,"local_temperature":21,"local_temperature_calibration":-2,"max_temperature":35,"max_temperature_limit":26,"min_temperature_limit":null,"preset":"program","preset_mode":"program","program":{"saturday_p1_hour":6,"saturday_p1_minute":0,"saturday_p1_temperature":21,"saturday_p2_hour":12,"saturday_p2_minute":0,"saturday_p2_temperature":21,"saturday_p3_hour":14,"saturday_p3_minute":0,"saturday_p3_temperature":21,"saturday_p4_hour":19,"saturday_p4_minute":0,"saturday_p4_temperature":19,"sunday_p1_hour":8,"sunday_p1_minute":0,"sunday_p1_temperature":21,"sunday_p2_hour":12,"sunday_p2_minute":0,"sunday_p2_temperature":21,"sunday_p3_hour":14,"sunday_p3_minute":0,"sunday_p3_temperature":21,"sunday_p4_hour":19,"sunday_p4_minute":0,"sunday_p4_temperature":19,"weekdays_p1_hour":4,"weekdays_p1_minute":30,"weekdays_p1_temperature":21,"weekdays_p2_hour":12,"weekdays_p2_minute":0,"weekdays_p2_temperature":21,"weekdays_p3_hour":15,"weekdays_p3_minute":0,"weekdays_p3_temperature":21,"weekdays_p4_hour":19,"weekdays_p4_minute":0,"weekdays_p4_temperature":19},"running_state":"heat","sensor":"IN","system_mode":"heat"}'
debug 2023-02-28 13:56:06: Received Zigbee message from 'Door Sensor 2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2023-02-28 13:56:06: MQTT publish: topic 'zigbee2mqtt/Door Sensor 2', payload '{"battery":70,"contact":true,"device_temperature":4,"linkquality":255,"power_outage_count":21143,"temperature":23,"voltage":2955}'
debug 2023-02-28 13:56:06: Received Zigbee message from 'Termostat Szary', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":25859}' from endpoint 1 with groupID null
[... cut ...] s_p4_minute":0,"weekdays_p4_temperature":19},"running_state":"heat","sensor":"IN","system_mode":"heat"}'

```
Seems that after shutting down power at entire home and turning it back on the network is back in more or less normal. I've also disconnected for a while the termostat from power. After powering it back on it still sends a lot of messages, on average 50 messages per minute. Could the device be faulty? Or was it always behaving this way and just this morning the network got stuck with it?

NietSinterklaas, 2023-03-01:
What to do if there is no `attributeReport` under Reporting?

[image]

Koenkk, 2023-03-02:
`attributeReport` is the message type, if you reduce the reporting of e.g. `appVersion` less `attributeReport` messages will come in.

NietSinterklaas, 2023-03-16:
Had set it on Minrep 3600 max rep 3600 change 5. Still keeps happening sadly.

[image]

NietSinterklaas, 2023-03-16:
Have tried it, set the min rep on 3600 max rep 3600 change on 5.

[image]

Still keeps happening.

closed 2023-04-23 by github-actions[bot]

clumsy-stefan, 2025-11-14:
I do have the same issue, also mainly with tuya switches that have power measurement included.

Has anyone found a solution to this?