Koenkk/zigbee2mqtt #16853: Moes Thermostat cannot set current_heating_setpoint in 0.5 steps (which it supports) [ _TZE200_aoclfnxz / TS0601 /  BHT-002-GCLZB ]

zeroflag0, 2023-03-01:
### What happened?

Can only set current_heating_setpoint in whole integers

### What did you expect to happen?

Setting 0.5°C-steps on all temperature-parameters, which the device supports.

### How to reproduce it (minimal and precise)

[image]
[image]

### Zigbee2MQTT version

1.30.1 commit: eb878d3

### Adapter firmware version

20221226

### Adapter

usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Debug log

 2023-03-01 07:04:26MQTT publish: topic 'zigbee2mqtt/Heating Livingroom', payload '{"child_lock":null,"current_heating_setpoint":22,"deadzone_temperature":null,"heat":"OFF","linkquality":167,"local_temperature":20,"local_temperature_calibration":-1,"max_temperature_limit":null,"min_temperature_limit":null,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":5,"saturday_p1_minute":0,"saturday_p1_temperature":21.5,"saturday_p2_hour":6,"saturday_p2_minute":30,"saturday_p2_temperature":22,"saturday_p3_hour":13,"saturday_p3_minute":30,"saturday_p3_temperature":22.5,"saturday_p4_hour":21,"saturday_p4_minute":30,"saturday_p4_temperature":20,"sunday_p1_hour":5,"sunday_p1_minute":0,"sunday_p1_temperature":21.5,"sunday_p2_hour":6,"sunday_p2_minute":30,"sunday_p2_temperature":22,"sunday_p3_hour":13,"sunday_p3_minute":30,"sunday_p3_temperature":22.5,"sunday_p4_hour":21,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":5,"weekdays_p1_minute":0,"weekdays_p1_temperature":21.5,"weekdays_p2_hour":6,"weekdays_p2_minute":30,"weekdays_p2_temperature":22,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22.5,"weekdays_p4_hour":21,"weekdays_p4_minute":30,"weekdays_p4_temperature":20},"running_state":"idle","sensor":"OU","system_mode":"off"}'
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":108}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":364}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":620}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":876}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1132}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1388}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:28Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1644}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1900}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":2156}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":2412}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":2668}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":2924}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":3180}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":3436}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":3692}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":3948}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4204}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4460}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4716}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4972}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":5228}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":5484}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":5740}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":5996}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":6252}' from endpoint 1 with groupID 0
Debug 2023-03-01 07:04:29Received Zigbee message from 'Heating Livingroom', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}]," [... cut ...] "sunday_p3_hour":13,"sunday_p3_minute":30,"sunday_p3_temperature":22.5,"sunday_p4_hour":21,"sunday_p4_minute":30,"sunday_p4_temperature":20,"weekdays_p1_hour":5,"weekdays_p1_minute":0,"weekdays_p1_temperature":21.5,"weekdays_p2_hour":6,"weekdays_p2_minute":30,"weekdays_p2_temperature":22,"weekdays_p3_hour":13,"weekdays_p3_minute":30,"weekdays_p3_temperature":22.5,"weekdays_p4_hour":21,"weekdays_p4_minute":30,"weekdays_p4_temperature":20},"running_state":"idle","sensor":"OU","system_mode":"off"}'

adorobis, 2023-03-01:
As far as I know it is a known limitation of the device when controlling over zigbee. Not specific to zigbee2mqtt.

adorobis, 2023-03-01:
And here is the information from Moes support confirming above limitation:
https://github.com/Koenkk/zigbee2mqtt/issues/14965#issuecomment-1368179306

closed 2023-04-09 by github-actions[bot]