Koenkk/zigbee2mqtt #16854: Moes BHT-002 is sometimes missing current_heating_setpoint from state object

odelma, 2023-03-01:
### What happened?

I am getting errors in Domoticz automation platform about missing current_heating_setpoint in the state object. This happens on average once or twice per hour. Not sure if it is an issue ("feature") with the devices or something that could be enhanced in zigbee2mqtt integration. Example status messages from devices:

```
{"child_lock":"UNLOCK","deadzone_temperature":1,"heat":"OFF","linkquality":96,"local_temperature":19,"local_temperature_calibration":-3,"max_temperature":35,"max_temperature_limit":25,"min_temperature_limit":5,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2_minute":0,"saturday_p2_temperature":0,"saturday_p3_hour":0,"saturday_p3_minute":0,"saturday_p3_temperature":0,"saturday_p4_hour":0,"saturday_p4_minute":0,"saturday_p4_temperature":0,"sunday_p1_hour":0,"sunday_p1_minute":0,"sunday_p1_temperature":0,"sunday_p2_hour":0,"sunday_p2_minute":0,"sunday_p2_temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"heat"}
{"child_lock":"UNLOCK","deadzone_temperature":1,"heat":"OFF","linkquality":96,"local_temperature":19,"local_temperature_calibration":-3,"max_temperature":35,"max_temperature_limit":25,"min_temperature_limit":5,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2_minute":0,"saturday_p2_temperature":0,"saturday_p3_hour":0,"saturday_p3_minute":0,"saturday_p3_temperature":0,"saturday_p4_hour":0,"saturday_p4_minute":0,"saturday_p4_temperature":0,"sunday_p1_hour":0,"sunday_p1_minute":0,"sunday_p1_temperature":0,"sunday_p2_hour":0,"sunday_p2_minute":0,"sunday_p2_temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"heat"}
{"child_lock":"UNLOCK","deadzone_temperature":1,"heat":"OFF","linkquality":96,"local_temperature":19,"local_temperature_calibration":-3,"max_temperature":35,"max_temperature_limit":25,"min_temperature_limit":5,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2_minute":0,"saturday_p2_temperature":0,"saturday_p3_hour":0,"saturday_p3_minute":0,"saturday_p3_temperature":0,"saturday_p4_hour":0,"saturday_p4_minute":0,"saturday_p4_temperature":0,"sunday_p1_hour":0,"sunday_p1_minute":0,"sunday_p1_temperature":0,"sunday_p2_hour":0,"sunday_p2_minute":0,"sunday_p2_temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"heat"}

{"child_lock":"UNLOCK","current_heating_setpoint":18,"deadzone_temperature":1,"heat":"OFF","linkquality":96,"local_temperature":19,"local_temperature_calibration":-3,"max_temperature":35,"max_temperature_limit":25,"min_temperature_limit":5,"preset":"hold","preset_mode":"hold","program":{"saturday_p1_hour":0,"saturday_p1_minute":0,"saturday_p1_temperature":0,"saturday_p2_hour":0,"saturday_p2_minute":0,"saturday_p2_temperature":0,"saturday_p3_hour":0,"saturday_p3_minute":0,"saturday_p3_temperature":0,"saturday_p4_hour":0,"saturday_p4_minute":0,"saturday_p4_temperature":0,"sunday_p1_hour":0,"sunday_p1_minute":0,"sunday_p1_temperature":0,"sunday_p2_hour":0,"sunday_p2_minute":0,"sunday_p2_temperature":0,"sunday_p3_hour":0,"sunday_p3_minute":0,"sunday_p3_temperature":0,"sunday_p4_hour":0,"sunday_p4_minute":0,"sunday_p4_temperature":0,"weekdays_p1_hour":0,"weekdays_p1_minute":0,"weekdays_p1_temperature":0,"weekdays_p2_hour":0,"weekdays_p2_minute":0,"weekdays_p2_temperature":0,"weekdays_p3_hour":0,"weekdays_p3_minute":0,"weekdays_p3_temperature":0,"weekdays_p4_hour":0,"weekdays_p4_minute":0,"weekdays_p4_temperature":0},"running_state":"idle","sensor":"OU","system_mode":"heat"}
```

Here there are three consequtive statuses missing the current_heating_setpoint value, then it is back again. Domoticz developers requested to raise this issue here (https://github.com/domoticz/domoticz/issues/5549#issuecomment-1420298649).

### What did you expect to happen?

Expected that all status messages should have current_heating_setpoint included.

### How to reproduce it (minimal and precise)

Include Moes BHT-002 to zigbee network and monitor for the resulting mqtt status messages to witness the occasional missing of current_heating_setpoint.

### Zigbee2MQTT version

1.30.1-dev commit: 61b1732

### Adapter firmware version

20210120

### Adapter

Electrolama ZZH!

### Debug log

This is not an error in zigbee2mqtt as such. But systems (like Domoticz) using zigbee2mqtt to autodiscover entities are affected as they expect current_heating_setpoint to be included in status messages.

insipiens, 2023-03-02:
I think this might be related to an odd behaviour we noticed sometimes [https://github.com/Koenkk/zigbee2mqtt/issues/4185#issuecomment-1386189566](https://github.com/Koenkk/zigbee2mqtt/issues/4185#issuecomment-1386189566)

The device seems to send a zigbee message before it has determined it's settings - see in the message below it shows local temp = 1.9 (in fact it was 15 degrees), and the set point is null.

Does your message only appear when the device is first powered up?

```
Mar  2 12:38:44 zigbee2mqtt npm[406]: Zigbee2MQTT:info  2023-03-02 12:38:44: MQTT publish: topic 'zigbee2mqtt/Moes_Room_Thermostat', payload '{"child_lock":null,"current_heating_setpoint":11,"deadzone_temperature":1,"heat":"OFF","linkquality":142,"local_temperature":-1.9,"local_temperature_calibration":-2,"max_temperature_limit":30,"min_temperature_limit":5,"preset":"hold","preset_mode":"hold","running_state":"idle","sensor":"IN","system_mode":"heat"}'
```

odelma, 2023-03-03:
Hi, no not in my case. The devices are on continuous power except on quite rare occasion (once or two
twice every few weeks) we turn the sauna on. Current_heating_setpoint is missing sometimes every hour (on couple of messages), sometimes there is +4h in between.

closed 2023-04-10 by github-actions[bot]

SSG1111, 2023-07-23:
Any progress on this? 
I have the same issue here, but the occurrence is much higher - every 5...10min (+ I have 6 devices, + message flooding).

Zigbee2MQTT version: 1.32.1-1
Adapter: Sonoff_Zigbee_3.0_USB_Dongle_Plus
```
Logger: homeassistant.helpers.template
Source: helpers/template.py:735
First occurred: 18:27:37 (39 occurrences)
Last logged: 18:29:21

Template variable warning: 'dict object' has no attribute 'current_heating_setpoint' when rendering '{{ value_json.current_heating_setpoint }}'
```

MaCrek22, 2025-01-23:
Have someone tried to contact MOES support for these kind of issues ? Or maybe ask if they have recent firmware update ?

SSG1111, 2025-01-23:
I have purchased couple more devices from MOES and found out that those are different from ones which I have purchased 2 years ago (different connectors on the screen panels, new devices are not supported yet by Z2M). Did try to contact MOES support, but silence from their side... I beleive that they are not checking/investigating the cases where thermostats are used without MOES bridge.

MaCrek22, 2025-01-23:
heh, looks, that it is not needed anymore :)
https://github.com/slacky1965/tuya_thermostat_zrd/tree/main

SSG1111, 2025-01-28:
need in my case, because device I have has diffent code - TZE200_aoclfnxz, which is not listed there...
I would agree, it looks 90% the same and functions 99% are similar.

anyway, thanks for the link, @MaCrek22