Koenkk/zigbee2mqtt #16859: Thermostatic valves periodically change its values in target_temperature _TZE200_yw7cahqs

UweLm, 2023-03-01:
### What happened?

Hi @all, I wrote some javascript classes to implement a heating control. Since a long time, I have the problem that 4 of 7 thermostats change their values in the property zigbee.0...target_temperature automatically and without intervention of the javascript classes. This bypasses the complete control of the temperatures.
I've switched from deconz to ZigBee, but this did not solve the problem.
After more than a year of searching, I now came across the same bug on another channel (zigpy): https://github.com/zigpy/zha-device-handlers/issues/1815.
The problem seems to have been fixed there: https://github.com/zigpy/zha-device-handlers/pull/1816
Is it possible to implement such a solution for the ZigBee adapter too?
Or am I on the subject completely wrong here?
Hence, some more data:
- iobroker: 6.3.5
- Node.js: v16.19.0(16.19.1)
- NPM: 8.19.3
- Zigbee: v1.8.10
- Thermostat manufacturer: _TZE200_yw7cahqs

In the meanwhile, I've updated my Zigbee Adapter to the Github version (repos:beta) 1.8.11 and the zigbee-herdsman-converters@15.0.51, zigbee-herdsman@0.14.94 dedupted.
See here: 
[image]

But it didn't change anything, the problem still occurs.
I've restarted iobroker as well as rebooted the raspi - didn't help out.

Is there any chance to fix this?

Thank you for any replies.

KR Uwe

### What did you expect to happen?

No automatic changes of the temperature by the device itself.

### How to reproduce it (minimal and precise)

Just wait, the changes happen every day.

### Zigbee2MQTT version

None

### Adapter firmware version

1.8.11 (Zigbee)

### Adapter

_TZE200_yw7cahqs

### Debug log

No log available, because the devices change their values automatically.

JohnRabbit1, 2023-03-28:
I had simular issues with the saswell.js file. Although my device is different: _TZE200_c88teujp.
After some investigation i discovered the  onEvent: tuya.onEventSetTime line does not do what it suppose to do, so i commented it out. And all problems were gone and after that i wrote my own external converter.

JohnRabbit1, 2023-03-28:
@Koenkk,

Yes it was. However I have changed the file a few months ago. I am not aware of recent changes. Even after the recent change to summertime i have a correct date/time. In the original saswell.js file the onEventSetTime function messed up the time: i.e. year was changed from 2023->2099. And that gave an unexpected result in the programmed schedule of course. I have attached the saswell.js file i am using now (yes, very dirty). However with some extra bonus functionality.
[saswell.zip](https://github.com/Koenkk/zigbee2mqtt/files/11089091/saswell.zip)

Koenkk, 2023-03-29:
@JohnRabbit1 initially the thermostat has to get its time from somewhere. When factory resetting the device or pulling out the batteries for 1 hour, is the time still correct?

JohnRabbit1, 2023-03-31:
@Koenkk I did the following:
1. Did a factory reset (long press + and PROG key);
2. Removed batteries for +/- 1,5 hours;
3. Forced remove the device from z2m
4. Reboot my Raspberry.
5. After that: checked time (year was 2099, probably some kind of NULL value). Month was correct. Day and hour not (old values)
6. Then: Add the device to z2m;
7. After some time (1 or 2 hours) day/time was correct
8. But keep in mind i am using my own saswell.js file mentioned above

Koenkk, 2023-04-01:
@JohnRabbit1 I've pushed a fix, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

UweLm, 2023-04-01:
@Koenkk Sorry for the late replay, was absent. I'll test it tomorrow and send you my feedback. I guess it will be installed using a Linux command. 

KR Uwe

 
 

Gesendet: Samstag, 01. April 2023 um 09:20 Uhr

JohnRabbit1, 2023-04-05:
@Koenkk: Sorry for my late response, but i am abroad coming month. So i cannot check your fix. However looking at your fix: i think it will partly solve the problem. In my solution i have an additional fix: fromZigbee: ......, hy_set_time_request] for time sync.
@UweLm: I am not familiar with iobroker. But the error you show looks like your files are in a different place:  /opt/iobroker/.... In my case they are in /opt/zigbee2mqtt/..... So you have to change that in the first few lines.

UweLm, 2023-04-19:
@JohnRabbit1: I've replaced the saswell.js on my spare RasPi4 running ioBroker connected via ZigBee to two thermostatic valves TZE200 which had the mentioned issue in #16859 with your copy. Of course, I've changed the directory in your version of saswell.js and up to now it works fine. No automatic changes of the target_temperature came up until now. This seems to be a positive indicator, but I want to monitor it a little more before I can be sure.
Can I use your saswell.js to replace the latest version on my "production" RasPi4 installation without triggering any problems?

Thanks a lot for your efforts!

 

KR Uwe

 
 

Gesendet: Mittwoch, 05. April 2023 um 09:56 Uhr

JohnRabbit1, 2023-04-20:
@UweLm: I am running this saswell.js script now for +/- 9 months, without any problems.  So i think it is stable enough. You can always put this saswell.js thing in the data directory and in the userinterface add it as an external converter. In this way you don’t have to delete the original one.

UweLm, 2023-04-20:
@JohnRabbit1: Rejoiced too soon...today the values of the two thermostatic valves have changed 3 times. No scripts are active, no entries in the log - nothing.
I haven't got that sentence: "You can always put this saswell.js thing in the data directory and in the userinterface add it as an external converter. In this way you don’t have to delete the original one."
The saswell.js is originally in /opt/iobroker/node_modules/zigbee-herdsman-converters/devices - for sure I can create a folder "/opt/data", copy your saswell.js into it and choose it in the ZigBee Adapter settings as an external converter. That means, it will not take the original saswell.js but yours in the subfolder "/opt/data"? Have I got that right?
 

Sorry, it's inconvenient for me to keep contacting you and bugging you about my problems with the saswell.js converter - I'm sure you have enough other things to do.
I appreciate you answering my mails - thanks for that!

 
KR Uwe

Gesendet: Donnerstag, 20. April 2023 um 12:09 Uhr

JohnRabbit1, 2023-04-20:
@UweLm: That’s a pity. As i said earlier: i’m not familiar with iobroker. I have used the installation as mentioned on zigbee2mqtt.io. 

It is installed in directory: /opt/zigbee2mqtt. A subdir: /opt/zigbee2mqtt/data contains all the config data like database etc. In my case it looked like:

 

configuration.yaml   coordinator_backup.json  database.db  database.db.backup  database.db.save  disk   log  saswell.js  state.json  ts011f.js

 

and saswell.js is the external converter

 

So you have to find this dir in iobroker

UweLm, 2023-05-21:
Rejoiced too soon...today (2023/04/20) the values of the two thermostatic valves have changed 3 times. No scripts are active, no entries in the log - nothing.
There might be a difference at the devices @JohnRabbit1 is using and the devices I have here. And I know it's hard to figure out what exactly is happening and how to fix it. Let me know if I can debug/check something. I work as a developer, but I'm not very experienced with Linux/Unix and the tools available, and I'm also completely lacking in the context of how the individual components act together. This makes searching for potentially problematic candidates quite difficult.
However, I have no problem with sending one of the thermostatic valves to you for testing, if that would help further

I appreciate your support - thanks for that!
 
KR Uwe

UweLm, 2023-06-21:
Still open.

closed 2023-07-29 by github-actions[bot]