Koenkk/zigbee2mqtt #16863: All Tuya TS110E dimmers broke after update! Cannot update state in all models

chojnicki, 2023-03-01:
### What happened?

I already reported issue with this dimmer in https://github.com/Koenkk/zigbee2mqtt/issues/16804 but it's not exactly related, that's why I need to make new one. In previous one it was about wrong support for new model/firmware `_TZ3210_k1msuvg6`. But I already have `_TZ3210_zxbtub8r` and `_TZ3210_wdexaypg` (2 gang) that was working before.

While debugging `_TZ3210_k1msuvg6` i updated Z2M (not sure what version I had before, but something around month old), and then tried Edge. But while doing this, my old dimmers broke. When I refresh state, it always shows as `on`, but it's not true. Not only that - it shows `on` with `brightness `0` - or maybe it really starts it with 0, not sure, and why would it do that.

So when I really click to turn on, it starts with brightness 0. What's worse - HA updates it in some period by itself (not sure why, but not important). Because of that, all my light become unusable, because I can't turn them on by automation (they start with brightness 0). 

Note that I DO NOT have custom converters. I also tried converter from #16804 shared by @Berlusconi13, and multiple other converters I found in old TS110E issues, but none of them fixes that problem. Of course same situation on both latest and edge version. I tried to tweak converter by myself, but no luck. My guess is that is somewhere deeper in tuya libs.

Since there is no easy way to downgrade addon in HomeAssistant - I'm without lights. My wife will kill me, so pleeease @Koenkk heeelp 🆘 

### What did you expect to happen?

Correct state update

### How to reproduce it (minimal and precise)

Refresh state

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

0x26680700

### Adapter

ConBee2

### Debug log

(turned off with max brightness from dashboard)
Info 2023-03-01 18:03:53MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":0,"brightness_1":254,"last_seen":"2023-03-01T17:04:31.057Z","linkquality":95,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"OFF","state_1":"OFF","switch_type":"momentary"}'

(click refresh state)
Debug 2023-03-01 18:03:59Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
Debug 2023-03-01 18:03:59Publishing get 'get' 'state' to 'office_ceiling_lamp'
Debug 2023-03-01 18:03:59Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-03-01 18:03:59MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":0,"brightness_1":254,"last_seen":"2023-03-01T17:04:37.486Z","linkquality":95,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"ON","state_1":"OFF","switch_type":"momentary"}'

Koenkk, 2023-03-06:
@chojnicki it won't fix the issue but I want to understand better what happens. Can you stop z2m, remove the device state from `data/state.json` start z2m and see if the `state_1` and `brightness_1` gets added to the published state?

chojnicki, 2023-03-06:
@Koenkk 
No, it's now. But I used converters, so maybe that's it.

**_TZ3210_zxbtub8r**

> Info 2023-03-06 19:54:59MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"last_seen":"2023-03-06T18:54:59.808Z","linkquality":159,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Debug 2023-03-06 19:55:00Received Zigbee message from 'office_ceiling_lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
Info 2023-03-06 19:55:00MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"last_seen":"2023-03-06T18:55:00.637Z","linkquality":159,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Debug 2023-03-06 19:55:04Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"brightness":""}'
Debug 2023-03-06 19:55:04Publishing get 'get' 'brightness' to 'office_ceiling_lamp'
Debug 2023-03-06 19:55:04Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID null
Info 2023-03-06 19:55:04MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"last_seen":"2023-03-06T18:55:04.279Z","linkquality":151,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Error 2023-03-06 19:55:04Publish 'get' 'brightness' to 'office_ceiling_lamp' failed: 'Error: Read 0xa4c138321f493293/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Info 2023-03-06 19:55:04MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'brightness' to 'office_ceiling_lamp' failed: 'Error: Read 0xa4c138321f493293/1 genLevelCtrl([\"currentLevel\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"office_ceiling_lamp"},"type":"zigbee_publish_error"}'
Debug 2023-03-06 19:55:41Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
Debug 2023-03-06 19:55:41Publishing get 'get' 'state' to 'office_ceiling_lamp'
Debug 2023-03-06 19:55:41Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-03-06 19:55:41MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"last_seen":"2023-03-06T18:55:41.646Z","linkquality":151,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Debug 2023-03-06 19:55:44Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/set' with data '{"state":"OFF"}'
Debug 2023-03-06 19:55:44Publishing 'set' 'state' to 'office_ceiling_lamp'
Info 2023-03-06 19:55:44MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"last_seen":"2023-03-06T18:55:44.419Z","linkquality":151,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Info 2023-03-06 19:55:44MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"last_seen":"2023-03-06T18:55:44.419Z","linkquality":151,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
Debug 2023-03-06 19:55:45Received Zigbee message from 'office_ceiling_lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":0}' from endpoint 1 with groupID null
Info 2023-03-06 19:55:45MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":0,"last_seen":"2023-03-06T18:55:45.230Z","linkquality":159,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
Debug 2023-03-06 19:55:45Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
Debug 2023-03-06 19:55:45Publishing get 'get' 'state' to 'office_ceiling_lamp'
Debug 2023-03-06 19:55:45Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-03-06 19:55:45MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":0,"last_seen":"2023-03-06T18:55:45.522Z","linkquality":159,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Debug 2023-03-06 19:55:49Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
Debug 2023-03-06 19:55:49Publishing get 'get' 'state' to 'office_ceiling_lamp'
Debug 2023-03-06 19:55:49Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-03-06 19:55:49MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":0,"last_seen":"2023-03-06T18:55:49.176Z","linkquality":159,"max_brightness":255,"min_brightness":1,"power_on_be [... cut ...]
> Debug 2023-03-06 19:58:29Received Zigbee message from 'living_room_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
> Info 2023-03-06 19:58:29MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_lamp', payload '{"brightness":1,"last_seen":"2023-03-06T18:58:29.336Z","linkquality":255,"min_brightness":131,"power_on_behavior":null,"state":"ON","switch_type":null}'

This how both of them looked like just after stop/start. I had to refresh manually every field.

Lights were off, but after refreshing state again state `ON` with `brightness` 0.

[image]

chojnicki, 2023-03-06:
@Koenkk I can even give you access to HA with Z2M and live stream to lamp if that will help you 😁

Koenkk, 2023-03-07:
```
(click refresh state)
Debug 2023-03-01 18:03:59Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
Debug 2023-03-01 18:03:59Publishing get 'get' 'state' to 'office_ceiling_lamp'
Debug 2023-03-01 18:03:59Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
```

Is `{"onOff":1}` logged when refreshing and the device is OFF?

chojnicki, 2023-03-07:
@Koenkk not sure if I understand correctly, so to be sure I will describe steps and provide logs again.

1. I turned light on by manually setting brightness to max.
2. Then turned it off by by clicking OFF on state. So light is for sure OFF.
[image]
3. Then I clicked refresh state, and it's now showing as ON in dashboard.
[image]

Logs from this step:

> Debug 2023-03-07 20:04:47Received MQTT message on 'zigbee2mqtt/living_room_ceiling_lamp/get' with data '{"state":""}'
> Debug 2023-03-07 20:04:47Publishing get 'get' 'state' to 'living_room_ceiling_lamp'
> Debug 2023-03-07 20:04:47Received Zigbee message from 'living_room_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
> Info 2023-03-07 20:04:47MQTT publish: topic 'zigbee2mqtt/living_room_ceiling_lamp', payload '{"brightness":0,"last_seen":"2023-03-07T19:04:47.189Z","linkquality":255,"min_brightness":131,"power_on_behavior":null,"state":"ON","switch_type":null}'

So yes, there is `{"onOff":1}`

Koenkk, 2023-03-07:
@chojnicki then the device reports the wrong state (1 = ON, 0 = OFF). I cannot do anything about that, it's a bug in the device firmware.

chojnicki, 2023-03-07:
@Koenkk But at least _TZ3210_zxbtub8r was working for months for me with Z2M until updates, and this device does not support OTA to be broken by firmware update 🤔

Koenkk, 2023-03-07:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS110E', ['_TZ3210_zxbtub8r', '_TZ3210_k1msuvg6']),
    model: 'TS110E_1gang_1',
    vendor: 'TuYa',
    description: '1 channel dimmer',
    fromZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
        .fromZigbee.concat([tuya.fz.power_on_behavior]),
    toZigbee: extend.light_onoff_brightness({disablePowerOnBehavior: true, disableMoveStep: true, disableTransition: true})
            .toZigbee.concat([tuya.tz.power_on_behavior]),
    exposes: [e.light_brightness()],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

chojnicki, 2023-03-07:
Nope, it removed min_brightness and max_brightness so for sure converted is loaded correctly, but refreshing state still gives exact same results.

chojnicki, 2023-03-07:
> This may be caused because we now control this device a little bit differently to workaround other bugs in the fw. Are there any other issues besides refreshing the state?

I described them in separate issue only for new model https://github.com/Koenkk/zigbee2mqtt/issues/16804.
But now I checked both again, on Edge(dev) version. No previous converter. And results:

**_TZ3210_zxbtub8r:**

1. Issue with refreshing state.
2. Cannot refres brightness `Publish 'get' 'brightness' to 'office_ceiling_lamp' failed: 'Error: Read 0xa4c138321f493293/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`
3. Turning light off also changes slider with brightness to 0 (which does not happen in any other light device I have in Z2M),

**_TZ3210_k1msuvg6:**

1. Issue with refreshing state.
2. Refreshing brightness changes slider to `1` (but in reality light is still bright).
3. Turning light off also changes slider with brightness to 0 (which does not happen in any other light device I have in Z2M),
4. Changing brightness to any value causes decrease -1 after second. So after setting it to 150, it changes by itself to 149.

Koenkk, 2023-03-08:
Can you provide the debug log when doing this with the latest known working version?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

chojnicki, 2023-03-08:
I did found way to downgrade Z2M in HA (addon) :/ I found git commands about switching branch, but I think this is for standalone z2m, no idea how to do this for HA addon.

Koenkk, 2023-03-08:
What was the last known working version? I can you provide you an external converter to match the behavior of that version.

chojnicki, 2023-03-08:
Like I said in first post, I updated Z2M (normal version, not edge) only while debugging new dimmer model, and I'm not sure what version was before, but probably around month old, so maybe 1.29.*

Koenkk, 2023-03-09:
@chojnicki this external converter gives the behaviour from 1.28.2:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS110E', manufacturerName: '_TZ3210_zxbtub8r'}, {modelID: 'TS110E', manufacturerName: '_TZ3210_ngqk6jia'}],
    model: 'TS110E_1gang',
    vendor: 'Lonsonho',
    description: 'Zigbee smart dimmer module 1 gang with neutral',
    extend: extend.light_onoff_brightness(),
};

module.exports = definition;

```

Can you try if it works with that? If not, since HA doesn't support downgrading, can you setup a separate z2m environment with the working version and provide me the debug logging?

chojnicki, 2023-03-09:
@Koenkk still exact same issues :/

Problem is - I have only one zigbee adapter, installed on computer with HA system. I cannot set up separate Z2M :(
I will try again to find a way to downgrade Z2M in HA, but not sure if it's even possible in any way.

chojnicki, 2023-03-09:
@Koenkk I forgot that I can just restore HA backup...

Reversed to 1.29.2-1. Tested on `_TZ3210_zxbtub8r` since I remember that `_TZ3210_k1msuvg6` was not working anyway on old version. But on both models refresh state is not bugged anymore :)

> Light off, clicking refresh state:
> 
> Debug 2023-03-09 18:17:47Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
> Debug 2023-03-09 18:17:47Publishing get 'get' 'state' to 'office_ceiling_lamp'
> Debug 2023-03-09 18:17:47Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
> Info 2023-03-09 18:17:47MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"brightness_1":0,"last_seen":"2023-03-09T17:17:47.915Z","linkquality":255,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"OFF","state_1":"ON","switch_type":"momentary"}'
> 
> 
> Click state ON:
> Debug 2023-03-09 18:20:16Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/set' with data '{"state":"ON"}'
> Debug 2023-03-09 18:20:16Publishing 'set' 'state' to 'office_ceiling_lamp'
> Info 2023-03-09 18:20:16MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"brightness_1":0,"last_seen":"2023-03-09T17:20:16.166Z","linkquality":191,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"OFF","state_1":"ON","switch_type":"momentary"}'
> Info 2023-03-09 18:20:16MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"brightness_1":0,"last_seen":"2023-03-09T17:20:16.166Z","linkquality":191,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","state_1":"ON","switch_type":"momentary"}'
> Debug 2023-03-09 18:20:16Received Zigbee message from 'office_ceiling_lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
> Info 2023-03-09 18:20:17MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"brightness_1":254,"last_seen":"2023-03-09T17:20:16.971Z","linkquality":199,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","state_1":"ON","switch_type":"momentary"}'
> 
> 
> Click refresh state:
> 
> Debug 2023-03-09 18:22:26Received MQTT message on 'zigbee2mqtt/office_ceiling_lamp/get' with data '{"state":""}'
> Debug 2023-03-09 18:22:26Publishing get 'get' 'state' to 'office_ceiling_lamp'
> Debug 2023-03-09 18:22:27Received Zigbee message from 'office_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
> Info 2023-03-09 18:22:27MQTT publish: topic 'zigbee2mqtt/office_ceiling_lamp', payload '{"brightness":254,"last_seen":"2023-03-09T17:22:27.022Z","linkquality":191,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","state_1":"ON","switch_type":"momentary"}'

Koenkk, 2023-03-11:
@chojnicki I think I understand what happens. In 1.29.2 there was a bug when refreshing state it would update `state_1` (which is still incorrect and not used) instead of `state`. To confirm this, stop z2m remove the device state from `data/state.json` and refresh again. If `state_1` appears this is the diff with 1.30.*.

chojnicki, 2023-03-11:
@Koenkk Yes that's correct

Koenkk, 2023-03-12:
I see, so then the issue was also there in 1.29.2 but in 1.29.2 there is another bug which causes the state to be published on the wrong property (`state_1` instead of `state`)

chojnicki, 2023-03-12:
Hm co I'm confused - what can be done about it? I understand that device firmware sucks, but at the same time on previous Z2M version it's at least usable :)

Koenkk, 2023-03-12:
@chojnicki device provides the wrong state so there is nothing we can do about that. What I can do is to disable the refresh button but this is tricky since I'm not sure this bug applies to all versions of this device. The easiest workaround would be to not click the refresh button

chojnicki, 2023-03-12:
@Koenkk problem is it's not just about refreshing in dashboard. Like eexplained previously, device updates state by itself after some time. So I turn light off, it becomes on with brightness 0 in HA by itself, so I cant turn light on again.

chojnicki, 2023-03-12:
@Koenkk more logs for you. I checked what's is up with that state reporting, and I see that device does report state correctly after turning light on/off directly in device (pins). Normally I'm not using that, just my own automations, by since device reports changed state, I thing it should do it also while refreshing or any other action?

I'm using new device that as never connected to Z2M/HA, on version 1.29.

Can you find something useful in that?

Light is on, turn it off by pins on device:

> Debug 2023-03-12 16:18:12Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
> Info 2023-03-12 16:18:12MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:18:12.231Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"ON","state_1":"OFF","switch_type":"momentary","switch_type_1":"momentary"}'

Now turn it on by pins on device:

> 2023-03-12 16:18:40Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
> Info 2023-03-12 16:18:40MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:18:40.660Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"ON","state_1":"ON","switch_type":"momentary","switch_type_1":"momentary"}'

I'm NOT touching dashboard (no refresh state) so device is updating it by itself as it should.

We can see that `state_1` changes to real state, so device reports it correctly.

Now tests in dashboard:

Light is on, turn it off by dashboard:

> Debug 2023-03-12 16:20:49Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/set' with data '{"state":"OFF"}'
> Debug 2023-03-12 16:20:49Publishing 'set' 'state' to 'bedroom_ceiling_lamp'
> Info 2023-03-12 16:20:49MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:20:49.896Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"ON","state_1":"ON","switch_type":"momentary","switch_type_1":"momentary"}'
> Info 2023-03-12 16:20:49MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:20:49.896Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"OFF","state_1":"ON","switch_type":"momentary","switch_type_1":"momentary"}'
> Debug 2023-03-12 16:20:49Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
> Info 2023-03-12 16:20:49MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:20:49.909Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"OFF","state_1":"OFF","switch_type":"momentary","switch_type_1":"momentary"}'

It updated state correctly in dashboard, but let's click refresh state anyway:

> Debug 2023-03-12 16:21:41Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/get' with data '{"state":""}'
> Debug 2023-03-12 16:21:41Publishing get 'get' 'state' to 'bedroom_ceiling_lamp'
> Debug 2023-03-12 16:21:41Received Zigbee message from 'bedroom_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
> Info 2023-03-12 16:21:41MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:21:41.452Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"OFF","state_1":"OFF","switch_type":"momentary","switch_type_1":"momentary"}'

Turn on in dashboard:

> Debug 2023-03-12 16:22:19Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/set' with data '{"state":"ON"}'
> Debug 2023-03-12 16:22:19Publishing 'set' 'state' to 'bedroom_ceiling_lamp'
> Info 2023-03-12 16:22:19MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:22:19.186Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"OFF","state_1":"OFF","switch_type":"momentary","switch_type_1":"momentary"}'
> Info 2023-03-12 16:22:19MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:22:19.186Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"ON","state_1":"OFF","switch_type":"momentary","switch_type_1":"momentary"}'
> Debug 2023-03-12 16:22:19Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
[... cut ...]
> Debug 2023-03-12 16:22:35Received Zigbee message from 'bedroom_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
> Info 2023-03-12 16:22:35MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"brightness_1":254,"last_seen":"2023-03-12T15:22:35.410Z","linkquality":255,"max_brightness":255,"min_brightness":4,"power_on_behavior":"previous","power_on_behavior_1":"previous","state":"ON","state_1":"OFF","switch_type":"momentary","switch_type_1":"momentary"}'

Koenkk, 2023-03-13:
@chojnicki when using z2m 1.30.2 and not using any external converter and clearing the state (stop z2m, remove from data/state.json and start z2m), does the `state_1` property appear again?

chojnicki, 2023-03-13:
@Koenkk I will make sure later, but if I remember correctly, 95% for no.

chojnicki, 2023-03-15:
@Koenkk did that, and not there is no `state_1` on latest version.
Still issue with state, but not opposite, after refresh it always shows as "OFF".

Removed from state.json, started addon and just waited:

> Debug 2023-03-15 16:51:38Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":64}' from endpoint 1 with groupID null
> Info 2023-03-15 16:51:38MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"last_seen":"2023-03-15T15:51:38.569Z","linkquality":111,"power_on_behavior":null,"state":null,"switch_type":null}'
> 

After refreshing everything (while light is off) and then clicking refresh state:

> Debug 2023-03-15 16:55:17Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/get' with data '{"state":""}'
> Debug 2023-03-15 16:55:17Publishing get 'get' 'state' to 'bedroom_ceiling_lamp'
> Debug 2023-03-15 16:55:17Received Zigbee message from 'bedroom_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
> Info 2023-03-15 16:55:17MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":254,"last_seen":"2023-03-15T15:55:17.746Z","linkquality":95,"max_brightness":196,"min_brightness":132,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
> 

Light is on, clicked refresh state - dashboard now shows it as off (but it is still on):

> Debug 2023-03-15 16:58:49Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/get' with data '{"state":""}'
> Debug 2023-03-15 16:58:49Publishing get 'get' 'state' to 'bedroom_ceiling_lamp'
> Debug 2023-03-15 16:58:49Received Zigbee message from 'bedroom_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
> Info 2023-03-15 16:58:49MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-15T15:58:49.857Z","linkquality":111,"max_brightness":196,"min_brightness":132,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
>

Koenkk, 2023-03-15:
> Light is on, clicked refresh state - dashboard now shows it as off (but it is still on):

- How did you turn the light on? I'm wondering why the state is still OFF.
- The state published by the device is incorrect (onOff == 0 means device is off). I cannot solve this incorrect state reporting in z2m.
```
Debug 2023-03-15 16:58:49Received Zigbee message from 'bedroom_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-03-15 16:58:49MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-15T15:58:49.857Z","linkquality":111,"max_brightness":196,"min_brightness":132,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
```

chojnicki, 2023-03-15:
@Koenkk I turned it on in dashboard

Koenkk, 2023-03-16:
When turning on via the dashboard I would expect the state is on. Can you provide me the debug logging of the following sequence:
1. Device is OFF
2. Turn device ON via the z2m dashboard
3. Do a refresh of the state

jacekmarek, 2023-03-17:
Hi @Koenkk 
I have severar TS110E_1gang_1.html, Zigbee Manufacturer: TZ3210_zxbtub8r, Z2M version 1.30.2.
When I refresh state on brightness on Z2M dashboard I got message:
```
debug 2023-03-17 17:11:21Received MQTT message on 'zigbee2mqtt/0xa4c138861f3733e6/get' with data '{"brightness":""}'
debug 2023-03-17 17:11:21Publishing get 'get' 'brightness' to '0xa4c138861f3733e6'
debug 2023-03-17 17:11:21Received Zigbee message from '0xa4c138861f3733e6', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID null
info 2023-03-17 17:11:21MQTT publish: topic 'zigbee2mqtt/0xa4c138861f3733e6', payload '{"last_seen":"2023-03-17T17:11:21+01:00","linkquality":255,"max_brightness":108,"min_brightness":9,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
error 2023-03-17 17:11:21Publish 'get' 'brightness' to '0xa4c138861f3733e6' failed: 'Error: Read 0xa4c138861f3733e6/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2023-03-17 17:11:21Error: Read 0xa4c138861f3733e6/1 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:998:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:274:21)
info 2023-03-17 17:11:21MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'brightness' to '0xa4c138861f3733e6' failed: 'Error: Read 0xa4c138861f3733e6/1 genLevelCtrl([\"currentLevel\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"0xa4c138861f3733e6"},"type":"zigbee_publish_error"}'
```

Koenkk, 2023-03-18:
@jacekmarek in your case the device does not support reading it (`UNSUPPORTED_ATTRIBUTE`)

jacekmarek, 2023-03-19:
>in your case the device does not support reading it (UNSUPPORTED_ATTRIBUTE)
New purchased TuYa products have limited functionality :( .

chojnicki, 2023-03-19:
@Koenkk here you go

```
Turn of via dashboard:
Debug 2023-03-19 18:13:39Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/set' with data '{"state":"ON"}'
Debug 2023-03-19 18:13:39Publishing 'set' 'state' to 'bedroom_ceiling_lamp'
Info 2023-03-19 18:13:39MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-19T17:13:39.066Z","linkquality":255,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
Debug 2023-03-19 18:13:39Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-03-19 18:13:39MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-19T17:13:39.081Z","linkquality":255,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Info 2023-03-19 18:13:39MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-19T17:13:39.109Z","linkquality":247,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Info 2023-03-19 18:13:39MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-19T17:13:39.109Z","linkquality":247,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
Debug 2023-03-19 18:13:39Received Zigbee message from 'bedroom_ceiling_lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2023-03-19 18:13:39MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-19T17:13:39.124Z","linkquality":247,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'

Refresh state:
Debug 2023-03-19 18:13:48Received MQTT message on 'zigbee2mqtt/bedroom_ceiling_lamp/get' with data '{"state":""}'
Debug 2023-03-19 18:13:48Publishing get 'get' 'state' to 'bedroom_ceiling_lamp'
Debug 2023-03-19 18:13:48Received Zigbee message from 'bedroom_ceiling_lamp', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info 2023-03-19 18:13:48MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling_lamp', payload '{"brightness":253,"last_seen":"2023-03-19T17:13:48.810Z","linkquality":231,"max_brightness":255,"min_brightness":1,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'

Now it's showing as off in dashboard, but light is still on.
```

Koenkk, 2023-03-20:
@chojnicki this is the same as we already found in https://github.com/Koenkk/zigbee2mqtt/issues/16863#issuecomment-1470353771

Device returns the wrong state when doing a GET. I cannot do anything about that, its a bug in the firmware of the device. The reason why this didn't happend in earlier z2m versions is because a GET would update the `state_1` (which is not used and incorrect) instead of `state`. Are there any issues when never refreshing the state?

Koenkk, 2023-03-21:
> The one with decreasing brightness to -1 by itself.

This one should be fixable, pushed a possible fix. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

> device or z2m/hs is updating this by itself.

Can you provide the debug log when the device state updates by itself?

jacekmarek, 2023-03-26:
>But I already have _TZ3210_zxbtub8r and _TZ3210_wdexaypg (2 gang) that was working before.

Hi @Koenkk 
I also have TS110E_2gang_1: _TZ3210_wdexaypg and  _TZ3210_3mpwqzuu ( Z2M current version 1.30.2.), both models have a problem with brightness - refresh state.
State of _TZ3210_3mpwqzuu is very strange also:

```
{
    "brightness_l1": 210,
    "brightness_l2": 251,
    "last_seen": "2023-03-26T18:12:47+02:00",
    "linkquality": 244,
    "max_brightness": 255,
    "min_brightness": 1,
    "state_l1": "OFF",
    "state_l2": "OFF",
    "switch_type_l1": "momentary",
    "switch_type_l2": "momentary",
    "power_on_behavior": "off",
    "min_brightness_l1": 1,
    "max_brightness_l1": 255,
    "power_on_behavior_l1": "off",
    "min_brightness_l2": 1,
    "max_brightness_l2": 255,
    "brightness": 209
}
```
The light l2 is ON, I set brightness:230 on Endpoint: l2:
```
debug 2023-03-26 18:22:14Received MQTT message on 'zigbee2mqtt/0xa4c138fe17711517/set' with data '{"brightness_l2":230}'
debug 2023-03-26 18:22:14Publishing 'set' 'brightness' to '0xa4c138fe17711517'
info 2023-03-26 18:22:14MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517', payload '{"brightness":209,"brightness_l1":210,"brightness_l2":251,"last_seen":"2023-03-26T18:22:13+02:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":255,"min_brightness":1,"min_brightness_l1":1,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"off","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2023-03-26 18:22:14MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517/l1', payload '{"brightness":210,"max_brightness":255,"min_brightness":1,"power_on_behavior":"off","state":"OFF","switch_type":"momentary"}'
info 2023-03-26 18:22:14MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517/l2', payload '{"brightness":251,"max_brightness":255,"min_brightness":1,"state":"ON","switch_type":"momentary"}'
info 2023-03-26 18:22:14MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517', payload '{"brightness":209,"brightness_l1":210,"brightness_l2":251,"last_seen":"2023-03-26T18:22:13+02:00","linkquality":236,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":255,"min_brightness":1,"min_brightness_l1":1,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"off","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
```
I check (publish:get) brightness on Endpoint: l2 now:  
```
debug 2023-03-26 18:23:27Received MQTT message on 'zigbee2mqtt/0xa4c138fe17711517/get' with data '{"brightness_l2":""}'
debug 2023-03-26 18:23:27Publishing get 'get' 'brightness' to '0xa4c138fe17711517'
debug 2023-03-26 18:23:27Received Zigbee message from '0xa4c138fe17711517', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":251}' from endpoint 2 with groupID null
info 2023-03-26 18:23:27MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517', payload '{"brightness":229,"brightness_l1":210,"brightness_l2":251,"last_seen":"2023-03-26T18:23:26+02:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":255,"min_brightness":1,"min_brightness_l1":1,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"off","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2023-03-26 18:23:27MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517/l1', payload '{"brightness":210,"max_brightness":255,"min_brightness":1,"power_on_behavior":"off","state":"OFF","switch_type":"momentary"}'
info 2023-03-26 18:23:27MQTT publish: topic 'zigbee2mqtt/0xa4c138fe17711517/l2', payload '{"brightness":251,"max_brightness":255,"min_brightness":1,"state":"ON","switch_type":"momentary"}'
```
and got brightness:251, I had previously set brightness:230.

jacekmarek, 2023-03-28:
@Koenkk 
>herdsman debug log when setting the brightness
In attached file
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/11088738/debug.txt)

Koenkk, 2023-03-28:
```
I check (publish:get) brightness on Endpoint: l1 now:

2023-03-28T11:25:09.688Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 253
2023-03-28T11:25:09.689Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":253,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":32,"attrData":255}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":53547,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2023-03-28 13:25:09: Received Zigbee message from '0xa4c13895232f30db', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":255}' from endpoint 1 with groupID null
```

Similar as what we saw with @chojnicki were it reported the wrong state when doing a GET, yours reports a wrong brightness after doing a GET (`'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":255}'`). This is an issue in the device firmware which I cannot fix from z2m.

jacekmarek, 2023-03-28:
@Koenkk Thanks for anwser.
Another Tuya devices with wrong firmware :(.
Today the next delivery of equipment, I will check how the new ones will work.

[image]

jacekmarek, 2023-04-16:
> When I refresh state on brightness on Z2M dashboard I got message:
> UNSUPPORTED_ATTRIBUTE
>@jacekmarek in your case the device does not support reading it (UNSUPPORTED_ATTRIBUTE)
Hi @Koenkk 
I bought a Tuya gateway to check the operation of these module  - TS110E_1gang_1 (Zigbee Manufacturer: TZ3210_zxbtub8r) and everything works fine. I did a test:
  - I turned on the switch from the TuYa application and set the brightness to 10% level.
  - I turned off the Tuya gateway (power off)
  - I changed the brightness to 70%  using the button
  - I turned off the module with the button
  - I turned on the Tuya gateway (power on)
  - after a few seconds the gateway correctly read the module status, on TuYa application show status of module: state = off, brightness = 70%. This proves that you can read the brightness level correctly from the module, when I try to read from Z2M, I get a message UNSUPPORTED_ATTRIBUTE, why? Where is the problem?

Koenkk, 2023-04-16:
> I also did a test with ZHA, also everything works fine,

That is interesting, do you have any ZHA debug logging so I can see what requests it send?

jacekmarek, 2023-04-17:
>That is interesting, do you have any ZHA debug logging so I can see what requests it send?
No, I don't because I did it quickly, I don't use ZHA integration in my HA, I've been using Z2M forever :). 
I'll try to collect the logs.

DivusJulius, 2023-05-07:
I have a _TZ3210_3mpwqzuu that also incorrectly reports the state when pressing the refresh button or reading the genOnOff Attribute using the dev console. This is only really a problem because either Home Assistant or Z2M will occassionally refresh the state automatically resulting in an incorrect reported status. 
I changed the value of the genOnOff Attribute using the Dev Console to 0 instead of 1. Now the Dimmer always reports the state as being off when it gets refreshed, essentially flipping the problem on its head.

The physical switches, set to toggle, are however always able to correcty switch the lamps on and off regardless of which state they report. 

@Koenkk 
I had this device setup in ZHA for 1h or so and have not noticed this problem. I then switched to Z2M because ZHA does not support the "Switch Type" setting that I need for this particular device. I'll set it up in ZHA again to see if the same problem exists and to capture some debug logs.

evangelo18, 2023-06-28:
I have 5 of these devices which I recently installed. 4 out of 5 work, however they are extremely slow to respond. Sometimes not at all. 1 of the 5 just refused to respond to any commands.

patapovich, 2024-01-25:
@Koenkk Is it possible to reopen this? issue still exists at least for _TZ3210_3mpwqzuu

Koenkk, 2024-01-26:
@patapovich what issues are still present with z2m 1.35.1?

patapovich, 2024-01-26:
> @patapovich what issues are still present with z2m 1.35.1?

Erratic brightness level reports in HA vs. setting, and the biggest issue is the state of L1 and L2 (on vs. off) usually being reported inverted when changed through Home Assistant.

I have a CC2531 stick on the way so that I can sniff the traffic and compare Zigbee2MQTT to the Tuya gateway, but it is still in transit.

Koenkk, 2024-01-28:
> usually being reported inverted when changed through Home Assistant.

Can you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

jacekmarek, 2024-01-29:
Will such logs be helpful? I turned off L1 via GUI Z2M (tab Exposes), now I can't turn on L1 using the button on device:
```
debug 2024-01-29 18:57:25Received MQTT message on 'zigbee2mqtt/0xa4c13895232f30db/set' with data '{"state_l1":"OFF"}'
debug 2024-01-29 18:57:25Publishing 'set' 'state' to '0xa4c13895232f30db'
info 2024-01-29 18:57:25MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db', payload '{"brightness":214,"brightness_l1":214,"brightness_l2":112,"last_seen":"2024-01-29T18:57:25+01:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":252,"min_brightness":14,"min_brightness_l1":14,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"ON","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2024-01-29 18:57:25MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l1', payload '{"brightness":214,"max_brightness":255,"min_brightness":14,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
info 2024-01-29 18:57:25MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l2', payload '{"brightness":112,"max_brightness":252,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
info 2024-01-29 18:57:25MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db', payload '{"brightness":214,"brightness_l1":214,"brightness_l2":112,"last_seen":"2024-01-29T18:57:25+01:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":252,"min_brightness":14,"min_brightness_l1":14,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2024-01-29 18:57:25MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l1', payload '{"brightness":214,"max_brightness":255,"min_brightness":14,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
info 2024-01-29 18:57:25MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l2', payload '{"brightness":112,"max_brightness":252,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
debug 2024-01-29 18:57:26Received Zigbee message from '0xa4c13895232f30db', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":0,"currentLevel":0}' from endpoint 1 with groupID null
info 2024-01-29 18:57:26MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db', payload '{"brightness":0,"brightness_l1":0,"brightness_l2":112,"last_seen":"2024-01-29T18:57:26+01:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":252,"min_brightness":14,"min_brightness_l1":14,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2024-01-29 18:57:26MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l1', payload '{"brightness":0,"max_brightness":255,"min_brightness":14,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
info 2024-01-29 18:57:26MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l2', payload '{"brightness":112,"max_brightness":252,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
debug 2024-01-29 18:57:33Received Zigbee message from '0xa4c13895232f30db', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db', payload '{"action":"off_l1","brightness":0,"brightness_l1":0,"brightness_l2":112,"last_seen":"2024-01-29T18:57:33+01:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":252,"min_brightness":14,"min_brightness_l1":14,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l1', payload '{"brightness":0,"max_brightness":255,"min_brightness":14,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l2', payload '{"brightness":112,"max_brightness":252,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db', payload '{"action":"","brightness":0,"brightness_l1":0,"brightness_l2":112,"last_seen":"2024-01-29T18:57:33+01:00","linkquality":255,"max_brightness":255,"max_brightness_l1":255,"max_brightness_l2":252,"min_brightness":14,"min_brightness_l1":14,"min_brightness_l2":1,"power_on_behavior":"off","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","state_l1":"OFF","state_l2":"ON","switch_type_l1":"momentary","switch_type_l2":"momentary"}'
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/action', payload 'off_l1'
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l1', payload '{"brightness":0,"max_brightness":255,"min_brightness":14,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
info 2024-01-29 18:57:33MQTT publish: topic 'zigbee2mqtt/0xa4c13895232f30db/l2', payload '{"brightness":112,"max_brightness":252,"min_brightness":1,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
```

pedroza26, 2024-04-03:
Issue still persist.. :(

satzmann666, 2024-04-22:
I am also experiencing some of these issues with my TS110E_2gang_1 (_TZ3210_3mpwqzuu), specifically: 
- On/off state of the HA entities and on the expose tab of Z2M is not updated correctly when changed using switch connected to the module, however the values of "state_l1" and "state_l2" change properly in the state tab.
- Turning the light off from within HA using the service, or on the expose tab of the device in Z2M also sets brightness to 0.
- Turning the light on using connected switch then sets the state to on, however the light does not come on, due to brightness being 0.
- Turning the light on using HA or Z2M expose tab after being turned off the same way and brightness changed to 0 restores previous brightness value.
- Using dev console to send regular command to endpoint 1 (light1), cluster 6 (genOnOff), id 0 (Off) properly turns off the light and does not change brightness.
- I also fail to understand what is the purpose of state keys without "_l1"/"_l2" suffix (ie. what is "brightness" for when there is already specific "brightness_l1" and "brightness_l2" for each light).
I will gladly provide any assistance in the form of debug logs if asked, considering that the onOff cluster seems to work properly I think this should be hopefully possible to remedy.

satzmann666, 2024-04-23:
So I have created external converter file by copying code from [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/7ee566b85c310c9f1b9cecb63d78e609d3b216e1/src/devices/tuya.ts#L6049) but excluded the `tzLocal.TS110E_light_onoff_brightness` converter and it seems to be working as intended. At least no more brightness changing to 0 when turned off in HA, but I would like to understand better what was the reason for this custom converter in the first place.

cbriddell, 2024-08-25:
Also adding to this, I have a dozen of these same 'Tuya' dimmers that have exactly the same issue. Dimmers constantly reporting wrong on/off states, wrong brightness levels, and also being jumpy. I've been going stir crazy for weeks trying to get to the bottom of it. Then I found this thread :)

RoGu777, 2024-09-03:
was now anyone able to fix the issue of brightness=0 for _TZ3210_k1msuvg6?

@satzmann666 could you please share your converter file?

kiksia95, 2024-09-21:
@satzmann666  could you please share your converter file?

RoGu777, 2024-09-22:
I was able to fix the problem in #23967 for _TZ3210_k1msuvg6, maybe it will help for other devices too

RoGu777, 2024-10-29:
Hi @satzmann666 @jacekmarek 
did you find any solution for wrongly refreshing of ON / OFF states or the brightness value?

I've created a new issue #24535, as this ticket seem to be ignored

mhu1987, 2024-11-21:
I can also confirm that this bug still exist with _TZ3210_k1msuvg6

mscv, 2024-11-25:
The state reporting issue shows up for me on:
- _TZ3210_k1msuvg6 (TS110E_1gang_2)
- _TZ3210_3mpwqzuu (TS110E_2gang_2)

To add to the previous discoveries:
- the devices **broadcast state changes properly** (i.e. when state changed either via onoff command or via physical switch).
- due to the bug in firmware the state changes (both physical and via command) do not **write** to genOnOff/onOff **attribute**
- the current converter for both devices binds reporting which adds to the confusion

Ideal solution (no idea if feasible):
- on state change received from the devices **write** to the genOnOff/onOff attribute with proper value

Alternative workaround:
- disable reporting/querying genOnOff/onOff for the devices

[3 more replies]

closed 2025-03-12 by github-actions[bot]