Koenkk/zigbee2mqtt #16869: Server stops running

tkarlsson, 2023-03-01:
### What happened?

After running a few days Zigbee2MQTT fails with the following error:

Zigbee2MQTT:debug 2023-02-27 17:10:41: Received Zigbee message from 'Huvudbrytare magasin', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"attributes":{"16384":[0,323504],"divisor":1000,"powerFactor":60},"clusterID":1794,"manufacturerCode":4190},"commandID":161,"frameCounter":128,"gppGddLink":10,"gppNwkAddr":14777,"options":22544,"payloadSize":23,"srcID":3791926705}' from endpoint 242 with groupID 13745
Zigbee2MQTT:info  2023-02-27 17:10:42: MQTT publish: topic 'zigbee2mqtt/Huvudbrytare magasin', payload '{"ac_frequency":50.1,"current_phase_a":0.4,"current_phase_b":0.19,"current_phase_c":0.64,"energy":323.504,"energy_phase_a":null,"energy_phase_b":null,"energy_phase_c":null,"linkquality":0,"power":176,"power_apparent":295,"power_factor":60,"power_phase_a":null,"power_phase_b":null,"power_phase_c":null,"voltage_phase_a":null,"voltage_phase_ab":407.8,"voltage_phase_b":null,"voltage_phase_bc":407.7,"voltage_phase_c":null,"voltage_phase_ca":408.4}'
Error: Write 0x1fff0001000000fe/1 genTime({"time":730833020}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

### Zigbee2MQTT version

1.30.2-dev

### Adapter firmware version

20210708

### Adapter

Sonoff

tkarlsson, 2023-03-01:
The version is actually the one before but couldn't find the version number. Digest:
96083f800b6d1b8efffafcccd623215f0cf04b7909249540d9b496184f9bd679

Koenkk, 2023-03-02:
Try updating your adapter firmware to the latest one, you are running an old version (`20210708`)

tkarlsson, 2023-03-05:
Upgraded and still have the same issue:

Zigbee2MQTT version
1.30.2-dev commit: [f865c97](https://github.com/Koenkk/zigbee2mqtt/commit/f865c97)
Coordinator type
zStack3x0
Coordinator revision
20221226

Zigbee2MQTT:info  2023-03-04 11:49:50: MQTT publish: topic 'zigbee2mqtt/Trefas slakteri', payload '{"ac_frequency":50.1,"current_phase_a":0,"current_phase_b":0,"current_phase_c":0.36,"energy":70.919,"energy_phase_a":null,"energy_phase_b":null,"energy_phase_c":null,"linkquality":0,"power":45,"power_apparent":84,"power_factor":53,"power_phase_a":null,"power_phase_b":null,"power_phase_c":null,"voltage_phase_a":null,"voltage_phase_ab":412.9,"voltage_phase_b":null,"voltage_phase_bc":411,"voltage_phase_c":null,"voltage_phase_ca":411.7}'
Error: Write 0x1fff0001000000fe/1 genTime({"time":731245763}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Koenkk, 2023-03-06:
What device is `0x1fff0001000000fe`? (can you provide your data/database.db file?)

tkarlsson, 2023-03-06:
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0029dc0d76","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1678116726256,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Router","ieeeAddr":"0xf082c0fffe89fa1d","nwkAddr":39135,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI Signal Repeater","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3,9,2821,4096,64636],"outClusterList":[25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI Signal Repeater","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190311","swBuildId":"2.3.086"}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0d76","endpointID":242}],"configuredReportings":[{"cluster":0,"attrId":5,"minRepIntval":3600,"maxRepIntval":14400}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190311","swBuildId":"2.3.086","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1678113392496,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"GreenPower","ieeeAddr":"0x00000000e203e53f","nwkAddr":58687,"manufId":null,"modelId":"GreenPower_254","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1678119121690,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"GreenPower","ieeeAddr":"0x00000000e203e1cf","nwkAddr":57807,"manufId":null,"modelId":"GreenPower_254","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1678119117940,"defaultSendRequestWhen":"immediate"}
{"id":5,"type":"Router","ieeeAddr":"0x70ac08fffe7b96ba","nwkAddr":48758,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI Signal Repeater","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3,9,2821,4096,64636],"outClusterList":[25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI Signal Repeater","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190311","swBuildId":"2.3.086"}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0d76","endpointID":1}],"configuredReportings":[{"cluster":0,"attrId":5,"minRepIntval":3600,"maxRepIntval":14400}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190311","swBuildId":"2.3.086","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1678111210651,"defaultSendRequestWhen":"immediate"}
{"id":6,"type":"Router","ieeeAddr":"0xe0798dfffee7e12f","nwkAddr":21492,"manufId":4880,"manufName":"AL001","powerSource":"Mains (single phase)","modelId":"WG001-Z01","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3],"outClusterList":[3,25],"clust [... cut ...] ,129,224,0,0,8,152]},"631":0,"633":500,"634":0,"635":1,"636":0,"637":0,"638":0,"640":0,"1034":100}},"hvacThermostat":{"attributes":{"localTemp":1620,"occupiedHeatingSetpoint":1550}},"genPowerCfg":{"attributes":{"batteryVoltage":33}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":25,"hwVersion":1,"dateCode":"Jun  7 2022","swBuildId":"0.0.0_0825","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1102440882},"lastSeen":1678116768830,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-03-06:
Pushed a fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

tkarlsson, 2023-03-08:
Seems like it did the trick. Been running for a day now without restarts but will monitor. Used to take 24-48h before the server shut down.

closed 2023-03-08 by Koenkk