Koenkk/zigbee2mqtt #16870: ZigBee2Mqtt not starting

wbsantos, 2023-03-02:
### What happened?

After the update to the 1.30.2-1 version, the server is no long starting successfully.

### What did you expect to happen?

It should start as usual.

### How to reproduce it (minimal and precise)

Not sure, on my server happens every time on start.

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

20210708

### Adapter

SonOff Plus

### Debug log

[21:49:20] INFO: Preparing to start...
[21:49:20] INFO: Socat not enabled
[21:49:21] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)

junosuarez, 2023-03-02:
From the debug log, it this error is happening when zigbee2mqtt is trying to load an External Converter at startup. You can troubleshoot to try to see which one is causing the problem by commenting them all out in your configuration.yaml (adding a `#` at the beginning of the line in yaml) and then adding them back in one at a time and restarting zigbee2mqtt until you see the one that breaks.

Can you share which converter is breaking for you?

It's possible that the converter will need to be upgraded to work with this new version.

wbsantos, 2023-03-02:
Thanks for the tip, I have found the issue.

I have written a custom converter for a tuya lamp model TS0505B. My script was using the actual tuya converters on the toZigbee property in order to add color_power_on_behavior and do_not_disturb.

However, I have found a commit in zigbee2mqtt-converters repo that changed the tuya_do_not_disturb and tuya_color_power_on_behavior to another file, and the two functions are now added by default to a new set of Tuya based models (including the model I own TS0505B).

Long story short, I removed my custom converter and had to do nothing more. Thanks for the help, I will be closing the issue.

To add information to help anyone else that might need it, my converter code (that I no longer need) was the follow:

```javascript
const extend = require('zigbee-herdsman-converters/lib/extend');

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

module.exports = [
    {
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_bfwvfyx1'}],
        model: 'TS0505B',
        vendor: 'Generic Lighting',
        description: 'Generic RGB Bulb',
        toZigbee: extend.light_onoff_brightness_colortemp_color().toZigbee.concat([
            tz.tuya_do_not_disturb, tz.tuya_color_power_on_behavior,
        ]),
        meta: {applyRedFix: true, enhancedHue: false},
        fromZigbee: extend.light_onoff_brightness_colortemp_color().fromZigbee,
        exposes: [e.light_brightness_colortemp_colorhs([153, 500]),
                  exposes.binary('do_not_disturb', ea.STATE_SET, true, false).withDescription('Do not disturb mode'),
                  exposes.enum('color_power_on_behavior', ea.STATE_SET, ['initial', 'previous', 'cutomized']).withDescription('Power on behavior state'),
        ],
    },
   /*{
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_jd3z4yig'}],
        model: 'TS0505B',
        vendor: 'Generic Lighting',
        description: '18W RGB Bulb',
        meta: {applyRedFix: true, enhancedHue: false},
        extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    },*/
    {
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_jd3z4yig'}],
        model: 'TS0505B',
        vendor: 'Generic Lighting',
        description: 'Generic RGB Bulb',
        toZigbee: extend.light_onoff_brightness_colortemp_color().toZigbee.concat([
            tz.tuya_do_not_disturb, tz.tuya_color_power_on_behavior,
        ]),
        meta: {applyRedFix: true, enhancedHue: false},
        fromZigbee: extend.light_onoff_brightness_colortemp_color().fromZigbee,
        exposes: [e.light_brightness_colortemp_colorhs([153, 500]).removeFeature('color_temp_startup'),
                  exposes.binary('do_not_disturb', ea.STATE_SET, true, false).withDescription('Do not disturb mode'),
                  exposes.enum('color_power_on_behavior', ea.STATE_SET, ['initial', 'previous', 'cutomized']).withDescription('Power on behavior state'),
        ],
    },
];
```

closed 2023-03-02 by wbsantos