Koenkk/zigbee2mqtt #16871: Tuya _TZE200_lu01t0zl Presence state recognized as boolean but Occupancy recognized as string

raphaabreu, 2023-03-02:
### What happened?

This is a continuation of issues #16467 and #16568.

Both issues were closed but I believe that I might have not expressed myself well on the last one.

Starting on 1.30.0, the Tuya TS0601 _TZE200_lu01t0zl sensor stopped reporting its occupancy and presence status correctly.

In 1.30.1, the presence status was fixed but occupancy was not and this is what I tried to report in issue #16568.

When I look into the MQTT debug info, I can see that one status is being reported as a boolean and the other is being reported as a string:

[image]

In HomeAssistant, the occupancy sensor (which was being reported as string) always have the "Unknown" state:

[image]

This is what I see from the frontend:

[image]

There is also this subtle difference between how the states are shown on the Exposes tab, presence which is boolean is shown with a capital F and occupancy is shown all lowercase:

[image]

### What did you expect to happen?

I expected that both occupancy and presence status were reported as booleans and therefore recognized correctly by home assistant.

### How to reproduce it (minimal and precise)

I have not done any customizations to my install other than installing zigbee2mqtt and adding this sensor.

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20210708

### Adapter

sonoff zigbee 3.0

raphaabreu, 2023-03-02:
One more piece of info... After going back to the previous issues I have decided to try one thing:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_lu01t0zl', '_TZE200_vrfecyku']),
    model: 'MIR-HE200-TY',
    vendor: 'TuYa',
    description: 'Human presence sensor with fall function TESTE!',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await tuya.sendDataPointEnum(endpoint, tuya.dataPoints.trsfTumbleSwitch, false);
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
    },
    exposes: [
        e.illuminance_lux(), e.presence(), e.occupancy(),
        exposes.numeric('motion_speed', ea.STATE).withDescription('Speed of movement'),
        exposes.enum('motion_direction', ea.STATE, ['standing_still', 'moving_forward', 'moving_backward'])
            .withDescription('direction of movement from the point of view of the radar'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('Sensitivity of the radar'),
        exposes.enum('radar_scene', ea.STATE_SET, ['default', 'area', 'toilet', 'bedroom', 'parlour', 'office', 'hotel'])
            .withDescription('Presets for sensitivity for presence and movement'),
        exposes.enum('tumble_switch', ea.STATE_SET, ['ON', 'OFF']).withDescription('Tumble status switch'),
        exposes.numeric('fall_sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(10).withValueStep(1)
            .withDescription('Fall sensitivity of the radar'),
        exposes.numeric('tumble_alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(5).withValueStep(1)
            .withUnit('min').withDescription('Tumble alarm time'),
        exposes.enum('fall_down_status', ea.STATE, ['none', 'maybe_fall', 'fall'])
            .withDescription('Fall down status'),
        exposes.text('static_dwell_alarm', ea.STATE).withDescription('Static dwell alarm'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [2, 'radar_sensitivity', tuya.valueConverter.raw],
            [102, 'occupancy', tuya.valueConverter.trueFalse1],
            [103, 'illuminance_lux', tuya.valueConverter.raw],
            [105, 'tumble_switch', tuya.valueConverter.plus1],
            [106, 'tumble_alarm_time', tuya.valueConverter.raw],
            [112, 'radar_scene', tuya.valueConverterBasic.lookup(
                {'default': 0, 'area': 1, 'toilet': 2, 'bedroom': 3, 'parlour': 4, 'office': 5, 'hotel': 6})],
            [114, 'motion_direction', tuya.valueConverterBasic.lookup(
                {'standing_still': 0, 'moving_forward': 1, 'moving_backward': 2})],
            [115, 'motion_speed', tuya.valueConverter.raw],
            [116, 'fall_down_status', tuya.valueConverterBasic.lookup({'none': 0, 'maybe_fall': 1, 'fall': 2})],
            [117, 'static_dwell_alarm', tuya.valueConverter.raw],
            [118, 'fall_sensitivity', tuya.valueConverter.raw],
            // Below are ignored
            [101, null, null], // reset_flag_code
            [104, null, null], // detection_flag_code
            [107, null, null], // radar_check_end_code
            [108, null, null], // radar_check_start_code
            [109, null, null], // hw_version_code
            [110, null, null], // sw_version_code
            [111, null, null], // radar_id_code
        ],
    },
};

module.exports = definition;
```

This external converter was copied from https://github.com/Koenkk/zigbee2mqtt/issues/16467#issuecomment-1421309711 but I changed `[102, 'occupancy', tuya.valueConverter.trueFalse2],` to `[102, 'occupancy', tuya.valueConverter.trueFalse1],` and it is now working for me:

[image]

Koenkk, 2023-03-02:
Can you confirm the `occupancy` also goes to `true` correctly?

raphaabreu, 2023-03-03:
Yes, it goes to `true` and `false`. 

[image]

[image]

Koenkk, 2023-03-04:
Great, integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-04 by Koenkk