Koenkk/zigbee2mqtt #16872: Danfoss Ally (014G2461 - eTRV0103) not reporting state to Z2M, but works with ZHA

jhettler, 2023-03-02:
### What happened?

Hi everyone, 
I have  Danfoss Ally (014G2461 - eTRV0103) TRV connected to Conbee I/RPi4/Hass.
I used Deconz - I was NOT able to pair TRV, that's why I migrated to Z2M - all devices connected and works without any issues, except TRV.

I can successfully pair and connect TRV, I can read and set all parameters (checked by setting display orientation on TRV - 
orientation is changed successfully, same for temperature and other parameters), but the trouble is that TRV does NOT report values back to Z2M (it looks like reporting is not working correctly). 

I have tried 3 TRVs (same type) and the behavior is the same. Can pair, connect, set, read parameters, but it doesn't send set values back to Z2M. 

Everything works fine with ZHA, but I like Z2M ❤️  My idea was about changing coordinator as Conbee I is pretty old, but as it works with ZHA, it is not probably HW issue on coordinator.

I have read this issue https://github.com/Koenkk/zigbee2mqtt/issues/13478 few times, but it looks like a bit different issue and no tips from the discussion helped.
 
Thanks for any hints!

### What did you expect to happen?

Updated/changed attribute is reported back to Z2M/Hass.

### How to reproduce it (minimal and precise)

Pair TRV to Z2M - everything fine, interviewed successfully, configured successfully.
I can see reporting attributes in Z2M Reporting tab.
I can set and read values manually from Z2M Exposes tab, but it never gets updated automatically.
[image]

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

0x26400500

### Adapter

Conbee I

### Debug log

Nothing special, unfortunately as it is not reporting I can't see something in logs or I can't correctly setup logging.

lux73, 2023-03-02:
Can't confirm this. All my four Allys works as expected.

Two of them eTRV0101 and two eTRV0103

There must be another Issue in your Installation

tbarnekov, 2023-03-09:
I have the Danfoss Ally 014G2461 / eTRV0100 and I've noticed that they're very sensitive to 'spamming'. It's like messages are queued up in the TRV and responses are only sent at certain intervals. If you send many requests in a short period you may experience it going seemingly unresponsive, but in reality it's processing the requests very slowly.

The documentation states that it's able to process ~650 messages/day using the default check-in and poll intervals. I suspect that if you send it 650 messages in quick succession it will only start to respond to new messages after 24 hours.

This means it's a little tricky to work with when you're trying to figure out stuff like how to set a schedule correctly. It doesn't take many attempts for it to stop responding. It also requires some thought when setting up automations. Make sure you don't flood it with commands.

jhettler, 2023-03-09:
I have migrated the infrastructure to ZHA, but still I would like to use Z2M, I will try to pair another Danfoss Ally TRV with Z2M and play with it again.

closed 2023-04-17 by github-actions[bot]

TheLion, 2023-11-02:
I am noticing a similar issue with my Danfoss TRV's.
When pressing the refresh icon in Z2M for a certain value, it gets retrieved and published to MQTT (and in my case Domoticz) immediately. But it doesn't update at regular intervals on its own.

It only happens with all Danfoss TRV's, other devices connected to Z2M work without issues

I'm not sure when this started happening, as the TRV's didn't get used or changed until it got colder outside some time ago.