Koenkk/zigbee2mqtt #16873: Gledopto power_on_behavior doesn't work

estevez-dev, 2023-03-02:
### What happened?

Gledopto light bulbs [GL-B-008P](https://www.zigbee2mqtt.io/devices/GL-B-008P.html#gledopto-gl-b-008p) `power_on_behavior` doesn't work. After powering the bulb off and back on the property resets to `previous`.

### What did you expect to happen?

When `power_on_behavior` is set to `off` the bulb should stay off after a power outage.

### How to reproduce it (minimal and precise)

1. Open the "Exposes" tab of the Gledopto bulb
2. Set `power_on_behavior` to `off`
3. Cut the power to the bulb
4. Power on the bulb
5. Observe the result

### Zigbee2MQTT version

 1.30.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

Koenkk, 2023-03-02:
Do none of the `power_on_behavior` options work? I know that most Gledopto devices don't support `power_on_behavior` but I thought the `*P` (pro) ones did.

estevez-dev, 2023-03-02:
None of them is working. I was so excited to find those options but looks like even Pro versions don't have this feature.

Koenkk, 2023-03-02:
Thanks for checking, I've disabled it for all Gledopto devices.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-02 by Koenkk