Koenkk/zigbee2mqtt #16874: [New device support]: Tyua Zigbee Smart Switch 63A

like-a-freedom, 2023-03-02:
### Link

https://aliexpress.ru/item/1005005160237939.html?spm=a2g2w.orderdetail.0.0.39dc4aa6X7iamP&sku_id=12000031921768899

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xa4c138e66584813c","nwkAddr":24285,"manufId":4417,"manufName":"_TZE204_wbhaespm","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i�\u0000\u0000\u0000i","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_wbhaespm","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1677745937314,"defaultSendRequestWhen":"immediate"}

### Comments

Now it's unsupported.

Device type: Router
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_wbhaespm
Support status: Not supported
IEEE Address: 0xa4c138e66584813c
Network address: 0x5EDD

nordeep, 2023-03-15:
Any chance to add `tuya.fz.power_outage_memory` or `tuya.tz.power_on_behavior`?

like-a-freedom, 2023-03-15:
> Any chance to add `tuya.fz.power_outage_memory` or `tuya.tz.power_on_behavior`?

I didn't find appropriate datapoints ¯\_(ツ)_/¯

nordeep, 2023-03-16:
Do you have original Tuya ZigBee Gateway and Tuya App? Could you check any settings about `_TZE204_wbhaespm` switch behavior after loss of power.
Unfortunately, mine always OFF after a power reset.

like-a-freedom, 2023-03-16:
Do, I don't have Tuya gateway, only sonoff zigbee usb stick.

closed 2023-04-24 by github-actions[bot]