Koenkk/zigbee2mqtt #16875: QS-Zigbee-C01 and QS-Zigbee-C03 Problems in last versions

antunespa, 2023-03-02:
### What happened?

I reported this issue in last version. Curtain modules QS-Zigbee-C01 and QS-Zigbee-C03 after 1.30* updates dont report state correctly, when open or closing dont send percentage of the state and when stop close only show closing and after a long time refresh state to closed, the modules are not reporting the state correctly in version 1.29.2-1 everything works like expected, returned do previous version from backup. Its impossible for me to upgrade from 1.29.2-1 this version is stable in my devices and everything work without problems.

### Zigbee2MQTT version

1.30.*

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB

Koenkk, 2023-03-02:
I've checked but I don't see any changes related to the QS-Zigbee-C01 on 1.29.2 vs 1.30.2. Can you provide the debug log when the device reports the state correctly (on 1.29.2) and not correctly (on 1.30.2)?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

antunespa, 2023-03-02:
I send you the logs and video of my screen, thank you for the dedication.

[29.zip](https://github.com/Koenkk/zigbee2mqtt/files/10875380/29.zip)
[30.zip](https://github.com/Koenkk/zigbee2mqtt/files/10875382/30.zip)

https://user-images.githubusercontent.com/79264899/222541006-af9c1973-6df6-4afe-81ad-966d9a290e55.mp4

https://user-images.githubusercontent.com/79264899/222541027-511e1a17-9464-4a20-8622-a9b3b320baad.MP4

Koenkk, 2023-03-04:
I need the debug logging, see https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging (set `log_level: debug`)

antunespa, 2023-03-04:
[29.zip](https://github.com/Koenkk/zigbee2mqtt/files/10889025/29.zip)
[30.zip](https://github.com/Koenkk/zigbee2mqtt/files/10889026/30.zip)

Koenkk, 2023-03-04:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS130F', manufacturerName: '_TZ3000_vd43bbfq'}, {modelID: 'TS130F', manufacturerName: '_TZ3000_fccpjz5z'}],
    model: 'QS-Zigbee-C01',
    vendor: 'Lonsonho',
    description: 'Curtain/blind motor controller',
    fromZigbee: [fz.cover_position_tilt, fz.tuya_cover_options],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.tuya_cover_calibration, tz.tuya_cover_reversal],
    meta: {coverInverted: true},
    exposes: [e.cover_position(), //exposes.enum('moving', ea.STATE, ['UP', 'STOP', 'DOWN']),
        exposes.binary('calibration', ea.ALL, 'ON', 'OFF'), exposes.binary('motor_reversal', ea.ALL, 'ON', 'OFF'),
        exposes.numeric('calibration_time', ea.STATE).withUnit('S').withDescription('Calibration time')],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

antunespa, 2023-03-04:
Like this right?

[image]

[image]

Koenkk, 2023-03-05:
yes!

antunespa, 2023-03-05:
The problem persist, i send you the log. 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10892224/log.txt)

Koenkk, 2023-03-06:
Can you give me a more detailed explanation of the issue? Based on the log I don't understand what the exact problem is.

antunespa, 2023-03-09:
Hi,

Sorry for the delay, i will send you two videos, one in 29 and another em 30.

As you can see in the videos in version 29 there is better communication of the status, percentages are displayed whenever you lower or raise, I can stop and resume perfectly, with version 30 it does not show the percentage when lowering or raising, it does not always respond to the stop and when I stop and go back down or up even when the process ends, it continues to display the opening or closing information even when the process has ended, it seems that it does not communicate the status well to the home assistant.

https://user-images.githubusercontent.com/79264899/224142049-74a41d0b-2e5c-4f26-92a9-2e0069d55f38.mp4
https://user-images.githubusercontent.com/79264899/224142076-de9cea05-6c5c-4af6-b665-d2b6d9638253.mp4

Koenkk, 2023-03-11:
I think this is caused by https://github.com/Koenkk/zigbee2mqtt/pull/16324 or https://github.com/Koenkk/zigbee2mqtt/pull/16614. @Drafteed this device has a `running` sensor, could you look into this?

Drafteed, 2023-03-12:
@antunespa

Hello!

In the previous version, your curtains didn't display the moving status, they just updated the progress status as a percentage. In this case, the status of the entity was set as stopped, even if they were moving.

In the new version, you have the **correct behavior**. Your curtain entity now knows that they are moving (state becomes to opening/closing). 

The percentage status in new version is not displayed only because of an bug/feature of the HA frontend `Tile Card` in case when state of curtains is opening/closing. 

Here is the place in the frontend repository: https://github.com/home-assistant/frontend/blob/69264b3448e9862025e78090a52af20f796aa32b/src/panels/lovelace/cards/hui-tile-card.ts#L227-L236
As you can see, the percentages are displayed only when the state of the curtains is `open`, and not `opening`, `closing`, `closed`.

You can [create an issue in their repository](https://github.com/home-assistant/frontend/issues/new/choose) to resolve your issue.

Drafteed, 2023-03-12:
> not always respond to the stop and when I stop and go back down or up even when the process ends, it continues to display the opening or closing information even when the process has ended, it seems that it does not communicate the status well to the home assistant.

About this. We need to find out whether your `moving` sensor returns correct state when you send `stop` command.

antunespa, 2023-03-13:
@Drafteed I opened a issue and they merged with another one https://github.com/home-assistant/frontend/pull/15809 i dont understand the process but is running.

Drafteed, 2023-03-14:
@antunespa Great. Changes will appear in the next release of Home Assistant (2023.4.0).

closed 2023-04-21 by github-actions[bot]