Koenkk/zigbee2mqtt #16876: Thread and zigbee support for sonoff zigbee dongle P

Sarnog, 2023-03-02:
### Is your feature request related to a problem? Please describe

no

### Describe the solution you'd like

Support for both zigbee and thread (matter) at the same time, like the SkyConnect.

### Describe alternatives you've considered

I'm waiting for this first, since it should be possible. Otherwise i need to buy another stick, maybe the skyconnect.

### Additional context

found some documentation:

https://www.silabs.com/documents/public/application-notes/an1256-using-sl-rcp-with-openthread-border-router.pdf

https://openthread.io/vendors/texas-instruments#cc2652

Koenkk, 2023-03-02:
This is out of scope for the Zigbee2MQTT project

closed 2023-03-02 by Koenkk

Sarnog, 2023-03-02:
Hoi Koen,

Ik denk dat je Nederlands bent, gezien je achternaam. 😊
Aangezien je ook de firmware's maakt (denk ik) voor de diverse zigbee
sticks, was ik benieuwd of je hier zicht op had? Of dat dit in de (nabije)
toekomst mogelijk wordt.

Bedankt voor je reactie!

Met vriendelijke groet,

Danny.