Koenkk/zigbee2mqtt #16877: Philips Hue dimmer - action_duration null

JCHH2, 2023-03-02:
### What happened?

I'm not sure which is possibly causing this, but I recently updated:
Z2M ➜ 1.30.2
HA ➜ 2023.3.0

I am now getting all my Hue dimmer switches `action_duration' reporting `unknown` in HA - example below:
[image]

The switches work fine and show OK for everything else.
[image]

...all dimmers are reporting `null` in Z2M - example below:
[image]

### What did you expect to happen?

I didn't expect a permanent state of `unknown` in HA or `null` in Z2M.

I have pressed the buttons to see idk it was only a start-up error.
I also restarted HA in case it was a 1-off upgrade error.

### How to reproduce it (minimal and precise)

It's a permanent status (for me) for:
Z2M ➜ 1.30.2
HA ➜ 2023.3.0

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

2023.3.0

### Adapter

Sonoff 3.0 USB stick ZBDongle-P / zStack3x0 / v20221226

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10870005/log.txt)

hellcry37, 2023-03-09:
I think this is solved in latest HA

JCHH2, 2023-03-09:
Still 'unknown' for me on Core 2023.3.2

JCHH2, 2023-03-16:
I see there is a 'duration' that is correct in Z2M but is not exposed.  Has there simply been a change form `action_duration` to `duration`?

```
{
    "action": "",
    "action_duration": null,
    "battery": 100,
    "brightness": 255,
    "counter": 1,
    "linkquality": 185,
    "update": {
        "installed_version": 1107324829,
        "latest_version": 1107324829,
        "state": "idle"
    },
    "update_available": false,
    "duration": 2.089
}
```

closed 2023-04-23 by github-actions[bot]