Koenkk/zigbee2mqtt #16878: [New device support]:

bam27beer, 2023-03-02:
### Link

https://www.zemismart.com/products/zm25-zb

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0x847127fffe1b711d","nwkAddr":38753,"manufId":4098,"manufName":"_TZE200_fzo2pocs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_fzo2pocs","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1677750770669}

### Comments

i have 5 x zemismart zigbee blind motors ,
i can use them via deconz ,
but want to try and use in 1 solution ,
model  looks like ZM25-ZB,
connected as 
[image]
[image]

TuYa | TS0601_cover, 
the weird thing is although i can control via the z2m dashboard ,its only exposing the link quality to mosquito

Koenkk, 2023-03-02:
You are using an outdated z2m version.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2023-03-02 by Koenkk

bam27beer, 2023-03-13:
cheers will check but could'nt find my own thread doh ,
using the home assistant add on and not showing any updates ,
just seen the new repo , do i uninstall the old add on before running the new seeing as its 1.18- v1.30+?