Koenkk/zigbee2mqtt #16879: HEIMAN SGMHM-I1 gas sensor is NOT working.

dmg244, 2023-03-02:
### What happened?

1. First of all HEIMAN SGMHM-I1 gas sensor is not battery device, but have battery_low  and tamper sensors.

2. Second gas detection is not working, gas status false every time.

**Log with start gas alarm is:**
```
debug 2023-03-02 13:15:51Received Zigbee message from 'GasSensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0
info 2023-03-02 13:15:51MQTT publish: topic 'zigbee2mqtt/GasSensor', payload '{"battery_low":false,"gas":false,"linkquality":156,"tamper":false}'
```

**Log with stop gas alarm is:**
```
debug 2023-03-02 13:15:53Received Zigbee message from 'GasSensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
info 2023-03-02 13:15:53MQTT publish: topic 'zigbee2mqtt/GasSensor', payload '{"battery_low":false,"gas":false,"linkquality":176,"tamper":false}'
```

### What did you expect to happen?

1. Remove tamper and battery_low sensors.
2. Correct description from "Combustible gas sensor" to "Methane gas sensor".  (Like HEIMAN SGPHM-I1 is "Propane gas sensor")
3. Correct code to work gas sensor.

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Debug log

debug 2023-03-02 13:15:51Received Zigbee message from 'GasSensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0
info 2023-03-02 13:15:51MQTT publish: topic 'zigbee2mqtt/GasSensor', payload '{"battery_low":false,"gas":false,"linkquality":156,"tamper":false}'

debug 2023-03-02 13:15:53Received Zigbee message from 'GasSensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
info 2023-03-02 13:15:53MQTT publish: topic 'zigbee2mqtt/GasSensor', payload '{"battery_low":false,"gas":false,"linkquality":176,"tamper":false}'

xtaska, 2023-03-15:
I have the same problem. The gas sensor is false all the time. When I press the test button, the sensor siren turns on, but the state of the gas sensor does not change!

Zigbee2MQTT version
1.30.2 commit: unknown
Coordinator type
zStack3x0
Coordinator revision
20210218

[image]

dmg244, 2023-03-15:
I'm use Zigbee Home Automation. All working fine. Also lamp BRHM8E27W70-I1 is not working in zigbee2mqtt but work in Zigbee Home Automation.

Koenkk, 2023-03-15:
Can you provide the data/database.db entry of this device?

dmg244, 2023-03-15:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0xe0798dfffeb1ce6a","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1677795409147,"defaultSendRequestWhen":"immediate"}
{"id":2,"type":"Router","ieeeAddr":"0x00158d00025e52b8","nwkAddr":32556,"manufId":4619,"manufName":"GS","powerSource":"Mains (single phase)","modelId":"BRHM8E27W70-I1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,4,3,6,8,5,768,4096,65533],"outClusterList":[25,65533],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorMode":1,"currentX":21168,"currentY":21561,"enhancedCurrentHue":35068,"currentSaturation":16}},"genBasic":{"attributes":{"manufacturerName":"GS","hwVersion":16,"powerSource":1,"swBuildId":"2000-1002"}},"genScenes":{"attributes":{"count":1,"currentScene":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":10,"stackVersion":2,"hwVersion":16,"dateCode":"20180628","swBuildId":"2000-1002","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1677801008260,"defaultSendRequestWhen":"immediate"}
{"id":3,"type":"Router","ieeeAddr":"0x000d6f0010dd1218","nwkAddr":2131,"manufId":4619,"manufName":"GS","powerSource":"Mains (single phase)","modelId":"SGMHM-I1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,9,1280,65523],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"hwVersion":16,"appVersion":22,"manufacturerName":"GS","modelId":"SGMHM-I1","powerSource":1,"dateCode":"2018.3.26","zclVersion":1}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffeb1ce6a","zoneState":1}},"genOnOff":{"attributes":{}},"genAlarms":{"attributes":{}}},"binds":[{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb1ce6a","endpointID":1},{"cluster":9,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb1ce6a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":22,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.26","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1677800610241,"defaultSendRequestWhen":"immediate"}
{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c1385728c92212","nwkAddr":16461,"manufId":4417,"manufName":"_TZ3000_qewo8dlz","powerSource":"Mains (single phase)","modelId":"TS0013","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�S�+\u0013�S�+\u0012","65506":55,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_qewo8dlz","zclVersion":3,"appVersion":80,"modelId":"TS0013","powerSource":3}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":2,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb1ce6a","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb1ce6a","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb1ce6a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1331047771},"lastSeen":1677801036827,"defaultSendRequestWhen":"immediate"}
```

dmg244, 2023-03-15:
For lamp BRHM8E27W70-I1 need add color temperature options. Because now only RGB mode support and impossible to switch lamp to "white light" mode.

Koenkk, 2023-03-16:
@OXOTH1K  since you added the SGMHM-I1, can you make sure it detects gas properly with z2m 1.30.2?

@dmg244 @xtaska 
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SGMHM-I1'],
    model: 'SGMHM-I1',
    vendor: 'HEIMAN',
    description: 'Combustible gas sensor',
    fromZigbee: [fz.ias_gas_alarm_2],
    toZigbee: [],
    exposes: [e.gas(), e.battery_low(), e.tamper()],
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed
-

@dmg244 regarding BRHM8E27W70, 
Could you check if the issue is fixed with the following external converter: 
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['BRHM8E27W70-I1'],
    model: 'BRHM8E27W70-I1',
    vendor: 'GS',
    description: 'Smart dimmable, RGB + white (E27 & B22)',
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

dmg244, 2023-03-16:
Yes both working now, but gas sensor don't have battery and tamper. It's AC powered device, don't have battery.

xtaska, 2023-03-16:
@Koenkk with ext_converter.js the issue is fixed.
Now when I press the test button, the gas alarm goes on. Thank you!

Koenkk, 2023-03-18:
Integrated the fixed and remove tamper and battery. Thanks!

You can keep using the external converter until the next release.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

dmg244, 2023-03-19:
Thank you! If possible correct description from "Combustible gas sensor" to "Methane gas sensor". (Like HEIMAN SGPHM-I1 have name "Propane gas sensor")

closed 2023-03-20 by Koenkk