Koenkk/zigbee2mqtt #16884: Brightness set to 1 on turn off

hamshiva, 2023-03-02:
### What happened?

Hi,

I am running v1.30.1. 
Turning on and off the light from z2m is working correctly but if I turn off the light (brightness previously set to 255) with z2m and turn it back on using a binded remote then the light is turned on with brigthness set to 1. 

And looking at the mqtt messages I can see that the brightness is set to 1 when the light is turned off with z2m but it stays set to 255 when turned off using the binded remote.

Turn off from z2m:
```
Mar 02 18:45:57 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:45:57: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":254,"last_seen":"2023-03-02T18:45:57.009Z","linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
Mar 02 18:45:57 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:45:57: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":254,"last_seen":"2023-03-02T18:45:57.009Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
Mar 02 18:45:57 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:45:57: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":1,"last_seen":"2023-03-02T18:45:57.786Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
Mar 02 18:45:57 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:45:57: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":1,"last_seen":"2023-03-02T18:45:57.799Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
```

Turn off from binded remote:
```
Mar 02 18:47:24 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:47:24: MQTT publish: topic 'zigbee2mqtt/WC dimmer switch', payload '{"action":"off","battery":null,"last_seen":"2023-03-02T18:47:24.872Z","linkquality":255,"update":{"installed_version":587699761,"latest_version":587699761,"state":"idle"},"update_available":false}'
Mar 02 18:47:24 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:47:24: MQTT publish: topic 'zigbee2mqtt/WC dimmer switch', payload '{"action":"","battery":null,"last_seen":"2023-03-02T18:47:24.872Z","linkquality":255,"update":{"installed_version":587699761,"latest_version":587699761,"state":"idle"},"update_available":false}'
Mar 02 18:47:24 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:47:24: MQTT publish: topic 'zigbee2mqtt/WC dimmer switch/action', payload 'off'
Mar 02 18:47:25 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:47:25: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":254,"last_seen":"2023-03-02T18:47:25.260Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
Mar 02 18:47:25 storez node[425]: Zigbee2MQTT:info  2023-03-02 18:47:25: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":254,"last_seen":"2023-03-02T18:47:25.914Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
```

I am wondering if z2m is not playing with the brightness when turned off, maybe for transition, but transition is set to 0 in my config.

Regards,

### What did you expect to happen?

When I turn the light on with the binded remote the brightness setting should be the same as before the light was turned off.

### How to reproduce it (minimal and precise)

Turn off the light from z2m interface (and see that the brightness goes to 1) then turn it on using a remote binded to the light.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

0x26660700

### Adapter

Conbee2

hamshiva, 2023-04-03:
Hi,
I am now running 1.30.2 and the issue is still there.
Regards,

sjorge, 2023-04-03:
I think this is by design due to the light having `turnsOffAtBrightness1` set ( see the list of meta options in de README https://github.com/Koenkk/zigbee-herdsman-converters)

When using a report it's the light that report that it's off and z2m updates it accordingly. If this is the case it should be visible in a debug log.

hamshiva, 2023-04-06:
Then why is the behavior different when the light is turned off using a binded switch ?
I mean when turned off with the binded switch the brightness value remains what it was before turning off and still the state is correctly updated to off. 

And btw 
> turnsOffAtBrightness1: Indicates light turns off when brightness 1 is set (default: false)

It doesn't says that the brightness is set to 1 when the light is turned off...

Did I missed something ?

sjorge, 2023-04-06:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/1e7a9807dfa3e70115c235560d8740654afad4bc/converters/toZigbee.js#L866-L1003

There are a bunch extra branch condition when this is set were it does stuff with the current cached brightness value too. A debug log is probably helpful to see if somehow the bulb still sends a brightness update or if it's purely z2m doing stuff.

hamshiva, 2023-04-06:
There are some logs:
[Brigthness_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/11172448/Brigthness_debug.log)

Let me know if you need something else.

sjorge, 2023-04-06:
Lets break this down, using a physical switch we get the following from the light itself:

1. report that it is on (?)
```
Apr 06 16:06:22 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:22: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID null
```
2. report that it is/was at ~42% brightness
```
Apr 06 16:06:22 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:22: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":107}' from endpoint 11 with groupID null
```

3. something sends a request to turn it (back) on using MQTT at full brightness (was this intentional?)
```
Apr 06 16:06:22 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:22: Received MQTT message on 'zigbee2mqtt/WC Light/set' with data '{"state":"ON","brightness":254}'
```

4. now it gets a bit confusing, something read the genOnOff.onOff state and the light replies that it is on (not sure what the origin is, doesn't look like it was mqtt
```
Apr 06 16:06:23 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:23: Received Zigbee message from 'WC Light', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID null
```

5. the light now reports that it is off (finally)
```
Apr 06 16:06:26 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:26: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
```

6. something triggers a read again for genOnOff.onOff and the light replies that it is indeed off.
```
Apr 06 16:06:27 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:27: Received Zigbee message from 'WC Light', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
```

7. the light reports it's now brightness is now at 100% (perhaps due to the /set form 3.)
```
Apr 06 16:06:27 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:06:27: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 11 with groupID null
```

@Koenkk any idea what could be triggering those reads? They do not seem to be mqtt initiated.

When turning it off with via mqtt:
1. it gets turned on twice, once with just state and once with state + brightness (100%)
```
Apr 06 16:07:09 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:07:09: Received MQTT message on 'zigbee2mqtt/WC Light/set' with data '{"state":"ON"}'
Apr 06 16:07:09 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:07:09: Received MQTT message on 'zigbee2mqtt/WC Light/set' with data '{"state":"ON","brightness":254}'
```

There are 2 state publishes in between, once with state already on (I think because opportunistic publishing is enabled) and before one with state still off which I don't understand why it gets pushed. @koenkk perhaps because last_seen is enabled? Would this cause an unchanged publish when it sees a `/set` for a device + an opportunistic one right after with the new state?

2. The light reports that it is now indeed on
```
Apr 06 16:07:09 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:07:09: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID null
Apr 06 16:07:09 storez node[391]: Zigbee2MQTT:info  2023-04-06 16:07:09: MQTT publish: topic 'zigbee2mqtt/WC Light', payload '{"brightness":254,"last_seen":"2023-04-06T16:07:09.567Z","linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16786432,"latest_version":16786432,"state":"idle"},"update_available":false}'
```

3. turned off via mqtt/frontend
```
Apr 06 16:07:15 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:07:15: Received MQTT message on 'zigbee2mqtt/WC Light/set' with data '{"state":"OFF"}'
```

4. The light returns brightness level of 1% + that it is off.
```
Apr 06 16:07:15 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:07:15: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 11 with groupID null
Apr 06 16:07:15 storez node[391]: Zigbee2MQTT:debug 2023-04-06 16:07:15: Received Zigbee message from 'WC Light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID null
```

@Koenkk I'm guessing this is because the switch probably sends genOnOff.onOff and the converter sends a levelMoveWithOnOff and we move it to 1 to turn it off instead ?

That seems to be about the only reason I can think off why the reported brightness is different.

hamshiva, 2023-05-07:
Hi,
I am now running 1.30.4 and the issue still exists.
Regards,

hamshiva, 2023-06-07:
Hi,
The issue is still there.
Regards,

hamshiva, 2023-07-13:
Hi,
I am now running 1.32.0 and the issue still exists.
Regards,

hamshiva, 2023-08-07:
Hi,
I am now running 1.32.2 and the issue still exists.
Regards,

hamshiva, 2023-09-13:
Hi,
I am now running 1.33.0 and the issue still exists.
Regards,

hamshiva, 2023-10-18:
Hi,
I am now running 1.33.1 and the issue still exists.
Regards,

hamshiva, 2023-11-04:
@Koenkk I never though about removing completly the value from the transition field. But it looks like it is working.

Many thanks !

closed 2023-11-04 by hamshiva

JoshSald, 2024-01-10:
I've been having these issues for a while with 2 separate lights with past versions where they turn on with a very low brightness level. I resolved the issue by re-pairing these lights with Z2M, with the hope that any visitors we have don't actually switch off power to the light otherwise I'd have to re-pair again. It wasn't the most ideal solution. I've just updated to **1.35.1** yesterday evening and 4 more lights have now been afflicted with this issue as well. 

Some points to note:
- No power was cut to any of these newly affected lights.
- All of the affected bulbs are Ikea Tradfri. 
- I use [Adaptive Lighting](https://github.com/basnijholt/adaptive-lighting) with some of them. 
- Transition has been removed on all of these lights already. 

Oddly enough, some of the lights claims to be at 254 brightness, but the actual visible brightness of the light is definitely around the 1-3% mark. The only fix I have right now is to lower the brightness by 1% and it works as it should. Here's an example of one of those lights (Adaptive Lighting is **INACTIVE** or rather never used on this light):

```
Debug 2024-01-10 09:36:47Received MQTT message on 'zigbee2mqtt/Kitchen Cabinet Lights/set' with data '{"state":"ON"}'
Debug 2024-01-10 09:36:47Publishing 'set' 'state' to 'Kitchen Cabinet Lights'
Info 2024-01-10 09:36:47MQTT publish: topic 'zigbee2mqtt/Kitchen Cabinet Lights', payload '{"brightness":203,"level_config":{"on_level":"previous"},"linkquality":36,"power_on_behavior":"previous","state":"ON","update":{"installed_version":65538,"latest_version":65538,"state":"idle"},"update_available":false}'
```

Here's the log of me turning on another light, adjusting its brightness, turning it off and turning back on again (Adaptive Lighting is **ACTIVE** on this light):

```
Debug 2024-01-10 09:27:27Received MQTT message on 'zigbee2mqtt/Living Room Bookshelf Lamp/set' with data '{"state":"ON"}'
Debug 2024-01-10 09:27:27Publishing 'set' 'state' to 'Living Room Bookshelf Lamp'
Info 2024-01-10 09:27:27MQTT publish: topic 'zigbee2mqtt/Living Room Bookshelf Lamp', payload '{"brightness":3,"color_mode":"color_temp","color_options":null,"color_temp":454,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":null,"state":"ON","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":false}'
Debug 2024-01-10 09:27:28Received MQTT message on 'zigbee2mqtt/Living Room Bookshelf Lamp/set' with data '{"state":"ON","brightness":3,"color_temp":454}'
Debug 2024-01-10 09:27:28Publishing 'set' 'brightness' to 'Living Room Bookshelf Lamp'
Debug 2024-01-10 09:27:28Publishing 'set' 'color_temp' to 'Living Room Bookshelf Lamp'
Info 2024-01-10 09:27:28MQTT publish: topic 'zigbee2mqtt/Living Room Bookshelf Lamp', payload '{"brightness":3,"color_mode":"color_temp","color_options":null,"color_temp":454,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":null,"state":"ON","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":false}'
Debug 2024-01-10 09:28:13Received MQTT message on 'zigbee2mqtt/Living Room Bookshelf Lamp/set' with data '{"state":"ON","brightness":254}'
Debug 2024-01-10 09:28:13Publishing 'set' 'brightness' to 'Living Room Bookshelf Lamp'
Info 2024-01-10 09:28:13MQTT publish: topic 'zigbee2mqtt/Living Room Bookshelf Lamp', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":454,"level_config":{"on_level":"previous"},"linkquality":69,"power_on_behavior":null,"state":"ON","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":false}'
Debug 2024-01-10 09:28:18Received MQTT message on 'zigbee2mqtt/Living Room Bookshelf Lamp/set' with data '{"state":"OFF"}'
Debug 2024-01-10 09:28:18Publishing 'set' 'state' to 'Living Room Bookshelf Lamp'
Info 2024-01-10 09:28:18MQTT publish: topic 'zigbee2mqtt/Living Room Bookshelf Lamp', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":454,"level_config":{"on_level":"previous"},"linkquality":69,"power_on_behavior":null,"state":"OFF","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":false}'
Debug 2024-01-10 09:28:22Received MQTT message on 'zigbee2mqtt/Living Room Bookshelf Lamp/set' with data '{"state":"ON"}'
Debug 2024-01-10 09:28:22Publishing 'set' 'state' to 'Living Room Bookshelf Lamp'
Info 2024-01-10 09:28:22MQTT publish: topic 'zigbee2mqtt/Living Room Bookshelf Lamp', payload '{"brightness":254,"color_mode":"color_temp","color_options":null,"color_temp":454,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":null,"state":"ON","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":false}'
Debug 2024-01-10 09:28:22Received MQTT message on 'zigbee2mqtt/Living Room Bookshelf Lamp/set' with data '{"state":"ON","brightness":3,"color_temp":454}'
Debug 2024-01-10 09:28:22Publishing 'set' 'brightness' to 'Living Room Bookshelf Lamp'
Debug 2024-01-10 09:28:22Publishing 'set' 'color_temp' to 'Living Room Bookshelf Lamp'
Info 2024-01-10 09:28:22MQTT publish: topic 'zigbee2mqtt/Living Room Bookshelf Lamp', payload '{"brightness":3,"color_mode":"color_temp","color_options":null,"color_temp":454,"level_config":{"on_level":"previous"},"linkquality":65,"power_on_behavior":null,"state":"ON","update":{"installed_version":65554,"latest_version":65554,"state":"idle"},"update_available":false}'
```
For reference, the other unaffected lights using Adaptive Lighting are at 100% brightness at the time I was testing this light.

As mentioned in the beginning, I could re-pair and fix all of these issues, but I was hoping there would be a far more effective solution than that.

amaisano, 2025-09-07:
Same issue as above on the current stable version of Z2M and latest firmware for my IKEA GU10 bulbs. RE-pairing them worked. It seems the level config gets "stuck" and cannot be set to anything but on level of "1."