Koenkk/zigbee2mqtt #16886: yrd226 & yrd-zbm-1

kuradi, 2023-03-02:
### What happened?

i have yale yrd226 lock and zigbee module yrd-zbm-1.
if i look defices list at https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/yale.js looks like my lock is supported, but i cant get any info from it.

db info: {"device":{"applicationVersion":255,"friendlyName":"ukselukk_maja","hardwareVersion":18,"ieeeAddr":"0x000d6f0018ade0dd","manufacturerID":4125,"manufacturerName":"Yale","model":"unknown","networkAddress":45056,"powerSource":"Battery","type":"EndDevice","zclVersion":1},"last_seen":"2023-03-02T19:17:58+02:00"}

and logs:
debug 2023-03-02 21:16:46: Device 'ukselukk_maja' announced itself
info  2023-03-02 21:16:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"ukselukk_maja","ieee_address":"0x000d6f0018ade0dd"},"type":"device_announce"}'
info  2023-03-02 21:16:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"ukselukk_maja"},"type":"device_announced"}'
debug 2023-03-02 21:16:48: Received Zigbee message from 'ukselukk_maja', type 'read', cluster 'genTime', data '["timeStatus"]' from endpoint 1 with groupID 0
warn  2023-03-02 21:16:48: Received message from unsupported device with Zigbee model 'NO LOCK PRESENT' and manufacturer name 'Yale'
warn  2023-03-02 21:16:48: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

so... help please.

### Zigbee2MQTT version

1.30.0

### Adapter firmware version

20210120

### Adapter

zStack3x0

kuradi, 2023-03-04:
also i can send sniff results

Koenkk, 2023-03-04:
`with Zigbee model 'NO LOCK PRESENT'` this seems to be some issue of the device itself. What you can try to insert a lock (since it states none is present?) and force remove the device through z2m frontend, restart z2m and re-pair it.

kuradi, 2023-03-04:
> `with Zigbee model 'NO LOCK PRESENT'` this seems to be some issue of the device itself. What you can try to insert a lock (since it states none is present?) and force remove the device through z2m frontend, restart z2m and re-pair it.

thanx. did it. same:
[image]

is mine similar case as https://github.com/Koenkk/zigbee2mqtt/issues/4019 ?

kuradi, 2023-03-05:
we may close this issue. seems this zigbee module is not compatible with lock.
[image]
p.s. buying first gen module! :)

closed 2023-03-05 by Koenkk