Koenkk/zigbee2mqtt #16892: Danfoss Ally (014G2461) & Popp (701721) OTA Update failed: "Aborted by device"

suaveolent, 2023-03-03:
### What happened?

Zigbe2mqtt says there is a new firmware for my Danfoss Ally and Popp thermostats available.
All my devices are running "01.18.0008 01.18". 
The device update runs fine, but is eventually aborted with the message:

`Update of '<device_name>' failed (Update failed with reason: 'aborted by device')`.

### What did you expect to happen?

The update should be installed successfully.

### How to reproduce it (minimal and precise)

Switch to OTA tab, click on "Check for new updates" (if update is not available), then click on "Update device firmware".

### Zigbee2MQTT version

1.30.2 commit: cdf62ea

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

lux73, 2023-03-03:
afaik there is no newer Firmware for Danfoss Ally available.

The x.20 Firmware is only for the newer eTRV0103 Allys because they have different Microcontroller.

No new Options in the x.20 FW

suaveolent, 2023-03-03:
Yes, I just found this thread:

https://github.com/Koenkk/zigbee-OTA/issues/264

Either way, the update should not be presented if it is not suitable for the device.

steff75, 2023-03-03:
The update was successful for my old Ally.
[image]

suaveolent, 2023-03-03:
Does your device have a "G" in front of the serial number?

steff75, 2023-03-03:
Where can I find the number?

steff75, 2023-03-03:
[image]

suaveolent, 2023-03-03:
Interesting. So you are indeed running the old version. 
Did you update from `01.18.0008 01.18`?

steff75, 2023-03-03:
Yes, I was also unsure at first whether I should risk the update.

lux73, 2023-03-03:
@steff75 

Thats really strange - your updated FW Date is from 2020...

Weird

steff75, 2023-03-03:
I don't know, why that is.
But the update was the one, that Z2M offered me. 
The process took about ~80 minutes.

suaveolent, 2023-03-03:
In general, there seems to be a problem with the Firmware built date. My devices are showing different built dates for the same firmware:

[image]

gecko17, 2023-03-03:
So we are in agreement, that the old Allies running 01.18 (and don't have the `G` in the serial number) shouldn't be updated to the new firmware, as it was only made for the new microcontroller. I think this is the reason Danfoss used a new `imageType` 288 in the 01.20 image.

If I'm correct, I think there should be a fix in the `index.json` by adding a new entry for the new 01.20 image with the new image type and revert the current Danfoss entry to its previous state, pointing to the 01.18 image?

So https://github.com/Koenkk/zigbee-OTA/issues/264 should be reopened to make those changes.

steff75, 2023-03-03:
It doesn't seem that simple, does it? 
I bought mine in November 2020, when the new microcontroller was not yet available. Nevertheless, I was able to update.

stefangehn, 2023-03-03:
> In general, there seems to be a problem with the Firmware built date. My devices are showing different built dates for the same firmware:

The incorrect firmware date/version has been reported before: #15649
Somehow this happens quite often with Danfoss Ally, removing the device, restarting z2m and then joining the device will at least fix that (seems to be a caching/display error).

steff75, 2023-03-03:
In case the log contains any information of interest, I have attached the part of the Danfoss Ally update here:

[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/10884899/log1.txt)

gedemagt, 2023-03-06:
I have the same issue. I downloaded the firmware from the official website, which also passes a .pdf along which describes the new firmware and that it belongs to imageType 288 (0x120). But when I try to flash it, it reports back an error saying "imageType mismatch".

Any ideas?

I downloaded the most recent from here: https://www.danfoss.com/en/products/dhs/smart-heating/smart-heating/danfoss-ally/danfoss-ally-support/#tab-software

gecko17, 2023-03-06:
> It doesn't seem that simple, does it? I bought mine in November 2020, when the new microcontroller was not yet available. Nevertheless, I was able to update.

I guess it is that simple... But of course, I don't **know** it for sure. It would make sense that they changed the imageType, so gateways like Z2M are not trying to push the update to the device, which is why I think the change made to the `index.json` was faulty. 

Just that it was possible to update the old devices to the new firmware for some devices doesn't mean it is supposed to work and the Danfoss documentation file that comes with the image clearly states that it should only be applied to the new micro controller devices IMHO...

steff75, 2023-03-07:
> documentation file that comes with the image clearly states that it should only be applied to the new micro controller devices

Basically, the PDF does not say that the OTA file is only for the new Allys. As the core statement of the document I see actually that the partners/third party manufacturers must make adjustments so that the eTRV0103 is also supported by their hardware.

But don't get me wrong, possibly the OTA file is really only for the new version. All I know right now is that my device is still working fine.

Greetings

gedemagt, 2023-03-07:
> Basically, the PDF does not say that the OTA file is only for the new Allys. As the core statement of the document I see actually that the partners/third party manufacturers must make adjustments so [...]

That is true. I am however still facing a imageType mismatch when I try to OTA update it. Wonder where that check is made and if i can force it somehow.

closed 2023-04-14 by github-actions[bot]

Masoholik, 2023-04-27:
> That is true. I am however still facing a imageType mismatch when I try to OTA update it. Wonder where that check is made and if i can force it somehow.

could you solve this somehow?

i get an "no image avaiable for image type 288" error. When i use an local index.json file and change "imageType" to 288 and copy the ota file to my server. then i also get "image type mismatch"

my OTA overview in z2m is showing me also a strange firmware version:

[image]