Koenkk/zigbee2mqtt #16894: [New device support]: LED2101G4

beliall2442, 2023-03-03:
### Link

https://www.ikea.com/fr/fr/p/tradfri-ampoule-a-led-e14-470-lumen-connecte-sans-fil-a-variateur-dintensite-spectre-blanc-globe-70518179/

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x1c34f1fffee20d9a","nwkAddr":17163,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 WS globe 470lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":17,"stackVersion":106,"hwVersion":1,"dateCode":"20211201","swBuildId":"1.1.003","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1677832806884,"defaultSendRequestWhen":"immediate"}

### Comments

The device is in the supported devices list but appear as unsupported in zigbee2mqtt 
Didn't find any issue for this devices, same in dev branch
Hope i don't miss something easy and make you Lost your Time :/

### Supported color modes

Color temp

### Color temperature range

{"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}

beliall2442, 2023-03-04:
I just find this model in the ikea.js
But for some reasons still list as Unsupported
Tried to add by myself but didn't figure how setup the external converter correctly

Regards

Koenkk, 2023-03-04:
Probably you are using an outdated z2m verison.

beliall2442, 2023-03-05:
Yes i'm on 1.27
Thx

beliall2442, 2023-03-05:
Yes i'm on 1.27
Thx

closed 2023-03-05 by beliall2442