Koenkk/zigbee2mqtt #16898: No power value for Legrand power socket 067775/741811

aurelien219, 2023-03-03:
### What happened?

I've no information about power and power_apparent for my 2 Legrand power socket 067775/741811

[image]

Maybe it's the same problem than the issue #16732. 

I tried to use the ext_converter.js but I've a problem of syntax file when I copy/paste the file and Zigbee2mqtt won't start.

### What did you expect to happen?

No value for power and power_apparent.

### Zigbee2MQTT version

1.30.1

### Adapter firmware version

20191106

### Adapter

Conbee 2

aurelien219, 2023-03-09:
Yes, I updated the power socket with the Legrand gateway.

LeorFinacre, 2023-03-12:
Hello,
I'm having the same product and trying to achieve the same goal : being able to read power consumption.
I also tried the following JS code but zigbee2mqtt does not restart.
It fails with this error : 
typeerror cannot read properties of undefined (reading 'options')

Maybe it's the same thing as aurelien219 ?
My product firmware is out of the box. I don't have the Legrand Gateway to update the firmware.

aurelien219, 2023-03-12:
I upgrade zigbee2mqtt to 1.30.2 yesterday (1.30.1 before) and I've the power value now. Which version are you using?

LeorFinacre, 2023-03-12:
According to the frontend application, I am in the 1.30.2 version too.
[image]

I don't have power consomption from this page 

[image]

LeorFinacre, 2023-03-12:
> Yes, I updated the power socket with the Legrand gateway.

Can you describe how you managed to update the firmware ?
I don't have the Legrand Gateway but maybe there is another way to update it's firmware ?

I have this info about firmware date and version :
Date du firmware
20180215
Version du firmware
00e��!u 
[image]

According to another site where they discuss about this subject, the new firmware update date should be at least : 
Firmware build date             20210304
Firmware version                  0023

aurelien219, 2023-03-12:
I have the gateway to upgrade firmwares of my devices. I don't know how to upgrade without the gateway.

I have this :
Date du firmware
20191106
Version du firmware
004a

LeorFinacre, 2023-03-13:
I'm back with some news.
I managed to get the manual OTA update working.

I'll see if it fixes the problem when it's done updating the firmware.

LeorFinacre, 2023-03-13:
That was it, now it's fixed and I have control on the LEDs and I have the power consumption working.