Koenkk/zigbee2mqtt #16900: OTA failed for LEDVANCE 4058075728981

Snowrisk, 2023-03-03:
### What happened?

OTA failed with Message

`Info 2023-03-03 17:04:36 Updating 'Leuchte Abstellkammer' to latest firmware
Info 2023-03-03 17:04:36 MQTT publish: topic 'zigbee2mqtt/Leuchte Abstellkammer', payload '{"brightness":254,"linkquality":0,"power_on_behavior":"previous","state":"ON","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":null}'
Info 2023-03-03 17:04:36 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Leuchte Abstellkammer"},"error":"Update of 'Leuchte Abstellkammer' failed (The value of \"offset\" is out of range. It must be >= 0 and <= 188380. Received 188384)","status":"error","transaction":"yq6u8-1"}'
Error 2023-03-03 17:04:36 Update of 'Leuchte Abstellkammer' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384)`

### What did you expect to happen?

Firmware should be updated

### How to reproduce it (minimal and precise)

Click "Update Device Firmware" in Frontend

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

 20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N) Zigbee USB-Stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10883890/log.txt)

Jeder321, 2023-03-18:
I got the exact same error for LEDVANCE 4058075729063 with the same adapter

```
info 2023-03-18 20:42:23 Updating 'Swiatlo' to latest firmware
info 2023-03-18 20:42:26 MQTT publish: topic 'zigbee2mqtt/Swiatlo', payload '{"linkquality":48,"power_on_behavior":null,"state":"OFF","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":null}'
info 2023-03-18 20:42:26 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Swiatlo"},"error":"Update of 'Swiatlo' failed (The value of \"offset\" is out of range. It must be >= 0 and <= 188380. Received 188384)","status":"error","transaction":"o4a9g-1"}'
error 2023-03-18 20:42:26 Update of 'Swiatlo' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384)
```

Snowrisk, 2023-04-18:
Any news on this? The Problem still exists.

aurban6, 2023-04-23:
I got the exact same error for LEDVANCE (SMART+ Classic P40 E14 dimmable white) Model: 4058075729100
`Zigbee2MQTT:error 2023-04-23 11:26:00: Update of 'Light1' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384)
Zigbee2MQTT:error 2023-04-23 11:26:12: Update of 'Light2' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384)`

gatnegre10, 2023-05-18:
I got the exact same error. No solution?

closed 2023-06-26 by github-actions[bot]

jjoergensen, 2023-07-16:
I have the same issue. Update of 'Lamp' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384. This is for LEDVANCE 4058075728981. I use Conbee II.

wojtasiq, 2023-12-22:
I still have this problem

Sueff1812, 2024-01-06:
Same problem

tom230764, 2024-01-28:
the same problem here

Locoblade, 2024-02-25:
Same problem here with a Ledvance 4058075729100 bulb

Koenkk, 2024-02-26:
can someone provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Locoblade, 2024-02-26:
@Koenkk Here you go

`Info 2024-02-26 20:49:47Updating 'Living Room Lamp' to latest firmware
Info 2024-02-26 20:49:47MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Living Room Lamp' to latest firmware","meta":{"device":"Living Room Lamp","status":"update_in_progress"},"type":"ota_update"}'
Debug 2024-02-26 20:49:48Received Zigbee message from 'Living Room Lamp', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"03143660"}' from endpoint 1 with groupID 0
Debug 2024-02-26 20:49:48OTA: Updating to latest '0xf0d1b8be24080c15' (P40 DIM T)
Debug 2024-02-26 20:49:48OTA: Using endpoint '1'
Debug 2024-02-26 20:49:48Received Zigbee message from 'Living Room Lamp', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51656288,"imageType":206,"manufacturerCode":4489}' from endpoint 1 with groupID 0
Debug 2024-02-26 20:49:48OTA: Got request '{"fieldControl":0,"manufacturerCode":4489,"imageType":206,"fileVersion":51656288}'
Debug 2024-02-26 20:49:48LedvanceOTA: call getImageMeta for P40 DIM T
Debug 2024-02-26 20:49:48OTA: Getting new image for '0xf0d1b8be24080c15' (P40 DIM T), latest meta {"fileVersion":52442720,"fileSize":188384,"url":"https://api.update.ledvance.com/v1/zigbee/firmwares/download?company=4489&product=206&version=3.32.54.96","sha256":"fb0c264961c05f3d93bf6e99660b3d5929e918aa90efc56e00f9db7295e80eaf"}
Debug 2024-02-26 20:49:48OTA: Update checksum validation succeeded for '0xf0d1b8be24080c15' (P40 DIM T)
Debug 2024-02-26 20:49:48Update of 'Living Room Lamp' failed (RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384)
Info 2024-02-26 20:49:48MQTT publish: topic 'zigbee2mqtt/Living Room Lamp', payload '{"brightness":254,"linkquality":18,"state":"OFF","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":true}'
Info 2024-02-26 20:49:48MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Living Room Lamp' failed (The value of \"offset\" is out of range. It must be >= 0 and <= 188380. Received 188384)","meta":{"device":"Living Room Lamp","status":"update_failed"},"type":"ota_update"}'
Info 2024-02-26 20:49:48MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Living Room Lamp"},"error":"Update of 'Living Room Lamp' failed (The value of \"offset\" is out of range. It must be >= 0 and <= 188380. Received 188384)","status":"error","transaction":"wwpqp-2"}'
Error 2024-02-26 20:49:48Update of 'Living Room Lamp' failed (The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384)
Debug 2024-02-26 20:49:48RangeError: The value of "offset" is out of range. It must be >= 0 and <= 188380. Received 188384 at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:88:9) at Buffer.readUInt32LE (node:internal/buffer:222:5) at parseSubElement (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:149:27) at parseImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:190:25) at getNewImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:600:19) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:430:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:269:41) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)`

Koenkk, 2024-02-27:
In the z2m directory, can you change the `node_modules/zigbee-herdsman-converters/lib/ota/common.js` to [this](https://gist.github.com/Koenkk/8f8d91f40420f8cc2646f87f44685d8e) and see what happens when trying to update?

Docker and HA users can use [these instructions](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt?tab=readme-ov-file#testing-changes-locally)

Locoblade, 2024-02-27:
@Koenkk  Sorry I've tried but those instructions have lost me,  I've tried editing the common.js file and updating it using vi but it despite saving seems to revert to the original file content when the add-on is rebooted.  I'm fairly clueless when it comes to Linux so I'm sure I'm doing something basic wrong but hopefully someone with a bit more knowledge can try it and report back. Is there an easier way I can upload the file via WinSCP or something?

wojtasiq, 2024-02-27:
For me the container doen't start after performing restart and when I start it from UI it reverts changes.

Locoblade, 2024-02-28:
> For me the container doen't start after performing restart and when I start it from UI it reverts changes.

Yep should have said but that's what happened for me too, the command line restart didn't do anything and restarting from the UI reset it

Koenkk, 2024-02-29:
What's in the container log?

wojtasiq, 2024-03-01:
@Koenkk Ok, my bad - I had an error on first line in JS after pasting your changes.
After restarting container OTA update succeded.

Koenkk, 2024-03-02:
@wojtasiq when checking for an OTA again now, it does say that there are no updates available anymore?

wojtasiq, 2024-03-02:
@Koenkk yes, no updates. New version of frimeware is `03203660`

Koenkk, 2024-03-02:
Great, pushed the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

ilwalti, 2024-04-15:
Hi all, I've updated the common.js file as above, but I'm still getting the size mismatch error... If possible, I would avoid the dev branch, for other matters. Any solution in the meantime?
Thanks @ all

```
info  2024-04-15 10:34:50: Updating 'Luce Walter' to latest firmware
info  2024-04-15 10:34:51: MQTT publish: topic 'zigbee2mqtt/Luce Walter', payload '{"brightness":102,"linkquality":152,"state":"OFF","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":null}'
info  2024-04-15 10:34:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xf0d1b8be24080bfd"},"error":"Update of 'Luce Walter' failed (Size mismatch)","status":"error"}'
error 2024-04-15 10:34:51: Update of 'Luce Walter' failed (Size mismatch)