Koenkk/zigbee2mqtt #16901: TS110E dimmer state stays ON

manodory, 2023-03-03:
### What happened?

I have this Zigbee Dimmer:
Manufacturer: _TZ3210_zxbtub8r
Model: TS110E
After turning it off, I've noticed it comes back to to ON state in home assistant, though the light won't actually turn on. 
I've checked the STATE status, refreshed it, and it is reporting ON. turning it off, then clicking the refresh button, and it comes back to the ON state (though the light itself actually stays off). This is causing HomeAssistant to get the wrong state of the light. 

### What did you expect to happen?

I expect the dimmer to report the currect state. 

### How to reproduce it (minimal and precise)

In the Exposes tab of the device, go to State. Turn it off and then refresh. it comes back to the ON state. 

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

0.6.10

### Adapter

Zigbee ESP32 GW (PoE IP Gateway)

chojnicki, 2023-03-03:
Same :( https://github.com/Koenkk/zigbee2mqtt/issues/16863

ash286, 2023-03-05:
I think this version has a major problem. My MOES touch dimmer did stop working after updating Z2M. Rolling back update works just fine.

veveme, 2023-03-06:
> I think this version has a major problem. My MOES touch dimmer did stop working after updating Z2M. Rolling back update works just fine.

@ash286 do you use home assistant? I cannot find a way to downgrade zigbee2mqtt addon.

ash286, 2023-04-08:
> > I think this version has a major problem. My MOES touch dimmer did stop working after updating Z2M. Rolling back update works just fine.
> 
> @ash286 do you use home assistant? I cannot find a way to downgrade zigbee2mqtt addon.

You should find a backup to this Z2M add-on, but only when created an automatic backup at a time of this add-on update. Default add-on backup won't happen prior to update when "Create backup before updating" option is unchecked. Backup of previous Z2M version should be in Settings >> System >> Backups.

closed 2023-05-17 by github-actions[bot]

rockzill, 2023-08-03:
Did anyone resolve this issue? I have the same problem with my TS110E, and it is super annoying.

manodory, 2023-08-03:
Hi,
I found myself replacing the problematic modules... So I guess not.

rockzill, 2023-08-03:
> Hi, I found myself replacing the problematic modules... So I guess not.
Which model did you choose for replacing the problematic one? The issue with TS110E changing its state to ON after refreshing is extremely annoying

manodory, 2023-08-03:
I replaced one of them with the one showed in the attached image.
The other one I replaced with another TS110E, the new one didn't have that
problem.
I think it was one specific buggy firmware that is no longer being
deployed. Just try to order 2 or 3 modules and see what works best for
you...