Koenkk/zigbee2mqtt #16906: Able to call a service or Home Assistant to Restart ZIgbeemqtt?

cloudbr34k84, 2023-03-04:
### Is your feature request related to a problem? Please describe

I find myself restarting Z2M when alot of devices go offline. and restart Z2M helps to get them back online. I also find that retart Z2M at night seems to help with repsonsivness as my network for some reason becomes slow and unrepsonsive.

### Describe the solution you'd like

It would be great to be able to restart Z2M using a service call with HA.

### Describe alternatives you've considered

having to manually go itno Z2M and settings and restart.

### Additional context

None required

fsedarkalex, 2023-03-24:
I would like to add on this: It should be a feature that can be disabled via config for security purposes

Koenkk, 2023-03-26:
To restart Zigbee2MQTT over MQTT see https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

elad-bar, 2023-04-10:
I implemented such functionality using NodeRED, but it should be simpler,
what needed is a call re-discover all devices as happens within the HA extension start function:
```javascript
        // MQTT discovery of all paired devices on startup.
        for (const entity of [...this.zigbee.devices(false), ...this.zigbee.groups()]) {
            this.discover(entity, true);
        }

        // Send availability messages, this is required if the legacy_availability_payload option has been changed.
        this.eventBus.emitPublishAvailability();
```

should be exposed as MQTT topic same as available for restart.

will try to get to it during the weekend

elad-bar, 2023-04-10:
will test it later this week and post PR if it works:
[add HA rediscover MQTT topic](https://github.com/elad-bar/zigbee2mqtt/commit/c8f186e4872981660a953b824fd7955e9d4418e1)

closed 2023-05-19 by github-actions[bot]