Koenkk/zigbee2mqtt #16911: Livarno Lux light bar RGB+CCT (black/white): Cannot set color anymore

Fridolin34, 2023-03-04:
### What happened?

Since updating to Z2M 1.30.2-1 I cannot change the color of this light anymore. It still can be switched on and off, but it is only white. 
The log indicates no errors. It shows that the color should be changed, but simply nothing happens.

### What did you expect to happen?

Since it's an RGB light, the colors should be adjustable like it's been during the last 2 years. Other lights from LIDL / Linvaro are still working fine.

### How to reproduce it (minimal and precise)

Just switch the light on and try to change the color.

### Zigbee2MQTT version

Current version: 1.30.2-1

### Adapter firmware version

20201128

### Adapter

CC2531

Fridolin34, 2023-03-12:
Am I the only one having this issue? Can I do anything to help solving this?

bkupidura, 2023-03-13:
@Fridolin34 after upgrade i see same problem with `TS0505A`. @Koenkk is there any data we can provide to help fixing this issue?

haissio, 2023-03-15:
Same problem here.
Running 1.30.2-1 with a electroama adapter (CC2652R)

Fridolin34, 2023-03-18:
I now switched back to 1.30.0 and it is working fine. Not really a solution, but there is nothing in the newer versions I really need, so a temporary solution.

Fridolin34, 2023-03-18:
Just regarding the change log, could this bug be related to this change? 
[Fix color handling for LIDL light bar (Koenkk/zigbee2mqtt#13421) #5514](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5514)

Koenkk, 2023-03-18:
A possible fix has just been pushed. Please check if it works now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Fridolin34, 2023-03-19:
I've just tried it out, but it is still not working, although I cannot say for sure, whether those changes are already in the version I'm using. When you say 'in a few hours', I guess it should be in the build I've just downloaded roughly 15 mins ago.

The log starts with:
Zigbee2MQTT:info  2023-03-19 09:54:07: Starting Zigbee2MQTT version 1.30.2-dev (commit #2c57bf8)
Zigbee2MQTT:info  2023-03-19 09:54:07: Starting zigbee-herdsman (0.14.100)

Koenkk, 2023-03-20:
@Fridolin34 could you provide the debug log when setting the color on the latest known working version vs that 1.30.2-dev?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Fridolin34, 2023-03-20:
Adding 'advanced: ...' to the configuration.xml, like described in the help linked above does not seem to work. But either way, the log level seems to be set correctly. In case you miss something, please let me know.

This is what I see, once I'm using 1.30.0:

`Debug 2023-03-20 18:01:00Received MQTT message on 'zigbee2mqtt/Lidl light/set' with data '{"state":"ON","brightness":162}'
Debug 2023-03-20 18:01:00Publishing 'set' 'brightness' to 'Lidl light'
Info 2023-03-20 18:01:00MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":253,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.172,"y":0.747},"color_mode":"xy","color_temp":129,"last_seen":"2023-03-20T18:00:59+01:00","linkquality":92,"state":"OFF"}'
Info 2023-03-20 18:01:00MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.172,"y":0.747},"color_mode":"xy","color_temp":129,"last_seen":"2023-03-20T18:00:59+01:00","linkquality":92,"state":"ON"}'
Debug 2023-03-20 18:01:05Received MQTT message on 'zigbee2mqtt/Lidl light/set' with data '{"state":"ON","color":{"x":0.136,"y":0.04}}'
Debug 2023-03-20 18:01:05Skipping state because of Home Assistant
Debug 2023-03-20 18:01:05Publishing 'set' 'color' to 'Lidl light'
Info 2023-03-20 18:01:05MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.172,"y":0.747},"color_mode":"xy","color_temp":129,"last_seen":"2023-03-20T18:01:04+01:00","linkquality":92,"state":"ON"}'
Debug 2023-03-20 18:01:05Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x588e81fffefb9cf2!
Info 2023-03-20 18:01:05MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.136,"y":0.04},"color_mode":"xy","color_temp":575,"last_seen":"2023-03-20T18:01:04+01:00","linkquality":92,"state":"ON"}'`

And now using 1.30.2-dev commit: [2c57bf8](https://github.com/Koenkk/zigbee2mqtt/commit/2c57bf8):

`Debug 2023-03-20 18:12:09Received MQTT message on 'zigbee2mqtt/Lidl light/set' with data '{"state":"ON"}'
Debug 2023-03-20 18:12:09Publishing 'set' 'state' to 'Lidl light'
Info 2023-03-20 18:12:09MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.136,"y":0.04},"color_mode":"xy","color_temp":575,"last_seen":"2023-03-20T18:12:08+01:00","linkquality":94,"state":"OFF"}'
Info 2023-03-20 18:12:09MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.136,"y":0.04},"color_mode":"xy","color_temp":575,"last_seen":"2023-03-20T18:12:08+01:00","linkquality":94,"state":"ON"}'
Debug 2023-03-20 18:12:11Received Zigbee message from '0xa4c13853363c5eb6', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,8232]}' from endpoint 1 with groupID 0
Info 2023-03-20 18:12:11MQTT publish: topic 'zigbee2mqtt/0xa4c13853363c5eb6', payload '{"child_lock":"UNLOCK","current":0,"energy":82.32,"indicator_mode":"off/on","last_seen":"2023-03-20T18:12:10+01:00","linkquality":47,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":235}'
Debug 2023-03-20 18:12:14Received MQTT message on 'zigbee2mqtt/Lidl light/set' with data '{"state":"ON","color":{"h":120.0,"s":100.0}}'
Debug 2023-03-20 18:12:14Skipping state because of Home Assistant
Debug 2023-03-20 18:12:14Publishing 'set' 'color' to 'Lidl light'
Info 2023-03-20 18:12:14MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.136,"y":0.04},"color_mode":"xy","color_temp":575,"last_seen":"2023-03-20T18:12:13+01:00","linkquality":94,"state":"ON"}'
Info 2023-03-20 18:12:14MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.136,"y":0.04},"color_mode":"xy","color_temp":575,"last_seen":"2023-03-20T18:12:13+01:00","linkquality":94,"state":"ON"}'
Debug 2023-03-20 18:12:14Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x588e81fffefb9cf2!
Info 2023-03-20 18:12:14MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":162,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"hs","color_temp":129,"last_seen":"2023-03-20T18:12:13+01:00","linkquality":94,"state":"ON"}'
`

It seems to be complaining about 'Missing colorTempPhysicalMin and/or colorTempPhysicalMax'

Fridolin34, 2023-03-20:
The only difference I can spot after comparing both logs is that in the second (non working) scenario, in the last line it says something about "color_mode":"hs" and before it has always used "color_mode":"xy".

Actually I don't know, why some of those lines seem to be duplicated, but at least this line is not there in the working solution, where constantly only xy is used.

Koenkk, 2023-03-21:
How do you change the color, via home assistant or the z2m dashboard/

Koenkk, 2023-03-22:
I need some more logging, could you provide the **herdsman** debug log when setting the color on the latest known working version vs 1.30.2?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Fridolin34, 2023-03-23:
Took me a while, but I hope this helps - again first the working scenario and then the not working one:

Zigbee2MQTT:debug 2023-03-23 20:18:25: Received MQTT message on 'zigbee2mqtt/light light/set' with data '{"state":"ON","color":{"x":0.157,"y":0.05}}'
Zigbee2MQTT:debug 2023-03-23 20:18:25: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2023-03-23 20:18:25: Publishing 'set' 'color' to 'light light'
2023-03-23T19:18:25.613Z zigbee-herdsman:controller:endpoint Command 0x588e81fffefb9cf2/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":10289,"colory":3277}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-03-23T19:18:25.615Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x588e81fffefb9cf2:52839/1 (0,0,1)
2023-03-23T19:18:25.618Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":52839,"destendpoint":1,"srcendpoint":1,"clusterid":768,"transid":56,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[1,12,7,49,40,205,12,0,0]}}
2023-03-23T19:18:25.619Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,103,206,1,1,0,3,56,0,30,9,1,12,7,49,40,205,12,0,0,97]
2023-03-23T19:18:25.633Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2023-03-23T19:18:25.634Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2023-03-23T19:18:25.634Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2023-03-23T19:18:25.635Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2023-03-23T19:18:25.637Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-23T19:18:25.642Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,56,254]
2023-03-23T19:18:25.651Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,56,254]
2023-03-23T19:18:25.653Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,56] - 254
2023-03-23T19:18:25.654Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":56}
2023-03-23T19:18:25.655Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-23T19:18:25.663Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,103,206,0,43]
2023-03-23T19:18:25.663Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,103,206,0,43]
2023-03-23T19:18:25.663Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [103,206,0] - 43
2023-03-23T19:18:25.664Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":52839,"relaycount":0,"relaylist":[]}
2023-03-23T19:18:25.664Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-03-23T19:18:25.722Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,0,3,103,206,1,1,0,89,0,211,193,38,0,0,5,8,12,11,7,0,103,206,29,162]
2023-03-23T19:18:25.722Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,0,3,103,206,1,1,0,89,0,211,193,38,0,0,5,8,12,11,7,0,103,206,29,162]
2023-03-23T19:18:25.723Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,0,3,103,206,1,1,0,89,0,211,193,38,0,0,5,8,12,11,7,0,103,206,29] - 162
2023-03-23T19:18:25.724Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":768,"srcaddr":52839,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":2539987,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,12,11,7,0]}}
2023-03-23T19:18:25.730Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":12,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":7,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":52839,"endpoint":1,"linkquality":89,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:info  2023-03-23 20:18:25: MQTT publish: topic 'zigbee2mqtt/light light', payload '{"brightness":253,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.172,"y":0.747},"color_mode":"xy","color_temp":129,"last_seen":"2023-03-23T20:18:25+01:00","linkquality":89,"state":"ON"}'
2023-03-23T19:18:25.740Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2023-03-23 20:18:25: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x588e81fffefb9cf2!
Zigbee2MQTT:info  2023-03-23 20:18:25: MQTT publish: topic 'zigbee2mqtt/light light', payload '{"brightness":253,"color":{"h":243,"hue":243,"s":100,"saturation":100,"x":0.157,"y":0.05},"color_mode":"xy","color_temp":581,"last_seen":"2023-03-23T20:18:25+01:00","linkquality":89,"state":"ON"}'
2023-03-23T19:18:26.060Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,25,92,1,1,0,55,0,191,197,38,0,0,8,24,229,10,8,5,33,66,9,25,92,29,56]
2023-03-23T19:18:26.060Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,25,92,1,1,0,55,0,191,197,38,0,0,8,24,229,10,8,5,33,66,9,25,92,29,56]
[... cut ...]
2023-03-23T19:29:28.540Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2023-03-23 20:29:28: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x588e81fffefb9cf2!
Zigbee2MQTT:info  2023-03-23 20:29:28: MQTT publish: topic 'zigbee2mqtt/Lidl light', payload '{"brightness":253,"color":{"h":120,"hue":120,"s":100,"saturation":100,"x":0.1724,"y":0.7468},"color_mode":"hs","color_temp":129,"last_seen":"2023-03-23T20:29:28+01:00","linkquality":86,"state":"ON"}'

Koenkk, 2023-03-24:
Could you provide the data/database.db entry of your device?

Fridolin34, 2023-03-24:
{"id":19,"type":"Router","ieeeAddr":"0x588e81fffefb9cf2","nwkAddr":52839,"manufId":4098,"manufName":"_TZ3000_gek6snaj","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_gek6snaj","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1679623200726,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-03-26:
@AndreKR this is probably due to https://github.com/Koenkk/zigbee-herdsman-converters/pull/5514

The device fingerprint contains two different models, TS0505A and TS0505B

```js
        fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_gek6snaj'},
            {modelID: 'TS0505B', manufacturerName: '_TZ3210_iystcadi'}],
        model: '14149505L/14149506L',
```

@Fridolin34 has the TS0505A, @AndreKR do you have the TS0505B? If yes, I think your fixes only have to be applied to the TS0505B.

AndreKR, 2023-03-28:
Yes
> Zigbee Model: TS0505B
> Zigbee Manufacturer: _TZ3210_iystcadi

Koenkk, 2023-03-29:
Thanks! @AndreKR @Fridolin34 is your device the `14149505L` or `14149506L`?

Fridolin34, 2023-03-29:
It seems to work, thanks a lot.

The label on my device says: 14149505L

AndreKR, 2023-03-29:
So does mine.

[image]

Koenkk, 2023-03-30:
I see, then I will use `14149505L/14149506L_1`/`14149505L/14149506L_2`

Fridolin34, 2023-04-01:
Working fine again since version 1.30.3-1.