Koenkk/zigbee2mqtt #16912: [Feature request]: Changeability of the primary MQTT identifier for Devices

Timmmy001, 2023-03-04:
### Is your feature request related to a problem? Please describe

When transmitting via MQTT, the friendly name is used as the primary identifier for Devices ("zigbee2mqtt/**Restroom_Light**/set:...").
The problem starts when you change the friendly name, which is not common, but does happen and makes follow up works.

### Describe the solution you'd like

As in normal programming, it might be better to use the fixed IEEE address as id and to transfer the friendly name as an additional value.

### Describe alternatives you've considered

No other alternatives.

### Additional context

Example:
[image]

Timmmy001, 2023-04-05:
I think it is still a good idea

Timmmy001, 2023-05-09:
Sorry, but I think it is still a good idea

P1Rebo, 2023-06-07:
Think also this is a good idea! Until now I used the description and was fine. But when integrating within HomeAssistant the description is not taken, e.g. As name of the device. In HA there can be an object_id which would also better fit to the zigbee2mqtt device identifier field. Friendly name could then be the name in HA.

P1Rebo, 2023-07-09:
Ping. Come on it's not that much to do ;-)

Timmmy001, 2023-07-11:
☕

closed 2023-08-18 by github-actions[bot]

ZaikinaEvgeniya-2, 2023-09-12:
###

Funkisl, 2024-09-23:
I also think it is still a good idea