Koenkk/zigbee2mqtt #16913: Zigbee2MQTT doesn't restart after Windows update and VirtualBox restart

robtperk, 2023-03-04:
### What happened?

I need to restart the computer occasionally to update or a power loss. When this happens, VirtualBox and HA restart, but Zigbee2MQTT doesn't. I have to unplug the Sonoff dongle and then restart Zigbee2MQTT manually to bring back my devices.

### What did you expect to happen?

I would expect Z2MQTT to restart automatically.

### How to reproduce it (minimal and precise)

Restart Windows with HA running under Oracle VirtualBox on the Windows box.

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

n/a

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

junosuarez, 2023-03-05:
How are you running zigbee2mqtt? If it's in Docker, you can use the `--restart always` flag (or if using Docker Compose, `restart: always` in docker-compose.yaml). Otherwise, you can search for how to run an application as a service for your operating system to have it run and restart automatically.

robtperk, 2023-03-05:
I'm running under HA OS on a VirtualBox under Windows 11. I do restart
VirtualBox automatically when Windows restarts, and that opens with the HA
instance, but don't know how to make HA restart the Zigbee2MQTT
automatically. I actually believe it does that, but the problem is, even if
I manually restart it after a reboot, the connections are lost, and I have
to manually disconnect and reconnect the Sonoff USB 3 Zigbee dongle, then
restart  Zigbee2MQTT. Then my devices reconnect.

junosuarez, 2023-03-07:
Ah! So it sounds like you're running zigbee2mqtt inside of a virtual machine. When you manually start it, it connects to the USB dongle, but when it's restarted, it can't connect to the USB device - do I have this right?

If so, VirtualBox needs to know which VM to reconnect the USB device to. This is possible using filters - see https://www.net-usb.com/virtual-usb/virtualbox-usb-passthrough/#How for a guide.

junosuarez, 2023-03-07:
If you get it working, let me know. If still no, can you post the logs from zigbee2mqtt?

robtperk, 2023-03-08:
Thanks. I've just ordered a refurbished Dell Wyse Intel 5, 8gb ram, 240gb
ssd for ~$125US. Cheaper and more powerful than the RPi 4B.
I'm going to move my HA from VirtualBox to the Dell running Ubuntu and HA
OS. I think that will solve my problem and at least use less power and I
can turn off the desktop with all it's attachments. I'll keep you posted. I
won't get the computer for about a week. Any suggestions or misgivings
before I set it up?

robtperk, 2023-04-08:
Ok, thanks.

closed 2023-05-17 by github-actions[bot]