Koenkk/zigbee2mqtt #16916: There's no OTA endpoint available for Xiaomi ZNGZDJ11LM

phoenixxie0, 2023-03-04:
### What happened?

[image]

### What did you expect to happen?

Xiaomi endpoint linked as expected in order to upgrade the firmware successfully.

### How to reproduce it (minimal and precise)

Go to OTA using the frontend UI
Click on check updates for the device Xiaomi ZNGZDJ11LM

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

 20221102

### Adapter

TTGO T-Internet-POE CC2652P

### Debug log

Debug 2023-03-05 00:06:16Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"0x00158d0007012b9b","transaction":"5jq8o-5"}'
Info 2023-03-05 00:06:16Checking if update available for '0x00158d0007012b9b'
Debug 2023-03-05 00:06:16Check if update available for '0x00158d0007012b9b' (lumi.curtain.aq2)
Info 2023-03-05 00:06:16MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0x00158d0007012b9b"},"error":"Failed to check if update available for '0x00158d0007012b9b' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"5jq8o-5"}'
Error 2023-03-05 00:06:16Failed to check if update available for '0x00158d0007012b9b' (Failed to find endpoint which support OTA cluster)
Debug 2023-03-05 00:06:16AssertionError [ERR_ASSERTION]: Failed to find endpoint which support OTA cluster at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:243:9) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:130:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:73) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:121:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1180:20) at Receiver.emit (node:events:513:28)
Debug 2023-03-05 00:06:27Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"5jq8o-6"}'

Koenkk, 2023-03-05:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.curtain.aq2'],
    model: 'ZNGZDJ11LM',
    description: 'Aqara roller shade motor',
    vendor: 'Xiaomi',
    whiteLabel: [{vendor: 'Xiaomi', model: 'SRSC-M01'}],
    fromZigbee: [fz.xiaomi_basic, fz.xiaomi_curtain_position, fz.xiaomi_curtain_position_tilt],
    toZigbee: [tz.xiaomi_curtain_position_state, tz.xiaomi_curtain_options],
    exposes: [e.cover_position().setAccess('state', ea.ALL),
        exposes.binary('running', ea.STATE, true, false)
            .withDescription('Whether the motor is moving or not')],
    ota: ota.zigbeeOTA,
    configure: async (device, coordinatorEndpoint, logger) => {
        utils.attachOutputCluster(device, 'genOta');
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

phoenixxie0, 2023-03-05:
The z2m can not start .
there is the log

[19:15:54] INFO: Preparing to start...
[19:15:54] INFO: Socat not enabled
[19:15:55] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:18
    ota: ota.zigbeeOTA,
         ^
ReferenceError: ota is not defined
    at /app/data/extension/externally-loaded.js:18:10
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

leroyloren, 2023-03-05:
Same issue for `GZCGQ11LM`
https://www.zigbee2mqtt.io/devices/GZCGQ11LM.html

Koenkk, 2023-03-06:
Whoops, updated https://github.com/Koenkk/zigbee2mqtt/issues/16916#issuecomment-1455027342

phoenixxie0, 2023-03-07:
[image]
Also can not update

Koenkk, 2023-03-07:
Can you provide the `data/database.db` entry of this device after starting z2m at least once with the external converter?

phoenixxie0, 2023-03-08:
I can not find the database.db file in the data. But have the database.db.backup in it.
there is the file
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/10915161/database.db.zip)

Koenkk, 2023-03-08:
I don't think the external converter has been loaded. To check, reconfigure the device by pressing the yellow refresh icon in the frontend. Only if this succeeds the external converter has been loaded.

phoenixxie0, 2023-03-08:
> I don't think the external converter has been loaded. To check, reconfigure the device by pressing the yellow refresh icon in the frontend. Only if this succeeds the external converter has been loaded.

After I can not update with add the - ext_converter.js , I removed it.
I will add it back and restart z2m now

KitHubek, 2023-03-08:
The same here for devices XIAOMI: SP-EUC01 and ZNCZ04LM
As You can see at screenshots some device has newer version some older. Older at list shows butto with avialable update, but if button clicked, see error at screenshot

[image]

[image]

phoenixxie0, 2023-03-10:
> > After I can not update with add the - ext_converter.js , I removed it.
> > I will add it back and restart z2m now
> 
> I don't understand what you mean, does the configure succeed or not?

I have add the "- ext_converter.js" back, and restart the z2m, but it also can not do a OTA, the file is the new ".db" file after I add the - ext_converter.js" back.

Koenkk, 2023-03-11:
Can you try to stop z2m, manually add `25` to the `inClusterList` in `data/database.db` of this device, check z2m and see if the issue is fixed?

KitHubek, 2023-03-11:
maby this help. At states:

**ZNCZ04LM**

```
{
    "auto_off": false,
    "consumer_connected": true,
    "consumption": 1988.42,
    "current": 0.54,
    "device_temperature": 29,
    "energy": 1988.42,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-03-11T13:16:03+01:00",
    "led_disabled_night": false,
    "linkquality": 255,
    "overload_protection": 2200,
    "power": 128.52,
    "power_outage_count": 5,
    "power_outage_memory": true,
    "state": "ON",
    "temperature": 32,
    "update": {
        "installed_version": "unknown",
        "state": "available"
    },
    "update_available": true,
    "voltage": 236
}
```

**SP-EUC01**

```
{
    "consumer_connected": true,
    "consumption": 67.24,
    "current": 0.13,
    "device_temperature": 29,
    "energy": 67.24,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-03-11T13:21:45+01:00",
    "linkquality": 255,
    "power": 34,
    "power_outage_count": 6,
    "power_outage_memory": true,
    "state": "ON",
    "temperature": 35,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": "available"
    },
    "update_available": true,
    "voltage": 230
}
```

KitHubek, 2023-03-11:
maby this help. At states:

**ZNCZ04LM**

```
{
    "auto_off": false,
    "consumer_connected": true,
    "consumption": 1988.42,
    "current": 0.54,
    "device_temperature": 29,
    "energy": 1988.42,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-03-11T13:16:03+01:00",
    "led_disabled_night": false,
    "linkquality": 255,
    "overload_protection": 2200,
    "power": 128.52,
    "power_outage_count": 5,
    "power_outage_memory": true,
    "state": "ON",
    "temperature": 32,
    "update": {
        "installed_version": "unknown",
        "state": "available"
    },
    "update_available": true,
    "voltage": 236
}
```

**SP-EUC01**

```
{
    "consumer_connected": true,
    "consumption": 67.24,
    "current": 0.13,
    "device_temperature": 29,
    "energy": 67.24,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-03-11T13:21:45+01:00",
    "linkquality": 255,
    "power": 34,
    "power_outage_count": 6,
    "power_outage_memory": true,
    "state": "ON",
    "temperature": 35,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": "available"
    },
    "update_available": true,
    "voltage": 230
}
```

```
    "update": {
        "installed_version": "unknown",
        "state": "available"
    },

    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": "available"
    },
```

phoenixxie0, 2023-03-11:
> Can you try to stop z2m, manually add `25` to the `inClusterList` in `data/database.db` of this device, check z2m and see if the issue is fixed?

I have added it. like this
`{"id":22,"type":"Router","ieeeAddr":"0x00158d0007012b9b","nwkAddr":28178,"manufId":4447,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.curtain.aq2","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,4,3,5,10,258,25,13,19],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.curtain.aq2","appVersion":31}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":31,"stackVersion":2,"hwVersion":1,"dateCode":"10-23-2020","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1678291464959,"defaultSendRequestWhen":"immediate"}`

But it still can not update.
`Error 2023-03-11 20:29:03Failed to check if update available for '书房窗帘' (Failed to find endpoint which support OTA cluster)
Debug 2023-03-11 20:29:03AssertionError [ERR_ASSERTION]: Failed to find endpoint which support OTA cluster at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:243:9) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/zigbeeOTA.js:130:19) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:73) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:121:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1180:20) at Receiver.emit (node:events:513:28)`

Koenkk, 2023-03-12:
Ah sorry, you need to add it to `outClusterList` instead of `inClusterList`

KitHubek, 2023-03-12:
i checked my database file, and all this device has 25 in inclusterList. Here example:

`{"id":31,"type":"Router","ieeeAddr":"0x04cf8cdf3c8a3175","nwkAddr":35688,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug.mmeu01","epList":[1,21,22,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,5,6,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.plug.mmeu01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"09-06-2019"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[100,16,1,3,40,38,152,57,51,51,85,66,149,57,16,15,38,68,150,57,0,224,18,69,151,57,250,206,98,67,5,33,1,0,154,32,0,8,33,22,1,7,39,0,0,0,0,0,0,0,0,9,33,0,11,11,32,0,155,16,1]},"252":0,"513":1,"514":0,"515":0,"519":1,"523":2200}},"genOnOff":{"attributes":{"245":117440681,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":81,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":53.29999923706055}}},"binds":[],"configuredReportings":[],"meta":{}},"22":{"profId":260,"epId":22,"devId":81,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"09-06-2019","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678657479367,"defaultSendRequestWhen":"immediate"}
`

KitHubek, 2023-03-12:
i checked my database file, and all this device has 25 in outclusterList. Here example:

`{"id":31,"type":"Router","ieeeAddr":"0x04cf8cdf3c8a3175","nwkAddr":35688,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug.mmeu01","epList":[1,21,22,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,2,3,4,5,6,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.plug.mmeu01","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"09-06-2019"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[100,16,1,3,40,38,152,57,51,51,85,66,149,57,16,15,38,68,150,57,0,224,18,69,151,57,250,206,98,67,5,33,1,0,154,32,0,8,33,22,1,7,39,0,0,0,0,0,0,0,0,9,33,0,11,11,32,0,155,16,1]},"252":0,"513":1,"514":0,"515":0,"519":1,"523":2200}},"genOnOff":{"attributes":{"245":117440681,"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":81,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":53.29999923706055}}},"binds":[],"configuredReportings":[],"meta":{}},"22":{"profId":260,"epId":22,"devId":81,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":22,"stackVersion":2,"hwVersion":1,"dateCode":"09-06-2019","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678657479367,"defaultSendRequestWhen":"immediate"}
`

phoenixxie0, 2023-03-13:
> Ah sorry, you need to add it to `outClusterList` instead of `inClusterList`

When I add it in the file , the device can do a ota. But the log show
`Debug 2023-03-13 10:38:20Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"书房窗帘","transaction":"edmzc-1"}'
Info 2023-03-13 10:38:20Checking if update available for '书房窗帘'
Debug 2023-03-13 10:38:20Check if update available for '0x00158d0007012b9b' (lumi.curtain.aq2)
Debug 2023-03-13 10:38:20Using endpoint '1'
Debug 2023-03-13 10:38:20Received Zigbee message from '书房窗帘', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":31,"imageType":257,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Info 2023-03-13 10:38:20MQTT publish: topic 'zigbee2mqtt/书房窗帘', payload '{"device_temperature":30,"last_seen":1678675100900,"linkquality":87,"motor_state":"stopped","options":{"hand_open":false,"reverse_direction":true},"position":0,"power_outage_count":21,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-03-13 10:38:20Configuring '书房窗帘'
Debug 2023-03-13 10:38:20Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":31}'
Error 2023-03-13 10:38:20Failed to configure '书房窗帘', attempt 2 (ReferenceError: utils is not defined at Object.configure (/app/data/extension/externally-loaded.js:21:9) at Configure.configure (/app/lib/extension/configure.ts:117:37) at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:82:62) at EventEmitter.emit (node:events:525:35) at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:52:22) at Controller.<anonymous> (/app/lib/zigbee.ts:66:27) at Controller.emit (node:events:513:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:727:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70))
Debug 2023-03-13 10:38:21Received Zigbee message from '大厅窗帘', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
Info 2023-03-13 10:38:22MQTT publish: topic 'zigbee2mqtt/大厅窗帘', payload '{"last_seen":1678675101984,"linkquality":65,"motor_speed":100,"options":null,"position":100,"running":false,"state":"OPEN"}'
Debug 2023-03-13 10:38:22ZigbeeOTA: downloaded main index
Debug 2023-03-13 10:38:22Is new image available for '0x00158d0007012b9b', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":31}', latest meta '{"fileVersion":33,"fileSize":189758,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220212141304_OTA_lumi.curtain.aq2_0.0.0_0033_20220124_2C69FB.ota","sha512":"07592976767a2cb8dd0faa643a15c4e022e0af06ac2dc6f0ba86bd5a3b6126dfefa794c4bfd899ad9883445bcc3b0a7df649341324469437cd74b64a49a6f168"}'
Debug 2023-03-13 10:38:22Update available for '0x00158d0007012b9b': YES
Info 2023-03-13 10:38:22Update available for '书房窗帘'
Info 2023-03-13 10:38:22MQTT publish: topic 'zigbee2mqtt/书房窗帘', payload '{"device_temperature":30,"last_seen":1678675100900,"linkquality":87,"motor_state":"stopped","options":{"hand_open":false,"reverse_direction":true},"position":0,"power_outage_count":21,"running":false,"state":"OFF","update":{"installed_version":31,"latest_version":33,"state":"available"},"update_available":null}'`

Koenkk, 2023-03-13:
Can you remove the external converter and try again?

phoenixxie0, 2023-03-14:
> Can you remove the external converter and try again?

I can not do that, because It has upgraded.. But I will remove it wheater it work

phoenixxie0, 2023-03-14:
When I check the OTA. Z2M give the log
`Debug 2023-03-14 08:27:13Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"书房窗帘","transaction":"mod53-2"}'
Info 2023-03-14 08:27:13Checking if update available for '书房窗帘'
Debug 2023-03-14 08:27:13Check if update available for '0x00158d0007012b9b' (lumi.curtain.aq2)
Debug 2023-03-14 08:27:13Using endpoint '1'
Debug 2023-03-14 08:27:13Received Zigbee message from '书房窗帘', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":33,"imageType":257,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Info 2023-03-14 08:27:13MQTT publish: topic 'zigbee2mqtt/书房窗帘', payload '{"device_temperature":28,"last_seen":1678753634125,"linkquality":83,"motor_state":"stopped","options":{"hand_open":false,"reverse_direction":true},"position":0,"power_outage_count":22,"running":false,"state":"OFF","update":{"installed_version":33,"latest_version":33,"state":"idle"},"update_available":null}'
Debug 2023-03-14 08:27:13Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":33}'
Debug 2023-03-14 08:27:14ZigbeeOTA: downloaded main index
Debug 2023-03-14 08:27:14Is new image available for '0x00158d0007012b9b', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":33}', latest meta '{"fileVersion":33,"fileSize":189758,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220212141304_OTA_lumi.curtain.aq2_0.0.0_0033_20220124_2C69FB.ota","sha512":"07592976767a2cb8dd0faa643a15c4e022e0af06ac2dc6f0ba86bd5a3b6126dfefa794c4bfd899ad9883445bcc3b0a7df649341324469437cd74b64a49a6f168"}'
Debug 2023-03-14 08:27:14Update available for '0x00158d0007012b9b': NO
Info 2023-03-14 08:27:14No update available for '书房窗帘'
Info 2023-03-14 08:27:14MQTT publish: topic 'zigbee2mqtt/书房窗帘', payload '{"device_temperature":28,"last_seen":1678753634125,"linkquality":83,"motor_state":"stopped","options":{"hand_open":false,"reverse_direction":true},"position":0,"power_outage_count":22,"running":false,"state":"OFF","update":{"installed_version":33,"latest_version":33,"state":"idle"},"update_available":null}'
Info 2023-03-14 08:27:14MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"书房窗帘","updateAvailable":false},"status":"ok","transaction":"mod53-2"}'
Debug 2023-03-14 08:27:18Received Zigbee message from '客房窗帘', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
Info 2023-03-14 08:27:18MQTT publish: topic 'zigbee2mqtt/客房窗帘', payload '{"last_seen":1678753639637,"linkquality":14,"options":null,"position":0,"running":false,"state":"CLOSE"}'`

Koenkk, 2023-03-14:
I see, so the issue is fixed now? Your device looks up-to-date now (if yes I can integrate the fix)

phoenixxie0, 2023-03-15:
> I see, so the issue is fixed now? Your device looks up-to-date now (if yes I can integrate the fix)

Yes，It have already up to date. But I don't know the whrater the bug come back while the image update next time.

closed 2023-04-23 by github-actions[bot]

phoenixxie0, 2023-06-04:
> I see, so the issue is fixed now? Your device looks up-to-date now (if yes I can integrate the fix)

Yes, after the modification, the update check can be performed normally.