Koenkk/zigbee2mqtt #16918: LLKZMK11LM toggled by buttons not connected through automation

djansen1987, 2023-03-04:
### What happened?

i got 2 [LLKZMK11LM](https://www.zigbee2mqtt.io/devices/LLKZMK11LM.html#xiaomi-llkzmk11lm) relays, and whenever I am replacing battery of switches/buttons the relays are toggling whenever i press the on/off button of these buttons. I am 10000% sure there are no automations or bindings created between these devices.

i have seen this happing with the following buttons: [E1524/E1810](https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810) and [81825](https://www.zigbee2mqtt.io/devices/81825.html#adurosmart-81825) ofcourse this is quite anoing because my computer is behide these relays and the impact is high when it suddenly turns of.

I have read this might be because the relays are the parrant of the button, and as you can see below in MAP that is the case. When these buttons are back in the sleepingroom and not near my workroom (studio) the problem is solved.

How can this happend and what can i do..??
[image]

### What did you expect to happen?

buttons not interfering with relays

### How to reproduce it (minimal and precise)

Please buttons near the relays and toggle main button

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20220219

### Adapter

CC2652R

djansen1987, 2023-04-05:
Please help

djansen1987, 2023-05-06:
No one ?

djansen1987, 2023-06-06:
Still an issue...

djansen1987, 2023-07-08:
Hi @Koenkk ,

This is still a big issue for me. I cannot use buttons around the relay switches from ikea as when the relay is the parent of the button it will toggle the relay without any automation or bind. How to solve this?

patrik-malina, 2023-07-12:
I have similar problem. Mine sonoff button randomly bind to xiaomi power plugs and turning it on/off.
I lost several days to look in all automations and scripts (didn't noticed that button actually turns them off).

I tried to remove/pair again button. No go.
Also cannot unbind device, it throws error.

djansen1987, 2023-07-12:
> I have similar problem. Mine sonoff button randomly bind to xiaomi power plugs and turning it on/off.
> I lost several days to look in all automations and scripts (didn't noticed that button actually [...]

I hope that someone will respond here with a solution,  I am afraid to use certain buttons near a relay switch... 

If you find an solution elsewhere please let me know here, I will do the same

Koenkk, 2023-07-13:
@djansen1987 this is probably caused by your switch being in group 0. Try to remove it by sending to `zigbee2mqtt/bridge/request/group/members/remove` payload `{"group": 0, "device":"LLKZMK11LM_DEVICE_NAME"}`

djansen1987, 2023-07-14:
Hi @Koenkk appropriate your response, but i am unsure on how to run such command. Can't find much details on how to send this kind of mqtt message manually. This is what is tried but if returns an error:

[image]

Koenkk, 2023-07-15:
This looks good but you probably need to encapsulate the payload in quotes (so `payload: "{....}"`)

djansen1987, 2023-07-15:
@Koenkk  Again thanks, the following command was accepted. But looking in to the log i think it did not work/do anything.

```yaml
service: mqtt.publish
data:
  topic: zigbee2mqtt/bridge/request/group/members/remove
  payload: '{"group": 0, "device":"Studio_Relay_2"}'
```

```
Zigbee2MQTT:debug 2023-07-15 19:33:42: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/remove' with data '{"group": 0, "device":"balkon"}'
Zigbee2MQTT:info  2023-07-15 19:33:42: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/remove', payload '{"data":{"device":"balkon"},"error":"Group '0' does not exist","status":"error"}'
Zigbee2MQTT:error 2023-07-15 19:33:42: Group '0' does not exist

Zigbee2MQTT:debug 2023-07-15 19:35:28: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/remove' with data '{"group": 0, "device":"Studio_Relay_1"}'
Zigbee2MQTT:info  2023-07-15 19:35:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/remove', payload '{"data":{"device":"Studio_Relay_1"},"error":"Group '0' does not exist","status":"error"}'
Zigbee2MQTT:error 2023-07-15 19:35:28: Group '0' does not exist
Zigbee2MQTT:debug 2023-07-15 19:35:31: Received Zigbee message from 'Studio_Relay_1', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-15 19:35:31: MQTT publish: topic 'zigbee2mqtt/Studio_Relay_1', payload '{"consumption":38.42,"current":0.01,"detection_period":505,"device_temperature":42,"energy":38.42,"interlock":false,"last_seen":"2023-07-15T17:35:31.886Z","linkquality":91,"power":0,"power_outage_count":150,"power_outage_memory":null,"state":"ON","state_l1":"OFF","state_l2":"OFF","temperature":45,"update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":231.36}'
Zigbee2MQTT:debug 2023-07-15 19:35:34: Received Zigbee message from 'Studio_Relay_2', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-15 19:35:34: MQTT publish: topic 'zigbee2mqtt/Studio_Relay_2', payload '{"consumption":32.78,"current":0,"detection_period":76,"device_temperature":37,"energy":32.78,"interlock":false,"last_seen":"2023-07-15T17:35:34.308Z","linkquality":65,"power":0,"power_outage_count":491,"power_outage_memory":null,"state":"ON","state_l1":"OFF","state_l2":"OFF","temperature":43,"update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":234.51}'
Zigbee2MQTT:debug 2023-07-15 19:35:44: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/remove' with data '{"group": 0, "device":"Studio_Relay_2"}'
Zigbee2MQTT:info  2023-07-15 19:35:44: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/remove', payload '{"data":{"device":"Studio_Relay_2"},"error":"Group '0' does not exist","status":"error"}'
Zigbee2MQTT:error 2023-07-15 19:35:44: Group '0' does not exist
```

Koenkk, 2023-07-16:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/76b452f7564e8db2654989ea097442d7

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed
- if it doesn't work, can you check if the configure succeeds? Press the yellow button in the z2m frontend -> device page

djansen1987, 2023-07-16:
Afraid it did not resolve the issue. When i added the ext.js and restarted Z2MQTT i saw all relays reconfiguring automatically. I also try manual to reconfigure. But when pressing one of the buttons the relays toggled again.

[image]

[image]
Here button "kinderbedje_sw" is toggling Relay  "Studio_relay_2"

During the toggle, this is de debug log trace i could find.

```
Zigbee2MQTT:debug 2023-07-16 16:02:58: Received Zigbee message from 'Wasmachine', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":343}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:02:58: MQTT publish: topic 'zigbee2mqtt/Wasmachine', payload '{"child_lock":"UNLOCK","current":0.34,"energy":181.46,"indicator_mode":"off/on","last_seen":"2023-07-16T14:02:58.978Z","linkquality":61,"power":48,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":242}'
Zigbee2MQTT:debug 2023-07-16 16:02:59: Received Zigbee message from 'Wasmachine', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":210}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/Wasmachine', payload '{"child_lock":"UNLOCK","current":0.21,"energy":181.46,"indicator_mode":"off/on","last_seen":"2023-07-16T14:02:59.991Z","linkquality":54,"power":48,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":242}'
Zigbee2MQTT:debug 2023-07-16 16:03:00: Received Zigbee message from 'Studio_Relay_1', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/Studio_Relay_1', payload '{"consumption":38.42,"current":0.66,"detection_period":505,"device_temperature":50,"energy":38.42,"interlock":false,"last_seen":"2023-07-16T14:03:00.548Z","linkquality":102,"power":145.26,"power_outage_count":150,"power_outage_memory":null,"state":"ON","state_l1":"ON","state_l2":"OFF","temperature":45,"update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":238.1}'
Zigbee2MQTT:debug 2023-07-16 16:03:00: Received Zigbee message from 'kinderbedje_sw', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/kinderbedje_sw', payload '{"action":"toggle","battery":50,"last_seen":"2023-07-16T14:03:00.800Z","linkquality":61,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/kinderbedje_sw', payload '{"action":"","battery":50,"last_seen":"2023-07-16T14:03:00.800Z","linkquality":61,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/kinderbedje_sw/action', payload 'toggle'
Zigbee2MQTT:debug 2023-07-16 16:03:00: Received Zigbee message from 'Studio_Relay_2', type 'attributeReport', cluster 'genOnOff', data '{"61440":124449828,"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/Studio_Relay_2', payload '{"consumption":32.78,"current":0.43,"detection_period":76,"device_temperature":40,"energy":32.78,"interlock":false,"last_seen":"2023-07-16T14:03:00.821Z","linkquality":54,"power":62.61,"power_outage_count":492,"power_outage_memory":null,"state":"ON","state_l1":"OFF","state_l2":"ON","temperature":43,"update":{"installed_version":47,"latest_version":47,"state":"idle"},"update_available":false,"voltage":240.73}'
Zigbee2MQTT:debug 2023-07-16 16:03:00: Received MQTT message on 'zigbee2mqtt/bedje/set' with data 'OFF'
Zigbee2MQTT:debug 2023-07-16 16:03:00: Publishing 'set' 'state' to 'bedje'
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/bedje', payload '{"child_lock":"UNLOCK","current":0,"energy":0.37,"indicator_mode":"off/on","last_seen":"2023-07-16T14:03:00.950Z","linkquality":14,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":241}'
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/bedje', payload '{"child_lock":"UNLOCK","current":0,"energy":0.37,"indicator_mode":"off/on","last_seen":"2023-07-16T14:03:00.950Z","linkquality":14,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":241}'
Zigbee2MQTT:debug 2023-07-16 16:03:00: Received Zigbee message from 'bedje', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:03:00: MQTT publish: topic 'zigbee2mqtt/bedje', payload '{"child_lock":"UNLOCK","current":0,"energy":0.37,"indicator_mode":"off/on","last_seen":"2023-07-16T14:03:00.957Z","linkquality":14,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":241}'
[... cut ...]
Zigbee2MQTT:debug 2023-07-16 16:03:26: Received Zigbee message from 'Wasmachine', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":53}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-07-16 16:03:26: MQTT publish: topic 'zigbee2mqtt/Wasmachine', payload '{"child_lock":"UNLOCK","current":0.24,"energy":181.46,"indicator_mode":"off/on","last_seen":"2023-07-16T14:03:26.021Z","linkquality":61,"power":53,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":242}'
```

djansen1987, 2023-07-16:
ps: when removing the ext.js and restarting. I saw this error passing by. But this is for a relay currently not connected, so think it can be ignorred.

```
Zigbee2MQTT:error 2023-07-16 16:12:36: Failed to configure 'Zonnescherm', attempt 1 (Error: Command 0x00158d00044c6f65/1 genGroups.remove({"groupid":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
```

Koenkk, 2023-07-17:
The issue is that the remote sends a command to group 0

```
Zigbee2MQTT:debug 2023-07-16 16:03:00: Received Zigbee message from 'kinderbedje_sw', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

Apparently it is not possible to remove the plug from group 0, so we can try to bind the remote to another group instead of 1. By default this remote should be bound to group 901. Did you already try to factory reset the remote and reconfigure it (might require to wake it up, otherwise it can fail)

djansen1987, 2023-07-18:
it only look like it is happening with Ikea buttons (multiple variants). Tried to re-pair the buttons, but as soon as the buttons are near the relays (and the become the parant in zigbee network) the behaviour happens. How to program the relay of button to move to another group?

djansen1987, 2023-07-19:
[image]
[image]

looks like gen onOff is not available for the LLKZMK11LM device. (Device Balkon is also a LLKZMK11LM) (does setting the group make it do what i do not want it to do ?)

patrik-malina, 2023-08-11:
I did it. (somewhat)
Deleted sonoff button.
created default binding group.
Paired button again.
It binds automatically to coordinator endpoint 1 (onoff and powercfg).
Then i can unbind it from the endpoint (only for powercfg).
Last step, bind onoff to created group.

It works now as it should. Only it shows me its offline, but works.

Thank you @Koenkk for tips.

[image]

patrik-malina, 2023-08-13:
Hmm, after fev days it started to turn on and off plugs again 😑
I'm desperate...

closed 2023-09-29 by github-actions[bot]

adynis, 2023-11-04:
Any other idea to fix this ?

Pospitch, 2024-01-14:
Is there any fix yet?

djansen1987, 2024-01-14:
Koen wrote on the linked topic it is a firmware issue at aquara. Someone neetlds to contact them...

leonardpitzu, 2024-02-07:
These devices haven't got a firmware update in ages... I guess only replacing them is an option.
In my case it's even worst as these devices control the lights in the entire house. Imagine the lights toggling for 1 second in the middle of the night... This happens randomly throughout the house.

vaidasL, 2024-09-26:
> Finger crossed... It working for several days. What id did is that i paired it not to all or coordinator, but to osram lamp. Then bind it to default group (i don't know if this matters) Need to try what will happen when lamp goes offline...

this worked for me as well!. avoid direct connection to coordinator and that might fix the problem. Mentioned the same in https://github.com/Koenkk/zigbee2mqtt/discussions/19900#discussioncomment-10766559

makskhats, 2024-11-10:
> These devices haven't got a firmware update in ages... I guess only replacing them is an option. In my case it's even worst as these devices control the lights in the entire house. Imagine the lights toggling for 1 second in the middle of the night... This happens randomly throughout the house.

I have exactly the same problem as you described. I found that it is due to the relay randomly losing power for short periods of time and when an external wired button is connected, the switching occurs. Without the external button, I just see the power loss counter accumulate. Have you found a solution to this problem?

george-cz, 2025-09-09:
Just sharing a solution here, there is a new Aqara firmware that soves this. A hub is required to upgrade. See #20545