Koenkk/zigbee2mqtt #16929: Cannot recall hue effects in device or group scenes

gdgib, 2023-03-05:
### What happened?

I added two scenes using the hue candle effect.  One scene to a group containing a single hue bulb, the other scene to the bulb itself.  When recalling the scenes (either one) the brightness and state of the bulb are set correctly.  Neither scene causes the effect to start again.

### What did you expect to happen?

I expected the effect to be applied to the bulb, when recalling either scene.

According to the CLIP v2 API (https://developers.meethue.com/develop/hue-api-v2/api-reference/#resource_scene_post) it should be possible to set effects in scenes (at least candle and fire).  What's more I've verified that the candle effect can be stored to multiple lights in a hue room (presumably implemented using a zigbee group) and recalled using the hue hub & app.  This definitely works, and there is no visible popcorn effect, suggesting the effect can be applied by the scene recall.

### How to reproduce it (minimal and precise)

* I have a hue bulb - https://www.zigbee2mqtt.io/devices/9290024687.html
* I have created a "Test Lights" group including only that bulb (endpoint 11) through the frontend
* I have created two scenes, one for the bulb, one for the group (see below for the JSON) by following https://www.zigbee2mqtt.io/guide/usage/scenes.html#creating-a-scene
* When recalling either scene, the state and brightness change, but **not the effect**

```
{
    "scene_add": {
        "ID": 0,
        "name": "Candle Example",
        "transition": 3,
        "state": "ON",
        "brightness": 64,
        "effect": "candle"
    }
}
```

To be sure, I also manually set this scene for the bulb through the UI, and then stored it.  Just like the above scenes created by adding the scenes, this third scene created by storing the scene does **NOT** apply the effect.

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

20220219

### Adapter

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-poe-2022

### Debug log

```
info  2023-03-05 09:37:46: Succesfully changed options
info  2023-03-05 09:37:46: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"rzpkc-1"}'
debug 2023-03-05 09:38:03: Received MQTT message on 'zigbee2mqtt/Test Lights/set' with data '{"scene_recall":0}'
debug 2023-03-05 09:38:03: Publishing 'set' 'scene_recall' to 'Test Lights'
debug 2023-03-05 09:38:03: Requested color_temp 582 is higher than maximum supported 500, using maximum!
info  2023-03-05 09:38:03: Successfully recalled group scene
info  2023-03-05 09:38:03: MQTT publish: topic 'zigbee2mqtt/Test Lights', payload '{"brightness":64,"color":{"h":243,"hue":243,"s":100,"saturation":100,"x":0.1603,"y":0.0513},"color_mode":"xy","color_temp":500,"state":"ON"}'
info  2023-03-05 09:38:03: MQTT publish: topic 'zigbee2mqtt/Hue Z2M Test Bulb', payload '{"brightness":64,"color":{"h":243,"hue":243,"s":100,"saturation":100,"x":0.1603,"y":0.0513},"color_mode":"xy","color_temp":500,"last_seen":"2023-03-05T09:28:39-08:00","linkquality":72,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16784922,"latest_version":16784922,"state":"idle"},"update_available":null}'
debug 2023-03-05 09:38:11: Received MQTT message on 'zigbee2mqtt/Hue Z2M Test Bulb/set' with data '{"scene_recall":0}'
debug 2023-03-05 09:38:11: Publishing 'set' 'scene_recall' to 'Hue Z2M Test Bulb'
info  2023-03-05 09:38:11: MQTT publish: topic 'zigbee2mqtt/Hue Z2M Test Bulb', payload '{"brightness":64,"color":{"h":243,"hue":243,"s":100,"saturation":100,"x":0.1603,"y":0.0513},"color_mode":"xy","color_temp":500,"last_seen":"2023-03-05T09:38:11-08:00","linkquality":10,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16784922,"latest_version":16784922,"state":"idle"},"update_available":null}'
debug 2023-03-05 09:38:11: Requested color_temp 582 is higher than maximum supported 500, using maximum!
info  2023-03-05 09:38:11: Successfully recalled scene
info  2023-03-05 09:38:11: MQTT publish: topic 'zigbee2mqtt/Hue Z2M Test Bulb', payload '{"brightness":64,"color":{"h":243,"hue":243,"s":100,"saturation":100,"x":0.1603,"y":0.0513},"color_mode":"xy","color_temp":500,"last_seen":"2023-03-05T09:38:11-08:00","linkquality":10,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16784922,"latest_version":16784922,"state":"idle"},"update_available":null}'
debug 2023-03-05 09:38:17: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"rzpkc-2"}'
info  2023-03-05 09:38:17: Succesfully changed options
info  2023-03-05 09:38:17: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"rzpkc-2"}'
```

gdgib, 2023-03-05:
Please note, I did also look at https://github.com/Koenkk/zigbee2mqtt/issues/15997.  However, that issue applies to setting effects on groups directly without the use of a scene.

While in a comment on that issue I note  that the Hue v2 API doesn't support setting effects on light groups, the Hue v2 API absolutely DOES support setting effects on scenes (https://developers.meethue.com/develop/hue-api-v2/api-reference/#resource_scene_post).

What's more, using the hue app allows me to create a scene with all of the lights set to the candle effect (I have to set them to candle one by one), save the scene, and then recall the scene and have the effect start without noticeably popcorn effects.

gdgib, 2023-04-05:
Still appears to be an issue.  Would love some help...

closed 2023-05-13 by github-actions[bot]

OUARZA, 2024-01-01:
Hello @gdgib ,
did you find a solution that seems to still be present.
Mathieu

gdgib, 2024-01-25:
Can confirm this is still broken.  I haven't forgotten, but obviously I haven't gotten time either.