Koenkk/zigbee2mqtt #16932: HAOS Addon 1.30.2-1 broken for MOES touch dimmer switch (Tuya TS0601_dimmer_1)

ash286, 2023-03-05:
### What happened?

Afer updating Zigbee2mqtt version from 1.30.1-1 to 1.30.2-1, it broke the communication with MOES touch dimmer switch (Tuya TS0601_dimmer_1) and the dimmer switch become unresponsive. It does not recognize the switch and does not complete interview upon re-adding to the Z2M network.

Upon rolling back to Z2M version 1.30.1-1, the switch has become normal so I can confirm it's the version 1.30.2-1 that has a bug!

### What did you expect to happen?

MOES touch dimmer switch did work fine before the update!! 

### How to reproduce it (minimal and precise)

Use Z2M version 1.30.2-1 and control the switch power on/off or change brightness (nothing works). However, it does take updates on switch status when physically power on/off switch but does not respond to any payloads. 

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 (SONOFF ZB Dongle -E)

veveme, 2023-03-06:
A lot of topics with dimmers recently , hope there will be any hotfix soon

ash286, 2023-03-08:
Yes, hope it will be fixed atleast in a next update. I have just rolled back Z2M to 1.30.1-1 for now.

ash286, 2023-03-08:
> A lot of topics with dimmers recently , hope there will be any hotfix soon

Yes, hope it will be fixed in a next update. I have just rolled Z2M back to previous version 1.30.1-1 for now.

Koenkk, 2023-03-08:
Can you provide the data/database.db entry of your device?

ash286, 2023-03-08:
@Koenkk Thank you for loonking into it. Yes, please see it below (ieeeAddr is changed). Let me know if you want anymore details. 

`{"id":3,"type":"Router","ieeeAddr":"0xa4c0000000a0000b","nwkAddr":27110,"manufId":4417,"manufName":"_TZE200_ip2akl4w","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[3,1,60237,0],"outClusterList":[17,4],"clusters":{"genBasic":{"attributes":{"65503":"|��+i","65506":32,"65508":0,"stackVersion":0,"dateCode":"","appVersion":64}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678295446916,"defaultSendRequestWhen":"immediate"}`

While using z2m 1.30.2-1, I was getting the following error when  I forced removed and tried to re-pair the switch

`info  2023-03-04 00:00:55: Zigbee: allowing new devices to join.
info  2023-03-04 00:00:55: Succesfully changed options
info  2023-03-04 00:00:55: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"lkr6e-1"}'
warn  2023-03-04 00:01:06: Device '0xa4c0000000a0000b' left the network
info  2023-03-04 00:01:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c0000000a0000b","ieee_address":"0xa4c0000000a0000b"},"type":"device_leave"}'
info  2023-03-04 00:01:13: Device '0xa4c0000000a0000b' joined
info  2023-03-04 00:01:13: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c0000000a0000b","ieee_address":"0xa4c0000000a0000b"},"type":"device_joined"}'
info  2023-03-04 00:01:13: MQTT publish: topic 'homeassistant/light/0xa4c0000000a0000b/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c0000000a0000b/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xa4c0000000a0000b/set","device":{"identifiers":["zigbee2mqtt_0xa4c0000000a0000b"],"manufacturer":"TuYa","model":"1 gang smart dimmer (TS0601_dimmer_1)","name":"0xa4c0000000a0000b"},"name":"0xa4c0000000a0000b","schema":"json","state_topic":"zigbee2mqtt/0xa4c0000000a0000b","unique_id":"0xa4c0000000a0000b_light_zigbee2mqtt"}'
info  2023-03-04 00:01:13: MQTT publish: topic 'homeassistant/sensor/0xa4c0000000a0000b/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c0000000a0000b/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c0000000a0000b"],"manufacturer":"TuYa","model":"1 gang smart dimmer (TS0601_dimmer_1)","name":"0xa4c0000000a0000b"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","name":"0xa4c0000000a0000b power on behavior","state_topic":"zigbee2mqtt/0xa4c0000000a0000b","unique_id":"0xa4c0000000a0000b_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2023-03-04 00:01:13: MQTT publish: topic 'homeassistant/select/0xa4c0000000a0000b/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c0000000a0000b/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c0000000a0000b/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0xa4c0000000a0000b"],"manufacturer":"TuYa","model":"1 gang smart dimmer (TS0601_dimmer_1)","name":"0xa4c0000000a0000b"},"entity_category":"config","icon":"mdi:power-settings","name":"0xa4c0000000a0000b power on behavior","options":["off","previous","on"],"state_topic":"zigbee2mqtt/0xa4c0000000a0000b","unique_id":"0xa4c0000000a0000b_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2023-03-04 00:01:13: MQTT publish: topic 'homeassistant/sensor/0xa4c0000000a0000b/countdown/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c0000000a0000b/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c0000000a0000b"],"manufacturer":"TuYa","model":"1 gang smart dimmer (TS0601_dimmer_1)","name":"0xa4c0000000a0000b"},"enabled_by_default":false,"name":"0xa4c0000000a0000b countdown","state_topic":"zigbee2mqtt/0xa4c0000000a0000b","unique_id":"0xa4c0000000a0000b_countdown_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.countdown }}"}'
info  2023-03-04 00:01:13: MQTT publish: topic 'homeassistant/number/0xa4c0000000a0000b/countdown/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c0000000a0000b/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c0000000a0000b/set/countdown","device":{"identifiers":["zigbee2mqtt_0xa4c0000000a0000b"],"manufacturer":"TuYa","model":"1 gang smart dimmer (TS0 [... cut ...]
error 2023-03-04 00:01:54: Failed to configure '0xa4c0000000a0000b', attempt 3 (Error: Read 0xa4c0000000a0000b/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18615 - 1 - 26 - 0 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7))`

Koenkk, 2023-03-09:
I've checked the converter but nothing has changed between 1.30.1 and 1.30.2. Therefore I expect this is caused by a change to the EZSP adapter integration. @kirovilya do you know if this is fixed in dev already?

kirovilya, 2023-03-10:
I'm not sure if this has been fixed, but need check. @ash286  can you check the work with the dimmer on the dev/edge version?
I don't even understand what the problem is. need [hardsman logs](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging).

ash286, 2023-03-12:
OK. Thank you for looking into this. I will check it with edge version or try to get hardsman logs next weekend and get back here. Funny enough, it does not have any problem with 1.30.1 so I am not sitting in a dark at the moment :)

ash286, 2023-04-08:
Hello @Koenkk and @kirovilya, apologies for the delay but I can confirm that this issue is not been fixed with even latest version 1.30.3-1. I have now generated hardsman logs from both versions (wroking and non-working) and tried to operate the dimmer from HA in both the case. Also, funny thing is that thogh the dimmer does on operate from HA dashboard using non-werking version 1.30.3-1, it does receive payload messages and change lighting status in HA dashboard button when the dimmer was operated physically. Since, hardsman logs are larger (>3000 lines in both logs), I would prefer to sent them to you via an email or some other private and direct means. Would you please send me your emails to send you logs?

P.S.: Oh, one strange thing I noticed is that restoring backup of 1.30.1-1 generated during update to 1.30.3-1 does not work. I need to restore it from revious back up of 1.30.1-1.

kirovilya, 2023-04-10:
@ash286 you can make an archive with logs and attach it here in the issue.

ash286, 2023-05-09:
@kirovilya Alright sorry again for the delay but please find the attached hardsman logs below. I can also confirm that the issue has remained unfixed even in the latest version 1.30.4-1. 

is it normal for ezsp to show radioTxPower=5? I did set it to 20 from the settings but it still shows 5!

[z2m-NOT-working_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/11436644/z2m-NOT-working_log.txt)
[z2m-working_log14.txt](https://github.com/Koenkk/zigbee2mqtt/files/11436646/z2m-working_log14.txt)

ash286, 2023-05-10:
I am sorry @Koenkk @kirovilya, I forgot to mention thaat the IEEE for TS0601 dimmer is 0xa7c3902891d3456a in above logs.

ash286, 2023-05-28:
Any luck @Koenkk @kirovilya with finding a bug? I hope you have some time to look at the hardsman logs. Looking forward to it becuase I'm still forced to use Z2M version 1.30.1-1.

closed 2023-07-06 by github-actions[bot]

ash286, 2023-08-15:
Hi @Koenkk @kirovilya I'm wondering if you find some time to look at my issue or help me with some clue to resolve this issue. I very much appreciate if would chat one to one and help solve this issue. I tried updating HA with latest Z2M 1.32.2-1 and issue is still remained. PLEASE PLEASE HELP

ash286, 2023-08-15:
> It's amazing how long you wait for help... have you tried something else? update stick firmware (my recomendation 7.1.1)? update z2m to latest version?

Oh great you replied. Many thanks for your quick response. I was just in a process of writing a new issue lol. I will certainly give a try with updating the stick firmware.

Latest z2m doesn't work, I just tried it few days ago in a hope to get it worked. Also, simply rolling back to 1.30.1-1 doen not work untill I fully reboot RPi4.

ash286, 2023-08-15:
Stick is connected via shielded usb 2.0 extension cable.
SSD is connected to usb3, but this was always been like there.

kirovilya, 2023-08-15:
move your usb3 device as far away from the stick as possible

kirovilya, 2023-08-15:
https://github.com/Koenkk/zigbee2mqtt/assets/8360230/c9909658-73c1-4277-abb3-13840f036e3b

ash286, 2023-08-15:
Yes, it is already far away. I am not sure if that interference would partially stop responding the switch in newer versions but not in all the versions before 1.30.1-1.

I'm using ZB dongle-E and has a latest firmware v6.10.3 not 7.1.1 right?

ash286, 2023-08-15:
I think the stick has a latest firmware 6.10.3.0 build 297 and I just found that 7.1.x is still alpha version and there is no way to downgrade/rollback option to 6.10.3 :(

Anymore suggestions please?

kirovilya, 2023-08-15:
no, the latest version is 7.3.1
version 7.1.1 has already been forgotten, but I consider it stable :)
try to update and check the operation of the devices.
[ncp-uart-sw_7.1.1.0_115200.zip](https://github.com/Koenkk/zigbee2mqtt/files/12346225/ncp-uart-sw_7.1.1.0_115200.zip)

I also consider the code for "ezsp" in z2m to be stable

ash286, 2023-08-15:
Alright. Thanks for the links for the firmware and instructions. I will update now and get back to you :)

ash286, 2023-08-16:
Here is an update: Yesterday I managed to update the stick firmware to MultiPAN and then to 7.1.1 but this dimmer switch didn't work with either (MultiPAN didn't work at all with all Zigbee devices). I tried then to update it with the latest version 7.3.1 but then stick got broke. Now stick doesn't respond in bootloader mode (SecureCRT) or Web Flasher :(

I have ordered new stick now, will update it to 7.3.1 to see if that works. Let me know @kirovilya any better suggestion please

ash286, 2023-08-16:
Okay. It's been very funny. Yesterday I tried bootloader mode with secureCRT but it was not getting into bootloader after pressing 1. I tried that couple of times and not worked but I could manager to update the firm to 7.1.1 agian.  But z2m now shows following error. Do I supposed to upload nvm3_initfile.gbl after uploading ncp-uart-sw_7.1.1.0_115200.gbl?

**Z2M Error:**
[12:30:58] INFO: Preparing to start...
[12:30:58] INFO: Socat not enabled
[12:31:00] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-08-16 12:31:07: Logging to console and directory: '/config/zigbee2mqtt/log/2023-08-16.12-31-07' filename: log.txt
Zigbee2MQTT:info  2023-08-16 12:31:07: Starting Zigbee2MQTT version 1.32.2 (commit #unknown)
Zigbee2MQTT:info  2023-08-16 12:31:07: Starting zigbee-herdsman (0.17.2)
Zigbee2MQTT:error 2023-08-16 12:31:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-08-16 12:31:27: Failed to start zigbee
Zigbee2MQTT:error 2023-08-16 12:31:27: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-08-16 12:31:27: Exiting...
Zigbee2MQTT:error 2023-08-16 12:31:27: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:551:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:349:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:59:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

kirovilya, 2023-08-16:
> Do I supposed to upload nvm3_initfile.gbl after uploading ncp-uart-sw_7.1.1.0_115200.gbl?

yes, if not start

kirovilya, 2023-08-16:
I don’t understand something, after upload 7.1.1 z2m doesn’t start?
What does it show in the logs during and after flashing?

Previously, I flashed the utility from Elelabs. `probe` command showed the current firmware
https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

ash286, 2023-08-16:
I don't know either. I'm using windows 10 at the moment so using secureCRT would be easier. Here are trace log of secureCRT:

```
begin upload
CCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCCC
Starting xmodem transfer.  Press Ctrl+C to cancel.[SERIAL-TRACE] : Waiting for 0 transmit operations to complete...
[SERIAL-TRACE] : Enabling binary mode...
[SERIAL-TRACE] : SERIAL-CONFIG: BAUD-RATE=115200
[SERIAL-TRACE] : SERIAL-CONFIG: PARITY-HONORED=no
[SERIAL-TRACE] : SERIAL-CONFIG: PARITY=none
[SERIAL-TRACE] : SERIAL-CONFIG: STOP-BITS=1
[SERIAL-TRACE] : SERIAL-CONFIG: DATA-BITS=8
[SERIAL-TRACE] : SERIAL-CONFIG: DSR-OUTPUT-FLOW-HONORED=no
[SERIAL-TRACE] : SERIAL-CONFIG: DTR-CONTROL=enable/high/on
[SERIAL-TRACE] : SERIAL-CONFIG: CTS-OUTPUT-FLOW-HONORED=no
[SERIAL-TRACE] : SERIAL-CONFIG: RTS-CONTROL=enable/high/on
[SERIAL-TRACE] : SERIAL-CONFIG: USE XON-XOFF=yes

Transferring nvm3_initfile.gbl...
    0%       0 KB     128 bytes/sec 00:00:00 ETA   0 Errors  [SERIAL-TRACE] : COMM-EVENT [tx]
```

it stays at 0% even after 10-15 minutes.

ash286, 2023-08-16:
I am getting a strong feeling feeling that 7.1.1 is no good firmware and no longer allow to upload other firmware nor nvm3_initfile.gbl

kirovilya, 2023-08-16:
ok :) then return the original firmware

ash286, 2023-08-16:
this is what I'm strugling about. How can I return to the original firmware or any other firmware other than 7.1.1? if you have 7.1.1 installed in your stick, have you tried changing from 7.1.1 to any other firmware (not multipan)?

Also, only 7.1.1 has a gbl file of size 260KB while all other firmware has a file size of 7KB!

ash286, 2023-08-16:
Belive me, I tried everything but I can flash only 7.1.1. or Multipan firmware but not any other firmware versions regardless new or older than 7.1.1

Would you spare 10-15 minutes over zoom/Teams/Meet? That would be great.

ash286, 2023-08-16:
Okay, I managed to make the stick working with Multipan firmware and below are firmware and software versions using .

```
Zigbee2MQTT version
    [1.32.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.32.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    EZSP v12

Coordinator revision
    7.3.1.0 build 0

Coordinator IEEE Address
    0xe0798dfffca485a1

Frontend version
    0.6.133
```

Then I tried a brand new ZBdongle-E with the original firmware and latest z2m with below specs,

```
Zigbee2MQTT version
    [1.32.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.32.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type
    EZSP v8

Coordinator revision
    6.10.3.0 build 297

Coordinator IEEE Address
    0xdc8e95fffb49a1f9

Frontend version
    0.6.133
```

I reset the network completely and paird two dimmers of the same model and it turned out that the both dimmers got paired but it now shows as unsupported devices (see the pic). So I think this is the issue that somehow they were fine and supported in the z2m version 1.30.1-1 and all previous z2m versions. Would you be able to look up for all supported devices in the version 1.30.1-1 and related changes in newer versions e.g. 1.30.1-2 and newer?

[image]

kirovilya, 2023-08-17:
I don’t see the name of the model in the picture, which means that during pairing it was not possible to read it - this once again tells me about problems with the radio, interference.
1) try not to use usb3
2) try to use the latest version of z2m (now 1.32.2)

ash286, 2023-08-17:
I don't think this is related the radio interference. That switch has LQI of 159 and it works absolutely fine whatsoever without any problem when I use z2m version 1.30.1-1 and of course while having ssd connected to usb3. Aslo when the switch was successfully paired and working in z2m 1.30.1-1 and if upgrade any newer version,  switch does immediately update its status in HA but it does not accept any command from HA, e.g. when I physically touch dimmer to turn on light the interface would show up status ON and similarly change to status OFF but I can not control it from the interface. How would you explain that if it is happening due to radio interference?

ash286, 2023-08-17:
I'm happy to buy you the dimmer swtich and you can show it working with the latest z2m 1.32.2 :))))

kirovilya, 2023-08-18:
write down the logs of the pairing process.
you have some errors when pairing the device - that's why it is shown as unsupported.

Do you happen to have a ZHA or Multiprotocol addon running in parallel? It can access the same port as specified in z2m -  must be disabled

ash286, 2023-08-21:
@kirovilya clearly it's not gonna work as the new z2m version has some issue with this dimmer switch. 

Here is the update: I coudn't remove 7.1.1 (I tried every flashing mehtod: webflasing, secureCRT and pythong scripts) and I fould out many people have ended up ino that non-resolvable problem ([problem mentioned-1](https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-v2-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-soc-mcu/442695/64?u=ashrad), [problem mentioned-2](https://github.com/Koenkk/zigbee2mqtt/discussions/13373#discussioncomment-3974926), [problem mentioned-3](https://github.com/Koenkk/zigbee-herdsman/issues/319#issuecomment-1674295841) ).

Config of the stick having non-replaceable firmware 7.1.1:

```
**CONFIG_PACKET_BUFFER_COUNT=254**
CONFIG_NEIGHBOR_TABLE_SIZE=26
**CONFIG_APS_UNICAST_MESSAGE_COUNT=64**
CONFIG_BINDING_TABLE_SIZE=32
**CONFIG_ADDRESS_TABLE_SIZE=8
CONFIG_MULTICAST_TABLE_SIZE=8**
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
**CONFIG_SOURCE_ROUTE_TABLE_SIZE=128**
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=12
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_END_DEVICE_BIND_TIMEOUT=60
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_APPLICATION_ZDO_FLAGS=0
**CONFIG_BROADCAST_TABLE_SIZE=128
CONFIG_MAC_FILTER_TABLE_SIZE=2
CONFIG_SUPPORTED_NETWORKS=2**
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=0
CONFIG_ZLL_RSSI_THRESHOLD=216
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
**CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=122**
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=128
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
CONFIG_GP_PROXY_TABLE_SIZE=5
EZSP_CONFIG_GP_SINK_TABLE_SIZE=0
```

Above stick config clearly match exactly with the config mentioned in the post ( [problem mentioned-3](https://github.com/Koenkk/zigbee-herdsman/issues/319#issuecomment-1674295841) ). While a brand new working stick having firmware factory v6.10.3 have a config as below:

```
**CONFIG_PACKET_BUFFER_COUNT=255**
CONFIG_NEIGHBOR_TABLE_SIZE=26
**CONFIG_APS_UNICAST_MESSAGE_COUNT=20**
CONFIG_BINDING_TABLE_SIZE=32
**CONFIG_ADDRESS_TABLE_SIZE=16
CONFIG_MULTICAST_TABLE_SIZE=16**
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
**CONFIG_SOURCE_ROUTE_TABLE_SIZE=200**
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=12
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_END_DEVICE_BIND_TIMEOUT=60
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_APPLICATION_ZDO_FLAGS=0
**CONFIG_BROADCAST_TABLE_SIZE=15
CONFIG_MAC_FILTER_TABLE_SIZE=15
CONFIG_SUPPORTED_NETWORKS=1**
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=0
CONFIG_ZLL_RSSI_THRESHOLD=216
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
**CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=9**
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=128
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
CONFIG_GP_PROXY_TABLE_SIZE=5
EZSP_CONFIG_GP_SINK_TABLE_SIZE=0
```

With the new stick I could use the latest firmware v7.3.1 and connected all MOES dimmers using z2m 1.30.1-1 without any problem and they are working absolutely normal. However, as soon as I update the latest z2m 1.32.2.1. I think the reason 7.1.1 can't be replaed with other firmware becuase of the diffferecen in above config. Can you @Koenkk @kirovilya  help me geting the configs back to factory setting?

kirovilya, 2023-08-22:
I see no reason why the dimmer stops working when updating z2m... maybe the converters have changed...
What is your dimmer model, "manufName":"_TZE200_ip2akl4w" ?

ash286, 2023-08-22:
> I see no reason why the dimmer stops working when updating z2m... maybe the converters have changed... What is your dimmer model, "manufName":"_TZE200_ip2akl4w" ?

Yes, that the correct dimmer model. See the fresh pic below while it is working with z2m 1.30.1-1

[image]

kirovilya, 2023-08-22:
Can you record herdsman log for 1.32.2 version? 
Аnd for the pairing process too.

I don't see dimmer-errors in your old logs https://github.com/Koenkk/zigbee2mqtt/issues/16932#issuecomment-1540975162