Koenkk/zigbee2mqtt #16934: Gledopto GL-C-007 not supported

xavierchan, 2023-03-06:
### What happened?

Gledopto GL-C-007 with firmware build on 20230220, it's not supported

### What did you expect to happen?

Device Gledopto GL-C-007 can be supported.

### How to reproduce it (minimal and precise)

Just use the latest Gledopto GL-C-007

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

20221226

### Adapter

cc2538

### Debug log

```shell
Zigbee2MQTT:warn  2023-03-06 18:06:39: Received message from unsupported device with Zigbee model 'GL-C-007' and manufacturer name 'GLEDOPTO'
```

Koenkk, 2023-03-06:
Is this the GL-C-007-1ID (white and color together) or GL-C-007-2ID (white and color separate) version?

xavierchan, 2023-03-07:
> Is this the GL-C-007-1ID (white and color together) or GL-C-007-2ID (white and color separate) version?

The old version we used before is GL-C-007-1ID, but new device can not identify it.

Koenkk, 2023-03-07:
Can you provide the data/database.db entry of this device?

xavierchan, 2023-03-08:
> Can you provide the data/database.db entry of this device?

Of course. Two device's entry is this.
```json
{"id":4,"type":"Router","ieeeAddr":"0xa4c1389db6f37e0f","nwkAddr":55155,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20230220","swBuildId":"20051203","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1678240479499,"defaultSendRequestWhen":"immediate"}
{"id":5,"type":"Router","ieeeAddr":"0x00124b00226fbc6f","nwkAddr":35449,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-007","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-007","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"3.0.0"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":155,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":528,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"3.0.0","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1678096721875,"defaultSendRequestWhen":"immediate"}
```

You can see, 0xa4c1389db6f37e0f is the new one, 0x00124b00226fbc6f is the old device we use before.

Koenkk, 2023-03-08:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-08 by Koenkk