Koenkk/zigbee2mqtt #16935: Light / Button Binding works the other way

Crumar, 2023-03-06:
### What happened?

Sonoff ZBMini is bound to group of Ikea GU 10 Lights, which are not attached to the ZBMini, but permanent Live wire.
[image]
zm_br = light group
buero spots schalter = ZBMini

When I operate the wall switches attached to the ZBMini, nothing happens (except relay click).
When I toggle the light group from Home Assistant, the relay clicks and state changes in Z2M GUI.

### What did you expect to happen?

I expected that toggling the wall switch will turn the lights on/off.
I did not expected the light turning of or on to toggle the relay.

### How to reproduce it (minimal and precise)

Create OnOff Reporting on ZBMini
Bind OnOff to Light Group
Unbind OnOff from Coordinator
Toggle Light Group from Home Assisstant

### Zigbee2MQTT version

1.30.1 commit: eb878d3d

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

Logs:
[image]

Crumar, 2023-04-05:
Anything that I could test?

closed 2023-05-13 by github-actions[bot]

ulysserevient, 2024-01-16:
Hi,
I've you fond a solution to this problem ?
Thx

Crumar, 2024-01-16:
No, but tested again last week, still same issue

ulysserevient, 2024-01-16:
ok, but have you find a module that support binding ?

Crumar, 2024-01-16:
Sadly, not.