Koenkk/zigbee2mqtt #16941: Sonoff S40 Lite Power On State UNSUPPORTED_ATTRIBUTE

ptr727, 2023-03-06:
### What happened?

I installed a [Sonoff S40 Lite](https://www.zigbee2mqtt.io/devices/S40ZBTPB.html#sonoff-s40zbtpb) in a cabinet to reset my AV equipment when it fails to work properly.  
I use HA to configure the plug power on state to be "on".  
Had a brief power failure last night, and this morning AV equipment was off.

Went into Z2M, set power to on.  
And then set power on behavior to on, and got an error:

```
Error 2023-03-06 07:48:23Publish 'set' 'power_on_behavior' to 'Sonoff-Plug-79B00F' failed: 'Error: Write 0x00124b002579b00f/1 genOnOff({"startUpOnOff":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

### What did you expect to happen?

Set the power on state to the desired value.

### How to reproduce it (minimal and precise)

Exposes power_on_behavior set to on.

### Zigbee2MQTT version

1.30.2 commit: cdf62ea

### Adapter firmware version

20221226

### Adapter

tubezb-cc2652p2-ethusb-2022

### Debug log

[z2mlog.log](https://github.com/Koenkk/zigbee2mqtt/files/10900370/z2mlog.log)

Koenkk, 2023-03-06:
Doesn't seem to be supported for the device, I've disabled it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-06 by Koenkk

scyto, 2023-03-16:
The database should also be updated IMO to be clear this is not the S40ZBTPB but actually the S40ZBTPB Lite
It is unclear to me if the non-lite version was ever released, but it is on their website (it supports power monitoring)