Koenkk/zigbee2mqtt #16942: Zemismart ZM25RX-08/30 no battery info

DariBer, 2023-03-06:
### What happened?

I have a tubular motor from Zemismart (see bellow) everything works fine, but I can't get the battery info to work, I just get a question mark.
Manufacturer: Zemismart
Model: ZM25RX-08/30
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_7eue9vhc

### What did you expect to happen?

Show the battery info in procent

### How to reproduce it (minimal and precise)

It's always like this

### Zigbee2MQTT version

1.30.2-1

### Adapter firmware version

 20220219

### Adapter

Sonoff ZBDongle-P

leo-na-rd, 2023-04-09:
got the same issue, kinda annoying.
Whoever can fix this, you're a hero!

nimmerpeople, 2023-05-01:
Same issue on a Zemimsmart TQL25 blind 
TS0601
by _TZE200_7eue9vhc

Koenkk, 2023-05-01:
To add support for battery %, we need to know the DP of this. (https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

chris1h1, 2023-08-16:
Hello,

Apologies for resurrecting a stale issue. I bought a Tuya Zigbee gateway and followed the instructions at [https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html); the resultant data follows:

`{
    "1": "Control",
    "101": "设置限位点",
    "106": "行程状态",
    "11": "Situation_set",
    "12": "Fault",
    "13": "Battery Percentage",
    "2": "Curtain position setting",
    "3": "Current curtain position",
    "4": "Mode",
    "5": "Motor Direction",
    "7": "Work State"
}`

Per Google Translate:

- 设置限位点 = set limit point
- 行程状态 = trip status

I see documentation regarding adding a completely new device, but I'm not sure how to best go about modifying an existing device; any pointers would be gratefully received.

Thanks.

DariBer, 2023-08-16:
how can we re-open this case so someone can try to ad the DP that @chris1h1 found out?

nimmerpeople, 2023-08-16:
Yes please. I never did work out how to this for mine and would love for it to work 

Thank you
:-)

chris1h1, 2023-08-16:
I tried to create an external converter definition for this already-supported device based on the various examples, with the understanding that the local external converter definition will take precedence over the delivered definition, but I'm unable to start Zigbee2MQTT with this in place:

```
Aug 16 11:45:08 zigbee2mqtt systemd[1]: Started zigbee2mqtt.
Aug 16 11:45:08 zigbee2mqtt npm[2599]: > zigbee2mqtt@1.32.1 start
Aug 16 11:45:08 zigbee2mqtt npm[2599]: > node index.js
Aug 16 11:45:09 zigbee2mqtt npm[2611]: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/tuya.js:312
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     return manufacturerNames.map((manufacturerName) => {
Aug 16 11:45:09 zigbee2mqtt npm[2611]:                              ^
Aug 16 11:45:09 zigbee2mqtt npm[2611]: TypeError: manufacturerNames.map is not a function
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at Object.fingerprint (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:308:30)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at /opt/zigbee2mqtt/data/extension/externally-loaded.js:43:23
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at Script.runInContext (node:vm:141:12)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at Script.runInNewContext (node:vm:146:17)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at Object.runInNewContext (node:vm:306:38)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at loadModuleFromText (/opt/zigbee2mqtt/lib/util/utils.ts:150:8)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at loadModuleFromFile (/opt/zigbee2mqtt/lib/util/utils.ts:157:12)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at Object.getExternalConvertersDefinitions (/opt/zigbee2mqtt/lib/util/utils.ts:167:25)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at getExternalConvertersDefinitions.next (<anonymous>)
Aug 16 11:45:09 zigbee2mqtt npm[2611]:     at new ExternalConverters (/opt/zigbee2mqtt/lib/extension/externalConverters.ts:12:20)
```

Perhaps my code has errors though. It's a mashup of [https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts), [https://github.com/Koenkk/zigbee2mqtt/issues/14571](https://github.com/Koenkk/zigbee2mqtt/issues/14571), and [https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html) (I have no idea what I'm doing):

```javascript

// Based on:
// https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts
// https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device

/*
Put this in /opt/zigbee2mqtt/data/ZM25RX-08.js
Edit configuration.yaml, add:
external_converters:
  - ZM25RX-08.js
*/

/*
From https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html:
{
    "1": "Control",
    "101": "设置限位点",
    "106": "行程状态",
    "11": "Situation_set",
    "12": "Fault",
    "13": "Battery Percentage",
    "2": "Curtain position setting",
    "3": "Current curtain position",
    "4": "Mode",
    "5": "Motor Direction",
    "7": "Work State"
}

设置限位点 = set limit point
行程状态 = trip status
*/

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: tuya.fingerprint('TS0601', '_TZE200_7eue9vhc'),
    model: 'TS0601_cover_battery',
    vendor: 'TuYa',
    description: 'Tubular motor',
    whiteLabel: [

        tuya.whitelabel('Zemismart', 'ZM25RX-08/30', 'Tubular motor (battery)', '_TZE200_7eue9vhc')
    ],
    fromZigbee: [legacy.fromZigbee.tuya_cover, fz.ignore_basic_report, tuya.fz.datapoints],
    toZigbee: [legacy.toZigbee.tuya_cover_control, legacy.toZigbee.tuya_cover_options, tuya.tz.datapoints],
    exposes: [
        e.battery(),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.composite('options', 'options', ea.STATE_SET)
        .withFeature(e.numeric('motor_speed', ea.STATE_SET)
            .withValueMin(0).withValueMax(255).withDescription('Motor speed'))
        .withFeature(e.binary('reverse_direction', ea.STATE_SET, true, false)
            .withDescription('Reverse the motor direction'))
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [13, 'battery', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

```

chris1h1, 2023-08-21:
@DariBer, @nimmerpeople: I've created [https://github.com/Koenkk/zigbee2mqtt/issues/18685](https://github.com/Koenkk/zigbee2mqtt/issues/18685) to resurrect this issue.